Hieromon/AutoConnect #100: Password protection / http auth

infiums, 2019-06-30:
Is there anyway to protect /_ac page with Server.authenticate(user, pass) and Server.requestAuthentication()?

Hieromon, 2019-07-04:
@infiums Sorry for my late reply.
As I prioritize other PR implementations now, I do not have enough time to examine your questions.
But roughly, your questions seem to contain useful hints that can improve the AutoConnect library.

I hasten to inform you about question2 as follows:
For now, there is no way other than using [retainPortal](https://hieromon.github.io/AutoConnect/apiconfig.html#retainportal) with [portalTimeout](https://hieromon.github.io/AutoConnect/apiconfig.html#portaltimeout) option to make the effect of handleClient() call effective in loop() while maintaining the SoftAP state.
A portalTimeout is a period of time to keep the captive portal state. While in Captive Portal state, AP_STA mode will be in effect, and the http request to detect the captive portal from the client device will be redirected to the AutoConnect menu by DNS hijacking.
The portalTimeout releases the captive portal state (stops the DNS server), but AutoConnect does not stop SoftAP if the retainPortal option is enabled. Documentation as [Inside AutoConnect::begin](https://hieromon.github.io/AutoConnect/lsbegin.html) describe its logic sequence.
Does the combination of these options meet your needs?

I will answer Question 1 within a few days, sorry to keep you waiting.

infiums, 2019-07-04:
AutoConenct

infiums, 2019-07-05:
@Hieromon, thank you for your response!
That seems to be a good solution for me! Not sure, that completely understand details and it is powerful, but it works. 
Great! Thank you!

Here example of code with portal running while loop() executing:

```
WebServer Server;
AutoConnect Portal(Server);
AutoConnectConfig Config;

void setup() {
  Config.portalTimeout = 15000; // Not sure the parameter to set
  Config.retainPortal = true;  
  Portal.config(Config);
  if (Portal.begin()) {
    Serial.println("Started, IP:" + WiFi.localIP().toString());
  }
}

void loop() {
  Serial.println("Loop with running portal");
  Server.handleClient();
  Portal.handleRequest();  
  if (WiFi.status() == WL_IDLE_STATUS) {
    ESP.restart(); delay(1000);
  }
  delay(1000);
}
```

Hieromon, 2019-07-06:
> `Config.portalTimeout = 15000; // Not sure the parameter to set`

Ok, I'll try to explain to you.
AutoConnect issues first WiFi.begin without SSID and password. WiFi.begin without SSID and password will try to connect to the SSID last connected. WiFi.begin will fail if there is no access point for that SSID nearby. Under that condition, You can find esp32ap in your smartphone's WiFi settings and tap it then the AutoConnect menu will pop up automatically. In the meantime, AutoConnect::begin does not exit. It does not return control to the sketch but executes WebServer::handleClient in itself.
The portalTimeout is the period when the auto pop-up is enabled. When the portalTimeout expires, AutoConnect::begin returns control to the sketch.

Hieromon, 2019-07-08:
@infiums I reviewed the current library code for adding authentication. 
I could not invent a way to add safely `WebServer::authenticate` and `WebServer::requestAuthentication` to each page placed in /_ac without modifying the library code.
Since AutoConnect leaves HTML generation and transmission to PageBuilder, there is a possibility that calling `WebServer::authenticate` method inside [PageBuilder](https://github.com/Hieromon/PageBuilder/blob/b5fe19c65e652bce7e48f262024fe72b8df0b5f1/src/PageBuilder.cpp#L102-L173) can apply as an effective way.
However, I will implement this change in the roadmap in the future and will not support it immediately.

closed 2019-07-26 by Hieromon