Hieromon/AutoConnect #105: /_ac results in blank page

Alives, 2019-07-12:
Using the latest version of all libraries.

It seems there might not be enough memory to build the page?  What are my options?

Sketch uses 540180 bytes (51%) of program storage space. Maximum is 1044464 bytes.
Global variables use 33732 bytes (41%) of dynamic memory, leaving 48188 bytes for local variables. Maximum is 81920 bytes.

Serial log shows:
[AC] Host:,URI:/_ac,generated:/_ac, allocated
[AC] Host:192.168.86.253,URI:/_ac,already allocated

Response:
<!DOCTYPE html><html><head><meta charset="UTF-8" name="viewport" content="width=device-width,initial-scale=1"><title>AutoConnect statistics</title><style type="text/css">html{font-family:Helvetica,Arial,sans-serif;font-size:16px;-ms-text-size-adjust:100%;-webkit-text-size-adjust:100%;-moz-osx-font-smoothing:grayscale;-webkit-font-smoothing:antialiased}body{margin:0;padding:0}.base-panel{margin:0 22px 0 22px}.base-panel>*>label{display:inline-block;width:3.0em;text-align:right}.base-panel>*>label.slist{width:auto;font-size:0.9em;margin-left:10px;text-align:left}input{-moz-appearance:none;-webkit-appearance:none;font-size:0.9em;margin:8px 0 auto}.lap{visibility:collapse}.lap:target{visibility:visible}.lap:target .overlap{opacity:0.7;transition:0.3s}.lap:target .modal_button{opacity:1;transition:0.3s}.overlap{top:0;left:0;width:100%;height:100%;position:fixed;opacity:0;background:#000;z-index:1000}.modal_button{border-radius:13px;background:#660033;color:#ffffcc;padding:20px 30px;text-align:center;text-decoration:none;letter-spacing:1px;font-weight:bold;display:inline-block;top:40%;left:40%;width:20%;position:fixed;opacity:0;z-index:1001}table{border-collapse:collapse;border-spacing:0;border:1px solid #ddd;color:#444;background-color:#fff;margin-bottom:20px}table.info,table.info>tfoot,table.info>thead{width:100%;border-color:#5C9DED}table.info>thead{background-color:#5C9DED}table.info>thead>tr>th{color:#fff}td,th{padding:10px 22px}thead{background-color:#f3f3f3;border-bottom:1px solid #ddd}thead>tr>th{font-weight:400;text-align:left}tfoot{border-top:1px solid #ddd}tbody,tbody>tr:nth-child(odd){background-color:#fff}tbody>tr>td,tfoot>tr>td{font-weight:300;font-size:.88em}tbody>tr:nth-child(even){background-color:#f7f7f7}table.info tbody>tr:nth-child(even){background-color:#EFF5FD}.luxbar-fixed{width:100%;position:fixed;top:0;left:0;z-index:1000;box-shadow:0 1px 3px rgba(0, 0, 0, 0.12), 0 1px 2px rgba(0, 0, 0, 0.24)}.luxbar-burger span,.luxbar-burger span::before,.luxbar-burger span::after{display:block;height:2px;width:26px;transition:0.6s ease}.luxbar-checkbox:checked~.luxbar-menu li .luxbar-burger span{background-color:transparent}.luxbar-checkbox:checked~.luxbar-menu li .luxbar-burger span::before,.luxbar-checkbox:checked~.luxbar-menu li .luxbar-burger span::after{margin-top:0}.luxbar-header{display:flex;flex-direction:row;justify-content:space-between;align-items:center;height:58px}.luxbar-menu-right .luxbar-burger{margin-left:auto}.luxbar-brand{font-size:1.6em;padding:18px 24px 18px 24px}.luxbar-menu{min-height:58px;transition:0.6s ease;width:100%}.luxbar-navigation{display:flex;flex-direction:column;list-style:none;padding-left:0;margin:0}.luxbar-menu a,.luxbar-item a{text-decoration:none;color:inherit;cursor:pointer}.luxbar-item{height:58px}.luxbar-item a{padding:18px 24px 18px 24px;display:block}.luxbar-burger{padding:18px 24px 18px 24px;position:relative;cursor:pointer}.luxbar-burger span::before,.luxbar-burger span::after{content:'';position:absolute}.luxbar-burger span::before{margin-top:-8px}.luxbar-burger span::after{margin-top:8px}.luxbar-checkbox{display:none}.luxbar-checkbox:not(:checked)~.luxbar-menu{overflow:hidden;height:58px}.luxbar-checkbox:checked~.luxbar-menu{transition:height 0.6s ease;height:100vh;overflow:auto}.dropdown{position:relative;height:auto;min-height:58px}.dropdown:hover>ul{position:relative;display:block;min-width:100%}.dropdown>a::after{position:absolute;content:'';right:10px;top:25px;border-width:5px 5px 0;border-color:transparent;border-style:solid}.dropdown>ul{display:block;overflow-x:hidden;list-style:none;padding:0}.dropdown>ul .luxbar-item{min-width:100%;height:29px;padding:5px 10px 5px 40px}.dropdown>ul .luxbar-item a{min-height:29px;line-height:29px;padding:0}@media screen and (min-width:768px){.luxbar-navigation{flex-flow:row;justify-content:flex-end;}.luxbar-burger{display:none;}.luxbar-checkbox:not(:checked)~.luxbar-menu{overflow:visible;}.luxbar-checkbox:checked~.luxbar-menu{height:58px;}.luxbar-menu .luxbar-item{border-top:0;}.luxbar-menu-right .luxbar-header{margin-right:auto;}.dropdown{height:58px;}.dropdown:hover>ul{position:absolute;left:0;top:58px;padding:0;}.dropdown>ul{display:none;}.dropdown>ul .luxbar-item{padding:5px 10px;}.dropdown>ul .luxbar-item a{white-space:nowrap;}}.luxbar-checkbox:checked+.luxbar-menu .luxbar-burger-doublespin span::before{transform:rotate(225deg)}.luxbar-checkbox:checked+.luxbar-menu .luxbar-burger-doublespin span::after{transform:rotate(-225deg)}.luxbar-menu-material,.luxbar-menu-material .dropdown ul{background-color:#263238;color:#fff}.luxbar-menu-material .active,.luxbar-menu-material .luxbar-item:hover{background-color:#37474f}.luxbar-menu-material .luxbar-burger span,.luxbar-menu-material .luxbar-burger span::before,.luxbar-menu-material .luxbar-burger span::after{background-color:#fff}</style></head><body style="padding-top:58px;"><div class="container">

Alives, 2019-07-12:
I read through the PageBuilder README and saw that the HTML is stored in SPIFFS, so I changed my flash to use 4M with 1M SPIFFS, but that didn't help.  I then tried a full flash, erase everything, but that didn't help (though it might not have worked, because it knew an SSID to connect to on reboot).

Hieromon, 2019-07-13:
AutoConnect does not use file input of PageBuilder. Your measures have no effect.
Use PageBuilder's debugging options to find out the available size of the heap.
https://hieromon.github.io/AutoConnect/faq.html#3-turn-on-the-debug-log-options

closed 2019-07-26 by Hieromon