Hieromon/AutoConnect #108: autoreconnect doesn't seem to be working

Alives, 2019-07-12:
After my captive portal times out, I'd expect a loop of WiFi.begin() -> Captive Portal again, then once an SSID is set, WiFi.begin().  I'm seeing the captive portal timeout, then nothing happens.

```
  config.apid = sensor_name.c_str();
  config.psk = F("");
  config.autoReconnect = true;
  config.portalTimeout = 60000;  // Timeout in 60s.
  portal.config(config);
  portal.begin();
```

Hieromon, 2019-07-13:
Once successfully connected to the SSID, AutoConnect loads and reconnects with its credential, even if disconnected once.
I can not reproduce the situation. autoReconnect works fine.

Please provide me a serial log based on the information written in the [documentation](https://hieromon.github.io/AutoConnect/faq.html#4-reports-the-issue-to-autoconnect-github-repository) to problem diagnostic.

Hieromon, 2019-07-26:
It closes because there is no answer. Please reopen if an answer occurs.

closed 2019-07-26 by Hieromon