Hieromon/AutoConnect #109: Notify application if Config Portal is opened?

HenrikSte, 2019-07-16:
Folks,
I just migrated from WifiManager to this awesome library as the mentioned one (no longer?) seems to work with ESP32.

There is one thing I am missing though: I want to notify my user that the portal is active and how to connect to it by displaying a text on an 2004 LCD. I show SSID and IP. In wifimanager there was a callback function I used.

Any thoughts?

THX!
Henrik

Hieromon, 2019-07-17:
@HenrikSte, Thank you for your interest in AutoConnect. It may not meet your needs, but I try explain.

Followings are quotes from the WiFiManager development branch as the [callback capability](https://github.com/tzapu/WiFiManager/blob/e0e0150f7815829b5b117ce4953fd7efeae8b358/WiFiManager.h#L192-L203) of WiFiManager (There are 6 callback features):
```cpp
void setAPCallback( std::function<void(WiFiManager*)> func );
void setWebServerCallback( std::function<void()> func );
void setConfigResetCallback( std::function<void()> func );
void setSaveConfigCallback( std::function<void()> func );
void setSaveParamsCallback( std::function<void()> func );
void setPreSaveConfigCallback( std::function<void()> func );
```
Of the above callbacks, One thing you need is probably the `setAPCallback`.  The `startUPCallback` will exit to the user function with A) the captive portal has been enabled by the `WiFiManager::AutoConnet`, or B) using the `WiFiManager::startConfigPortal` directly.
An exit that works close to this is the [onDetect](https://hieromon.github.io/AutoConnect/advancedusage.html#captive-portal-start-detection) exit routine in AutoConnect. When AutoConnect launches a captive portal, it calls a user function registered as the [**AutoConnect::onDetect**](https://hieromon.github.io/AutoConnect/api.html#ondetect) exit routine. It is also enabled if you start the captive portal directly with the [**AutoConenctConfig::immediateStart**](https://hieromon.github.io/AutoConnect/apiconfig.html#immediatestart) option.

AutoConnect will pass the [SoftAP IP address](https://hieromon.github.io/AutoConnect/api.html#ondetect) of the ESP module during Captive Portal as a parameter to the onDetect exit routine implemented by the user sketch. Also, in its exit routine, the WiFi.SSID function can know the SoftAP's SSID.
So I think that you can implement your primary requirements with a sketch like this:

```cpp
AutoConnect portal;
LiquidCrystal_I2C lcd(0x27, 20, 4);

// An exit to display the portal starting and identification
void diplayPortal(IPAddress softapip) {
   lcd.clear();
   lcd.setCursor(0, 0);
   lcd.printf("SoftAP:%s", softapip.toString().c_str()); 
   lcd.setCursor(0, 1);
   lcd.print("SSID:%s", WiFi.SSID().c_str());
}

void setup() {
  lcd.begin();
  lcd.backlight();
  portal.onDetect(displayPortal);  // Register onDetect exit routine.
  portal.begin();
}

void loop() {
  portal.handleClient();
}
```

HenrikSte, 2019-07-17:
@Hieromon 

Thank you so much for your fast, kind and extensive reply!
This works just perfect.

Please excuse that I have not found the section in the FAQ.

Cheers and keep up the great work!

Henrik

closed 2019-07-17 by HenrikSte

Hieromon, 2019-07-18:
@HenrikSte, You are welcome. Its tip was already mentioned in the "[Advanced Usage](https://hieromon.github.io/AutoConnect/advancedusage.html)" chapter of the documentation, but I revised it can also be referenced from the FAQ.

Please do not hesitate to post to an issue if you think it is better to improve.
Thank you for your contribution.