Hieromon/AutoConnect #113: Memory (PROGMEM) usage of EEPROM

OoM-JaN, 2019-07-25:
Great work here!

I have a question about the memory usage of Autoconnect. I've been building a program and added Autoconnect. Since I've done that I see that Autoconnect is EEPROM space where I store my data. Here is my code (partially):

 ```EEPROM.begin(512);
 EEPROM.put(0, variable);
 eeAddress += sizeof(int);
 EEPROM.put(eeAddress, variable);
 EEPROM.commit();
```

After I uses Autoconnect the data of my variable has been changed. Which memory does Autoconnect use or where can I best store my variable?

Hieromon, 2019-07-26:
I added a workaround for your problem to the FAQ of the documentation.
Quick measures to avoid the issue is below:
```cpp
AutoConnect portal;
AutoConnectConfig config;
void setup() {
  config.boundaryOffset = 64;  // Reserves the size required for EEPROM data.
  portal.config(config);
  portal.begin();
}
```
Please refer to the [FAQ](https://hieromon.github.io/AutoConnect/faq.html#data-saved-to-eeprom-is-different-from-my-sketch-wrote) of the documentation.

OoM-JaN, 2019-07-26:
So if I understand correctly, AutoConnect uses the EEPROM memory from 0 to 63. The memory from 64 and upwards is 'untouched'? Perhaps something to add in the FAQ?

Hieromon, 2019-07-26:
The boundaryOffset shifts the AutoConnect credential area. `boundaryOffset=64` means that AutoConnect uses it from the 64th byte. The area for user sketches is 0-63.

OoM-JaN, 2019-07-26:
0-63, clear, that's enough space for my project. Thanks!
But just out of curiosity, as beginner programmer, which space is used by AutoConnect? Or can I check which memory is used with some kind of function (check.lastused.memory.from.64-)?

Hieromon, 2019-07-26:
Have you read the reference of ESP8266 Arduino core? Its [documentation](https://arduino-esp8266.readthedocs.io/en/latest/filesystem.html) mentions the flash layouts that the ESP8266 Arduino core assigns. Incidentally, PROGMEM and EEPROM are different memory areas.
A guide to the size of binary sketch used by AutoConnect can be found in the [FAQ](https://hieromon.github.io/AutoConnect/faq.html#how-much-memory-does-autoconnect-consume).
Also, there is no upper limit on the size of the EEPROM area for credentials consumed by AutoConnect. See the chapter on [Saved credentials access](https://hieromon.github.io/AutoConnect/credit.html) of the documentation for the size of one entry of authentication information.

closed 2019-07-27 by OoM-JaN