Hieromon/AutoConnect #114: AutoConnectAux

OoM-JaN, 2019-07-26:
The more I understand of AutoConnect , the more I love it. Thanks for all the work!
I'm trying to implement the AutoConnectAux API but some examples would be appreciated to help me understand the code. 

I'n not sure if it's due to bad coding on my part or a bug but the ACText.caption isn't recognized by Arduino . See picture where the header text turns orange, the caption stays black:

[image]

Hieromon, 2019-07-26:
> the ACText.caption isn't recognized by Arduino.

Colorize in Arduino IDE does not IntelliSense. It is simply matched by a regular expression using the definition of keyword.txt. Is the string `header` present in the keyword.txt definition of another library?
So, did your sketch code give a compilation error? If there are no compilation errors, that is the correct syntax.

OoM-JaN, 2019-07-26:
I noticed my captioncolour was different from you example but no error was given during compiling.  It  isn't working yet but that is mainly because of my beginner programming experience :-)

OoM-JaN, 2019-07-27:
Finally got it to work. And I must say: I'm impressed. Works perfectly :-)
For those who are struggling as well:

In GLOBAL place the menu.JSON (here aux), the aux.handler and AutoConnectAux page;
```
static const char aux[] PROGMEM = R"(
    {
      "uri" : "/aux",
      "title" : "Aux.",
      "menu" : true,
      "element" : [
    {
      "name": "doortime",
      "type": "ACInput",
      "label": "Time to close"
    },
    {
      "name": "send",
      "type": "ACSubmit",
      "value": "SEND",
      "uri": "/settings"
    }
  ]
}
)";

String onPage(AutoConnectAux& aux, PageArgument& args) {
  AutoConnectSubmit& send = aux["send"].as<AutoConnectSubmit>();
  if (WiFi.isConnected())
    send.enable = (WiFi.getMode() == WIFI_STA);
  else
    send.enable = false;
  return String();
}

AutoConnectAux    page;
```

In void.setup () place the following code before portal.begin:
```
page.load(aux);
page.on(onPage);
portal.join(page);
```

closed 2019-07-27 by OoM-JaN