Hieromon/AutoConnect #115: AutoConnectConfig API parameters

OoM-JaN, 2019-07-26:
Reading thru the manual I found a minor spelling error:
AutoConenct has to be AutoConnect.
[image]

It is still not clear how to use this api. I've tried the next code but it keeps failing:
```
AutoConnectConfig config("myweb","");
config.staip = IPAddress(192,168,4,1);
Portal.config(Config);
```
`'Config' does not name a type`
 what am I doing wrong?

Hieromon, 2019-07-26:
@OoM-JaN, 
> AutoConenct has to be AutoConnect.

Thank you notify. It had been fixed.

Hieromon, 2019-07-26:
Use the [AutoConnectConfig::apip](https://hieromon.github.io/AutoConnect/apiconfig.html#apip) to change the IP address of the ESP module that is in Soft AP state when launching the captive portal.

OoM-JaN, 2019-07-26:
My stupid mistake! With apip it worked instantly. Thanks!

closed 2019-07-27 by OoM-JaN