Hieromon/AutoConnect #116: Putting variables to Custom Web pages with JSON in AutoConnectAux

OoM-JaN, 2019-07-27:
I've been reading the library pages over and over, tried a lot of different steps and although I have it almost working, there is one thing I can't successfully implement; putting a variable into a custom webpage.

I managed to add the menu item, wrote the Handler for it and retrieve the entered variable.
I read the variable in the Handler with:
myvariable = server.arg("doortime").toInt();

But how do I put myvariable into the input field?
I think it needs to be done in the section between the ***-es but once i change anything with myvariable it fails.

Here is my code (only the relevant sections)
```
static const char aux[] PROGMEM = R"(
    {
      "uri" : "/aux",
      "title" : "SETTINGS",
      "menu" : true,
      "element" : [
    {
      "name": "doortime",
      "type": "ACInput",
      "label": "Time to close [s] ",
      "value": ""
    },
    {
      "name": "send",
      "type": "ACSubmit",
      "value": "SEND",
      "uri": "/settings"
    }
  ]
}
)";

AutoConnectAux    page;

String onPage(AutoConnectAux& aux, PageArgument& args) {
  AutoConnectSubmit& send = aux["send"].as<AutoConnectSubmit>();
  if (WiFi.isConnected())
    send.enable = (WiFi.getMode() == WIFI_STA);
  else
    send.enable = false;
  return String();
}

/************************************************************************************************
portal.load(aux);
    portal.on("/aux", [](AutoConnectAux& aux, PageArgument& args) {
      AutoConnectText& ac_echo = aux.getElement<AutoConnectText>("doortime");
    ac_echo.value = args.arg("doortime");
  return String();  
  });
************************************************************************************************/

  page.load(aux);
  page.on(onPage);
  portal.join(page);
  
  if (portal.begin()) {
      Serial.println("Started, IP: " + WiFi.localIP().toString());  
  }
  else{
    Serial.println("Connection failed.");
    while (true) { yield(); }
  }

```

Hieromon, 2019-07-28:
You defined the element named 'doortime' is AutoConnectInput with JSON. But the '/ aux' on-handler accesses it as AutoConnectText.
The getElement casts from the AutoConnectElement type which is a variant of all elements, to the actual declared type. Conversion to other types is not possible.

OoM-JaN, 2019-07-28:
So I got it wrong. But how to fix it?

Hieromon, 2019-07-28:
The primary fix is ​​to change `AutoConnectText& ac_echo = aux.getElement<AutoConnectText>("doortime");` to `AutoConnectInput& ac_echo = aux.getElement<AutoConnectInput>("doortime");`.
But the '/ aux' handler is taken out of doortime and assigned to doortime, so it seems to be meaningless.
What do you want to do with the doortime value entered in a custom web page?

OoM-JaN, 2019-07-28:
I'm programming a wifi-enabled chicken coop door. I tried to store setting inside the AutoConnect menu but I think it's simpler to just add another page & handler. The AutoConnectAux is just too complex to use for me.

closed 2019-07-28 by OoM-JaN

Hieromon, 2019-07-28:
AutoConnectElements itself maintains the values ​​entered in the custom web page. You do not need a special code to save it.