Hieromon/AutoConnect #120: how to force to forget saved credential?

jose365sc, 2019-08-15:
Is there any API or other means to force to forget saved WiFi connection params?

I'd like to equip a feature to forget everything, or factory-reset all saved configuration including wifi configs, for user to do configuration again from scratch for trouble shooting purposes.  Having "Reset only wifi settings" feature would be also better. I'm wondering if there's any API or means to erase or reset saved AutoConnect Wifi settings, which can be called on user's button click or such action.

If I understand correctly, AutoConnect saves its permanent data by its own, other than ESP-sdk's wifi configs. Only ESP.eraseConfig() may not be enough, I assume.

Any advice would be appreciated!

Hieromon, 2019-08-18:
I'm not sure what the factory-reset mean, but you can erase the saved credentials as  follows:

1. To erase the credentials AutoConnect saved

    You can use the [AutoConnect API](https://hieromon.github.io/AutoConnect/credit.html) for accessing the saved credentials in EEPROM. It can read/write and erase credentials. In particular, use the [del](https://hieromon.github.io/AutoConnect/credit.html#del) to erase a credential.
    An example using this API is [**Credential.ino**](https://github.com/Hieromon/AutoConnect/blob/master/examples/Credential/Credential.ino) that lists and deletes credentials with the Web interface.

2. Also, to erase the WiFi connection memorized by SDK

    You can use ESP.eraseConfig, but I don't think it fits your purpose. ESP.eraseConfig will erase the flash sectors used individually by the SDK. As a result, the data other than saved credentials will be deleted too. And ESP32 arduino core does not have ESP.eraseConfig interface.
    **WiFi.disconnect** can be used instead to clear the connection information that the arduino core memorized using esp_set_config.
    ```cpp
    // For ES8266
    WiFi.persistent(false);
    WiFi.disconnect(true);
    WiFi.persistent(true);
    WiFi.disconnect(true);

    // For ESP32
    WiFi.disconnect(true, true);
   ```

jose365sc, 2019-08-18:
Thanks for your comment!
I assume AutoConnect API you mentioned is AutoConnectCredential::del(const char* ssid).
it needs to enumerate saved SSID beforehand and there's no API to delete all at once.

I'll try WiFi.disconnect() following your advice!

Hieromon, 2019-08-18:
> there's no API to delete all at once.

It designed on my intention. Clearing all credentials is a separate need, which can be implemented individually in a sketch with code like this:

```cpp
void deleteAllCredentials(void) {
  AutoConnectCredential credential;
  struct station_config config;
  uint8_t ent = credential.entries();

  while (ent--) {
    credential.load((int8_t)0, &config);
    credential.del((const char*)&config.ssid[0]);
  }
}
```
This topic closed due to present how. Reopen if necessary.
Thank you.

closed 2019-08-18 by Hieromon