Hieromon/AutoConnect #121: AutoConnect Portal is not opening.

krutarthtrivedi, 2019-08-21:
Hello Hieromon,
Greetings!

Board: ESP32 Development Board
IDE: Arduino

I am using Autoconnect library with STA static IP assignment configuration. Once ESP gets connected with my router, I am able to open Autoconnect portal using "staip" provided by me in configuration

. But if router is reset or power off and then getting power on, the Autoconnect portal is not getting open using "staip". Kindly look into this. Thanks.

Following is my configuration:
```
IPAddress local_IP(192, xxx, xx, 100);
IPAddress gateway(192, xxx, xx, 1);
IPAddress subnet(255, xxx, xxx, 0);
IPAddress DNS1(8, 8, 8, 8);
IPAddress DNS2(8, 8, 4, 4);

  Config.boundaryOffset = CREDENTIAL_OFFSET;
  Config.autoReconnect = true;
  Config.apid = "XYZ";
  Config.psk = "9876543210";
  Config.staip = local_IP;
  Config.staGateway = gateway;
  Config.staNetmask = subnet;
  Config.dns1 = DNS1;
  Config.dns2 = DNS2;
  Portal.config(Config);
```

krutarthtrivedi, 2019-08-21:
> Hello Hieromon,
> Greetings!
>
> Board: ESP32 Development Board
> IDE: Arduino
>
> I am using Autoconnect library with STA static IP assignment configuration. Once ESP gets connected with my router, I am able [...]

[image]

Hieromon, 2019-08-29:
Sorry for reply late. I prioritize support for arduino-esp32 1.0.3. So, I have not yet tried to reproduce posted issue.
It doesn't seem to be an AutoConnect issue, but could you reproduce the issue with the Core Debug level set to "Debug"?
Please post the debug message again.

krutarthtrivedi, 2019-08-31:
> Sorry for reply late. I prioritize support for arduino-esp32 1.0.3. So, I have not yet tried to reproduce posted issue.
> It doesn't seem to be an AutoConnect issue, but could you reproduce the issue with the Core Debug level set to "Debug"?
> Please post the debug message again.

krutarthtrivedi, 2019-09-05:
Hello Hieromon,

Sorry for the late reply. Followings are the responses as per asked.

1. On startup, ESP32 searched for WiFi and gets connected with the WiFi router using static credentials provided through AutoConnect config.

[image]

2. After connection has been established, I opened ESP32 IP address inside a web browser and the Portal gets opened.

[image]
 
3. Now I switched off the router and ESP32 shows the following output.

[image]

4. ESP32 keeps searching for WiFi.

[image]

5. Now I turned on the router and after some time, ESP32 gets automatically connected to the same WiFi router.

[image]

6. Once ESP32 has been connected with WiFi router, I tried to reopen the ESP32 IP Address inside a web browser and it shows the following error.

[image]

Kindly look into this. 

Thanks.
Krutarth Trivedi.

Hieromon, 2019-09-06:
@krutarthtrivedi Thank you for your reports. I cannot reproduce the issue yet. Here is my test log:
```
12:12:38.822 -> [W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 200 - BEACON_TIMEOUT
12:12:38.822 -> [D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
12:12:38.856 -> [D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
12:12:38.856 -> [D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
12:12:40.905 -> [D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
12:12:40.905 -> [W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 201 - NO_AP_FOUND
12:12:40.905 -> [D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
... SAME AS ABOVE
13:00:42.076 -> [D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
13:00:42.076 -> [D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
13:00:44.116 -> [D][WiFiGeneric.cpp:337] _eventCallback(): Event: 5 - STA_DISCONNECTED
13:00:44.116 -> [W][WiFiGeneric.cpp:353] _eventCallback(): Reason: 201 - NO_AP_FOUND
13:00:44.116 -> [D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
13:00:44.150 -> [D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
13:00:44.150 -> [D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
13:00:45.137 -> [D][WiFiGeneric.cpp:337] _eventCallback(): Event: 7 - STA_GOT_IP
13:00:45.137 -> [D][WiFiGeneric.cpp:381] _eventCallback(): STA IP: 172.20.10.4, MASK: 255.255.255.240, GW: 172.20.10.1
13:00:45.137 -> [D][WiFiGeneric.cpp:337] _eventCallback(): Event: 7 - STA_GOT_IP
13:00:45.171 -> [D][WiFiGeneric.cpp:381] _eventCallback(): STA IP: 172.20.10.4, MASK: 255.255.255.240, GW: 172.20.10.1
13:01:13.950 -> [AC] Host:172.20.10.4,URI:/_ac,already allocated
13:01:13.950 -> [AC] Host:172.20.10.4,URI:/_ac,already allocated
13:01:13.983 -> [D][WiFiClient.cpp:509] connected(): Disconnected: RES: 0, ERR: 128
```
As you can see from the log, TCP connection recovered automatically even if WIFi disconnected about 50 minutes.
Your case seems similar to the [esp32 issue](https://github.com/espressif/arduino-esp32/issues/1464) that remains unresolved. The issue implies that once the WiFi data link is recovered after the router restarts, but the TCP socket once closed will not reopen.
However, I encountered a successful case during a reproduction test of this problem when I tested a cell phone with desalted as a router.

After restarting the router, does the esp32 respond when a ping is thrown after WiFi reconnected? If the esp32 module does not respond to the ping, TCP remains closed. It is also necessary to reconnect the client browser to the router after restarting.

If we may not clear the cause, I will consider improving the AutoConnect library as a workaround.  The phenomenon of not automatically recovering after disconnected WiFi can be fatal to IoT devices.
Also, the workaround will look like this:

- AutoConnect::handleClient function improvement
    If IP is assigned again after STA_DISCONNECTED is detected, AutoConnect will execute WiFi.begin again if the elapsed time of that period is 5 to 10 seconds or more.

krutarthtrivedi, 2019-09-07:
@Hieromon Thanks for the workaround. Currently, I am using 0.9.11 release. I'll check the same using the latest release today and will share the results. Meanwhile, Can you please send me the arduino program, using which you performed the similar test. It would be beneficial for me to understand the problem better and I can produce some test result, if the same issue still arises. 

Thanks,
Krutarth Trivedi.

Hieromon, 2019-09-07:
@krutarthtrivedi 
> Can you please send me the arduino program, using which you performed the similar test.

It is here and is one of the example sketches included in upcoming v1.0.0 very soon.
https://github.com/Hieromon/AutoConnect/blob/Enhance/v099/examples/ConfigIP/ConfigIP.ino

So, Have you tried ping after reconnecting?

Hieromon, 2019-09-09:
@krutarthtrivedi I released v1.0.0 which has amended [Config.ino](https://github.com/Hieromon/AutoConnect/blob/master/examples/ConfigIP/ConfigIP.ino). Please, you use this one with AutoConnect v1.0.0 for the reproduction test.
In ESP32 with AutoConnect v1.0.0, Saved credentials area has moved from EEPROM to Preferences. So, past saved credentials will be unreached. You will need to reconnect again using a captive portal at least once.

krutarthtrivedi, 2019-09-09:
> @krutarthtrivedi
>
>
> It is here and is one of the example sketches included in upcoming v1.0.0 very soon.
> https://github.com/Hieromon/AutoConnect/blob/Enhance/v099/examples/ConfigIP/ConfigIP.ino
>
> So, [...]

As instructed, I have started testing the same using v1.0.0 and ESP32 1.0.2 SDK. I am unaware of how to use ping command. Please guide me through this. So, I can try to give a ping from ESP32.

Besides, I tried pinging using cmd command after router reboot and this is what i got. First Ping request is of before router reboot and the preceding one is of after router reboot.

[image]

Thanks,
Krutarth Trivedi.

Hieromon, 2019-09-09:
@krutarthtrivedi It has become clear what is happening. TCP remains closed after the WiFi data link has established again after the router restarts.
Can you try something more test before I implement the workaround? because I cannot reproduce the issue.
Please try as following procedure:
- The sketch to use is Config.ino
- Insert the following sentence immediately after `portal.begin()` inside `setup()`.
  ```cpp
  WiFi.setAutoReconnect(true);
  ```
- Test router restart after connection established

Even with the above patch applied, if the same problem occurs, you need to implement a more complex restore sequence in the event handler. It is a workaround that I may implement by AutoConnect.

krutarthtrivedi, 2019-09-11:
@Hieromon I tried as per instructed, no luck!!! Tried with different routers as well but nothing worked for me.
I can't even see any ping event taking place for me! 

This is the Debug log I got. I found none changes in this using v1.0.0. WiFi is showing WiFi_Connected() event but actual TCP connection is not being established.

> [W][WiFiGeneric.cpp:351] _eventCallback(): Reason: 201 - NO_AP_FOUND
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:379] _eventCallback(): STA IP: 192.168.31.100, MASK: 255.255.255.0, GW: 192.168.31.1
[D][WiFiGeneric.cpp:336] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:379] _eventCallback(): STA IP: 192.168.31.100, MASK: 255.255.255.0, GW: 192.168.31.1

Please look into this matter as I am in the middle of the development phase and got stuck in this.

Thanks,
Krutarth Trivedi.

krutarthtrivedi, 2019-09-11:
@Hieromon I tried HTTP GET request to google.com, just to replicate ping feature.  Here's the code.

```
void setup() {
  delay(1000);
  Serial.begin(115200);
  Serial.println();
  AutoConnectConfig Config;
  
  Config.boundaryOffset = CREDENTIAL_OFFSET;

  Config.hostName = "XXX";
  Config.staip = local_IP;
  Config.staGateway = gateway;
  Config.staNetmask = subnet;
  Config.dns1 = DNS1;
  Config.dns2 = DNS2;
  
  Portal.config(Config);

  if (Portal.begin()) {
    Serial.println("WiFi connected: " + WiFi.localIP().toString());
    delay(3000);
  }
}

void testClient(const char * host, uint16_t port)
{
  Serial.print("\nconnecting to ");
  Serial.println(host);

  WiFiClient client;
  if (!client.connect(host, port)) {
    Serial.println("connection failed");
    return;
  }
  client.printf("GET / HTTP/1.1\r\nHost: %s\r\n\r\n", host);
  while (client.connected() && !client.available());
  while (client.available()) {
    Serial.write(client.read());
  }

  Serial.println("closing connection\n");
  client.stop();
}

void loop() {
  Portal.handleClient();

  wifi_status = WiFi.status();

  if(wifi_status == WL_CONNECTED)
  {
    testClient("google.com", 80);
  }

  else
  {
    Serial.println("No WiFi connection");
  }
  delay(5000);
}
```

On startup, I got connection established to "google.com" and got the GET response. Besides, ping to ESP32 from computer connected in the same network works fine. Here's the screenshot of the same.

[image]

After router reset, ESP32 is automatically connected with Router but couldn't establish connection with "google.com" and presents the error shown in the following figure. 

[image]

Everything works good after resetting ESP32. So, I think this is something related to TCP/IP connection reestablishment or socket reopen but I am not sure at this point of time.

Kindly look into this.

Thanks,
Krutarth Trivedi.

krutarthtrivedi, 2019-09-13:
@Hieromon 

ESP32 Repository: 1.0.3-rc3
The suggested changes for WiFi.disconnect()  has been performed.

**Test sketch with reconnection** is working as expected like after re-establishment of WiFi connection, first GET request is showing ERROR but after that connection is being re-established and GET request is working good. 

**Test sketch without reconnection** is still showing the same socket error.

Now, What's your view on this?

Thanks,
Krutarth Trivedi.

Hieromon, 2019-09-13:
@krutarthtrivedi Thank you for the result report. It's kind of deeply rooted and tricky.
It seems to me that the root of the causes is in the lwIP, not the arduino core. Anyway, I will post an issue to the arduino-esp32 core repository. We may get some hints from the contributors.

krutarthtrivedi, 2019-09-16:
@Hieromon  Thanks a lot for your support! Let's wait for the contributors to reply. So, I can plan accordingly for my ongoing development.

Thanks,
Krutarth Trivedi.

Hieromon, 2019-09-16:
You can work around this problem in your development project diverting "Test sketch with reconnection" as above like as:
```cpp
#include <WiFi.h>
#include <WebServer.h>
#include <AutoConnect.h>

IPAddress local_IP(xxx, xxxx, xxx, xxx);
IPAddress gateway(xxx, xxxx, xxx, xxx);
IPAddress subnet(xxx, xxxx, xxx, xxx);
IPAddress DNS1(xxx, xxxx, xxx, xxx);
IPAddress DNS2(xxx, xxxx, xxx, xxx);

wifi_event_id_t eventDiscon;
wifi_event_id_t eventGotIP;
wifi_event_id_t eventRescon;
bool staDiscon;
bool staGotIP;
bool rip;
unsigned long tmDiscon;
unsigned long tmPeriod;

AutoConnect portal;
AutoConnectConfig config;

void cbDiscon(WiFiEvent_t e, WiFiEventInfo_t info) {
  Serial.println("[E] SYSTEM_EVENT_STA_DISCONNECTED");
  staDiscon = true;
  tmDiscon = millis();
}

void cbGotIP(WiFiEvent_t e, WiFiEventInfo_t info) {
  Serial.println("[E] SYSTEM_EVENT_STA_GOT_IP");
  staGotIP = true;
}

void onDiscon(void) {
  staDiscon = false;
  staGotIP = false;
  rip = false;
  eventDiscon = WiFi.onEvent(cbDiscon, SYSTEM_EVENT_STA_DISCONNECTED);
  eventGotIP = WiFi.onEvent(cbGotIP, SYSTEM_EVENT_STA_GOT_IP);
}

void removeEvent(void) {
  staDiscon = false;
  staGotIP = false;
  WiFi.removeEvent(eventDiscon);
  WiFi.removeEvent(eventGotIP);
}

void cbRescon(WiFiEvent_t e, WiFiEventInfo_t info) {
  Serial.println("[E] Connection restored");
  WiFi.removeEvent(eventRescon);
  onDiscon();
}

void restoreIP(unsigned long tmWhile) {
  if (!rip) {
    if (staDiscon & staGotIP) {
      if (millis() - tmDiscon > tmWhile) {
        rip = true;
        removeEvent();
        WiFi.reconnect();
        eventRescon = WiFi.onEvent(cbRescon, SYSTEM_EVENT_STA_GOT_IP);
      }
    }
  }
}

void setup() {
  delay(1000);
  Serial.begin(115200);
  Serial.println();

  config.staip = local_IP;
  config.staGateway = gateway;
  config.staNetmask = subnet;
  config.dns1 = DNS1;
  config.dns2 = DNS2;

  portal.config(config);
  if (portal.begin()) {
    Serial.println("connected:" + WiFi.localIP().toString());
    onDiscon();
  }
  delay(3000);
  tmPeriod = millis();
}

void testClient(const char * host, uint16_t port) {
  Serial.print("\nconnecting to ");
  Serial.println(host);

  WiFiClient client;
  if (!client.connect(host, port)) {
    Serial.println("connection failed");
    return;
  }
  client.printf("GET / HTTP/1.1\r\nHost: %s\r\n\r\n", host);
  while (client.connected() && !client.available());
  while (client.available()) {
    Serial.write(client.read());
  }

  Serial.println("closing connection\n");
  client.stop();
}

void loop() {
  if (millis() - tmPeriod > 5000) {
    tmPeriod = millis();
    wl_status_t wifi_status = WiFi.status();
    if (wifi_status == WL_CONNECTED)
      testClient("google.com", 80);
    else
      Serial.println("No WiFi connection");
  }
  restoreIP(5000);
  portal.handleClient();
}
```
However, this code is risky a little. Even if the arduino-core reconnect completes successfully, the reconnect attempt is performed twice. This is a detrimental effect of returning WL_CONNECTED with  WiFi.status even if TCP is still closed.

krutarthtrivedi, 2019-09-16:
@Hieromon Okay, I will workaround on this and will let you know if I found any solution for this. Meanwhile, Let's hear from the contributors first.

Thanks,
Krutarth Trivedi.

Hieromon, 2019-12-31:
Close once due to no activity record for one month.
Reopen when new reports are posted.

closed 2019-12-31 by Hieromon