Hieromon/AutoConnect #127: Incompatible with ESP8266httpUpdate.h

Javinator9889, 2019-09-14:
After upgrading to version `v1.0.1` from `v0.9.12`, while using an OTA server, the library `#include <ESP8266httpUpdate.h>` appears to be missing either overwritten by AutoConnect - here you have the compilation error I got:
```
main:1239:5: error: 'ESPhttpUpdate' was not declared in this scope
     ESPhttpUpdate.setLedPin(LED_PIN, HIGH);
     ^
exit status 1
'ESPhttpUpdate' was not declared in this scope
```
I am forced to downgrade to `v0.9.12` until there is a workaround or it has been fixed.

Thanks for your awesome job

Hieromon, 2019-09-15:
As you pointed out, if you use ESP8266httpupdateh in a user sketch, coexistence with AutoConnectUpdate may not be fully considered.  It may occur in the order of include directives.
AutoConnectUpdate.h defines **NO_GLOBAL_HTTPUPDATE** internally, so if your sketch include ESP8266HttpUpdate.h after AutoConnect.h, the ESPhttpUpdate global variable will be lost.

v1.0.1 has the following two workarounds.
1. Disable it if you don't need AutoConnectUodate.
https://github.com/Hieromon/AutoConnect/blob/3938f5af648cb9cbcd510354c96f000989062dcc/src/AutoConnectDefs.h#L33
    Committed out the above to disable AutoConnectUpdate.

2. Include AutoConnect.h after ESP8266httpUpdate.h

Besides, I study modify it so that it can compile correctly without depending on the order of include directives and will release a patch for this.
Thank you for your contribution.

Javinator9889, 2019-09-15:
Hi
I have tested some possible solutions, such as `undef` the `AUTOCONNECT_USE_UPDATE` macro:
```c++
#if OTA_ENABLED
#ifdef AUTOCONNECT_USE_UPDATE
#undef AUTOCONNECT_USE_UPDATE
#endif
#include <ESP8266httpUpdate.h>
#endif
```
But this does not work, so the only working workaround I found is just adding the `AutoConnect` library at the end of the `#include` section:
```c++
#if OTA_ENABLED
#ifdef AUTOCONNECT_USE_UPDATE
#undef AUTOCONNECT_USE_UPDATE
#endif
#include <ESP8266httpUpdate.h>
#endif

// AutoConnect libraries - must be defined here for OTA compatibility
#include <AutoConnect.h>
#include <AutoConnectCredential.h>
#include <AutoConnectAux.h>
```
Thanks for your support - I think this issue can be closed

Hieromon, 2019-09-16:
@Javinator9889 Could you describe your solution? Do you assume that a series of macro statements beginning with **OTA_ENABLED** as above will be coded by a user sketch?
If so, I think it's enough to simply write `#include <AutoConnect.h>` after `#include <ESP8266httpUpdate.h>`. What is your point of view?

AutoConnectUpdate uses a strategy that minimizing global variables defined by esp8266/esp32 core libraries as possible to save memory. That's why I define NO_GLOBAL_HTTPUPDATE in AutoConnectUpdate.h, but it doesn't work if user sketch includes AutoConnect.h before ESP8266httpUpdate.h. (ESP8266httpUpdate.h is evaluated only once per compilation)
Therefore, if AutoConnect.h is included before ESP8266httpUpdate.h, Compiler cannot export ESPhttpUpdate at the preprocessor execution timing.
I was initially thinking to avoid this problem by using preprocessor instructions, but for now, I think there is no other way to avoid this problem with the order of the #include directives.
Do you have any workarounds that should be implemented on the AutoConnect library side?

Javinator9889, 2019-09-16:
Hi @Hieromon 
I finally managed to get a working workaround **saving sketch size also** (about _~1%_ in ESP8266).
I will explain what I have:

1. A **macro** `#define OTA_ENABLED 1` for saving space (sometimes I do not want sketch to be updated via OTA - e.g.: I am doing tests).
2. If I enabled OTA, then include in the sketch `ESP8266httpUpdate.h` lib.
3. Enable also functions that uses OTA.

Now, with the latest version of `AutoConnect`, this behaviour is not working as there is a new update feature. So, the workaround I found is:
```c++
/* ALL INCLUDE HEADERS THAT DO NOT USE THE OTA FUNCTIONALITY
--------------------------------------------------------------
*/
// It is necessary to include the AutoConnecDefs.h here
#include <AutoConnectDefs.h>

// Define whether will the OTA be enabled or not
#define OTA_ENABLED 1 // can be 'true' also

// If the OTA has been enabled, we #undef the AUTOCONNECT_USE_UPDATE macro
#if OTA_ENABLED
    #ifdef AUTOCONNECT_USE_UPDATE
        // That is why the AutoConnectDefs library should be included before
        // so it is possible to undefine the MACRO.
        #undef AUTOCONNECT_USE_UPDATE
    #endif 
#endif 

// Now, it is safe to include the AutoConnect library
// even before including the ESP8266httpUpdate one:
#include <AutoConnect.h>
#include <AutoConnectCredential.h>
#include <AutoConnectAux.h>

// Including the ESP8266httpUpdate here is safe as the ESPhttpUpdate
// is still available
#if OTA_ENABLED
    #include <ESP8266httpUpdate.h>
#endif 
/*
void setup(void) {
    SETUP CODE
}

void loop(void) {
    LOOP CODE
}

+-----------------------------+
| Other functions & utilities |
+-----------------------------+

*/

#if OTA_ENABLED
void otaUpdate(void) {
    if (WiFi.status() == WL_CONNECTED) {
        t_httpUpdate_return returnValue = ESPhttpUpdate.update(client, OTA_URL, VERSION);
        // There is no error such that "ESPhttpUpdate was not declared in this scope"
    }
}
#endif 
```
The full code I am using can be found here: https://s.javinator9889.com/FVHWyT

Hieromon, 2019-09-17:
@Javinator9889 Thank you for comments.
I agree with your approach but may not be able to maintain backward compatibility of AutoConnect header file structure. Please post the issue if it affects your approach in the future. I will correspond.

PR #129 (2019-09-17): Patches/v102
Fixes #125
Fixes #127
Fixes #128
closed 2019-09-17 by Hieromon