Hieromon/AutoConnect #128: Passphrase shown while trying to connect to ap with ssid of 32 characters length

jstammi, 2019-09-16:
The existing wlan's ssid is of 32 chars length, the max allowed AFAIK.
It shows up in the menu from "Configure new AP".
After entering the passphrase and pressing the "Apply" button, the ssid directly followed by the passphrase is shown on the progress page while trying to connect. Obviously AutoConnect expects a 0-terminated ssid. But such is not given with ssid of max length possible.

In the serial concole a related message appears: "[E][WiFiSTA.cpp:124] begin(): SSID too long or missing". IMHO this is wrong, too, as AFAIK spec allows the ssid to be of 32 characters length. But this is not AutoConnect's fault, I will have to raise another issue for arduino-esp32.

jstammi, 2019-09-16:
https://github.com/espressif/arduino-esp32/issues/3218

Hieromon, 2019-09-17:
@jstammi Yours pointed out is valid. I will release a fix by v1.0.2.
Thank you for your contribution.

PR #129 (2019-09-17): Patches/v102
Fixes #125
Fixes #127
Fixes #128
closed 2019-09-17 by Hieromon