Hieromon/AutoConnect #132: Static IP on AC Menu + Password change

Andreadj, 2019-09-26:
Hi Hieromnon,
first let me tell you once again: Great work!!! I will bring you a beer!
Simple question: It should be possible put an AC element check to enable/disable DHCP with settings for Static IP on the "Configure new AP" menu? I have a lot of places without dhcp and I don't want everytime to connect to wemos to change Static ip settings...
Last one: How should I change APpsk from a page on the captive portal? I tried different things but nothing seems to work...
Sorry if simple question... I'm a newbie...

Andreadj, 2019-09-27:
Ok, password problem solved.

Hieromon, 2019-09-30:
It is currently not possible to configure a static IP with the 'Configure new AP' menu. However, you can create a sketch that allows each SSID links to the static IP in advance concerning the **ConfigIP.ino** example.

Andreadj, 2019-09-30:
Ok, thanks. Actually I can't place an external switch, I will look for other solution. It should be a nice implementation... think about it...

Hieromon, 2019-09-30:
@Andreadj 
> I can't place an external switch

If you are expecting something, I can consider it.
Do you have requirements which something to be a trigger as an alternative to an external switch?

Andreadj, 2019-09-30:
@Hieromon 
thanks for the availability but I don't want to waste your time, that's not a priority. I'm assembling a 20W RGBW Wi-Fi Battery Led light for architectural purpose drived with Artnet or sACN from App (Aurora DMX, others), physical Consolle or PC (Qlight, Freestyler, others).  Artnet is quite an old protocol and different old hardware use static ip. It was only to support old hardware... Unfortunately the light is as small as possible and I don't have a mm left for other switches or anything. The only way is through software configuration. So don't worry. 
There's still a lot of work to do... in case I'll ask you! Thanks

Andreadj, 2019-09-30:
I appreciate it!!! ;-)

Hieromon, 2019-10-01:
For devices based on UDP/IP derived protocols such as Art-Net, it is not so unusual to assign IP addresses on the device side. I agree with your request.
In the next enhance turn, the Config New menu will support the enable/disable DHCP client and the storing of static IP addresses to the credentials.

[image]

Andreadj, 2019-10-01:
Wow that's an Awesome news!!! Please let me know I can I offer you a beer or a contribute to your work!

Hieromon, 2019-10-08:
@Andreadj Can I have your help instead of a beer?
I staged an enhanced AutoConnect v1.1.0 to the development branch as [Enhance/v110](https://github.com/Hieromon/AutoConnect/tree/Enhance/v110). It can configure static IP using Config New menu. I need to improve the quality before releasing the v1.1.0.
Please test it if you have time and let me know the occurred issues and your suggestion.

v1.1.0 has features and limitations as follows:
- You can configure static IPs in the **Configure new AP** menu. Uncheck **Enable DHCP** in the menu and specify static IPs. However, it does not check the syntax of the entered IP address. If the entered IP addresses are wrong, the WiFi connection cannot be established.

- AutoConnect v1.1.0 will save the Static IP configuration along with its SSID credential. When you select a saved SSID in the **Open SSIDs** menu, it will attempt to connect with the saved static IP configuration. If its SSID is DHCP configured, AutoConnect will act as a DHCP client.

- It was lost AutoConnectCredential backward compatibility. Credentials saved by AutoConnect v1.0.3 (or earlier) will not work properly with AutoConnect v1.1.0. You need to erase the flash of the ESP module using the esptool before the test sketch uploading.  
    ```
    esptool -c esp8266 (or esp32) -p [COM_PORT] erase_flash
    ```
- If you try to connect to another AP that is connected currently, you will need to disconnect wifi once and relaunch the captive portal. **Configure new AP** menu can be called even if the module already connected in STA mode, but the browser will display an error when the new connection established also the web session will lose the connection. (because the IP address changes dynamically during the http session)

- I tested on ESP32 core 1.0.3 and ESP8266 core 2.5.2.

Maybe bugs remain still. Should be continuing Search-and-Destroy tactics.

Andreadj, 2019-10-09:
Hi Hieromon,

I’m starting trying Enhance/v110 with my sketch (that is still a work in progress).  Actually I’m not even able to visualize the Menu “Configure new AP” page.

I can see the “Configure new AP” page 1 time each 10 times pressed. Sometimes everything disappear… it seems very unstable…

It’s a bug or there’s a problem with my sketch?

Attached if you want to check…

I will try with examples sketches…

 

Andrea Croce

Hieromon, 2019-10-10:
@Andreadj Thank you for a testing report.
To tell the truth, a similar phenomenon occurs in my environment. Probably a bug, but I cannot yet identify the situation where the problem occurs.
I initially thought it would occur if the access point WIFI signal was weak. However, if a similar phenomenon occurs in your environment, the causes may be different.
Please provide me the conditions as follows when you had encountered this phenomenon:
- A type of the esp module. ESP8266 or ESP32
- Arduino core version
- LwIP variant if you use ESP8266 (You can see it in Arduino IDE's tool menu)
- CPU Frequency
- Saved credentials are empty, or its AP has established once. (its credential stored in the flash already)
- DHCP or Static IP, or Both
- It occurs at the beginning immediately after erasing flash, or it occurs in Config new AP after leaving some AP.

If possible, upload your sketch to Gist in the secret mode and let me know the URL. I will test it.

Hieromon, 2019-10-10:
@Andreadj Thank you for providing the sketch. I reproduced the issue using your sketch.
I found the direct cause. It insufficient heap. Due to the addition of the input boxes for static IP configuration, insufficient the string buffer to generate HTML.
I will try somehow but don't expect it. If I fail, I'll treat you to a beer.

Andreadj, 2019-10-10:
Ok, so in general I have to be careful about how much I load on the HTML page or is it just a problem with the "Configure new AP" page? I mean it is the sum of the buffer used for all the pages or only the one that must display? Or maybe string buffer is something else?

Hieromon, 2019-10-11:
@Andreadj 
> so in general I have to be careful about how much I load on the HTML page or is it just a problem with the "Configure new AP" page?

Both. The string buffer is actually a heap area. At least, it temporarily needs unfragmented continuous heap space of the double size of the HTML length that should be generated. With the v110 enhancement, the HTML size of **"Config new AP"** has increased beyond 10KB. (In other words, 20KB of continuous free heap space is required)

1. PageBuilder's structural issues. (Not suitable for long HTML generation)
AutoConnect entrusts HTML generation to [PageBuilder](https://github.com/Hieromon/PageBuilder) that makes heavy use of String::concatenation and causes memory fragmentation. By adding an IP address input box to the **"Config new AP"**, heap fragmentation occurred more than v1.0.3, and "temporarily continuous free space in the heap" required for HTML generation was lost.  
So, I have modified the Enhance/v110 branch today to prevent fragmentation as much as possible, but it is not enough for your sketch. You can try a modified Enhance/v110 branch again.

2. Your sketch uses a lot of global variables.
Which is one of the factors that squeeze the heap. In particular, DynamicJsonBuffer as **startUpPage** variable is allocated globally as 5KB, but this buffer can be moved into the **startUpConfig** function. It also saves DevName, DMXAddress and more as your parameters to saveData.json, so you will end up using ArduinoJson. Therefore, Your sketch natively defines AutoConnectElements globally, but reading from a JSON document is more memory efficient. (Defining AutoConnectElements natively in a sketch is only effective for reducing memory consumption when a sketch does not use JSON at all)

The screenshot below shows the free memory size as a result of moving the **startUpPage** Json buffer to the **startUpConfig** function. Its size is the total size of the free space pieces, not the continuous free space size. Moreover, LXDMXWiFi library is not included.

[image]

I think you will make a completed sketch, the heap will probably run out. So, I will try to modify your sketch to work. A huge typhoon will hit Japan this weekend. I cannot be going out anyway so I have time for this work.

Andreadj, 2019-10-11:
Hieromon all this is really too kind of you so let me at least offer you, not a beer, but a case of beers!!! Please let me know how can I send you some money to eat a Pizza with friends offered by me!!!

Andreadj, 2019-10-11:
however... I miss to add an Upload file page (in case to update firmware by OTA and the needings to upload json files... I was reading it can be possible with update creating a spiffs.bin but I get a magic byte error....) and the information to send to MQTT. I think it should not be too much...

Andreadj, 2019-10-11:
Last thing... don't you think that having to create a page for each element like Update, Upload, MQTT, etc .. is too expensive and cumbersome? It would not be better to be able to use Update, Upload etc .. as individual elements to be added to the page you want? For example I'd like to have Update (for OTA update), Upload (to upload json files) and a "Format Spiffs" button (this is yet possible) on the same page but I think i should not be possible at the moment... Well, I don't know if creating less page will free memory to better generate HTML... in case consider it! Just my 2 cents... 
Anyway thanks again for your amazing work!!!

Hieromon, 2019-10-13:
@Andreadj It was a very intense typhoon! Surprisingly, I was not affected and fixed MobileD.
Modified source codes are [here](https://gist.github.com/Hieromon/db0b7881a1c3788c1e19b4c7035d2c02), You can download from its Gists.
I reviewed the source code structure and made some modifications. As a result, it consists of four files.
- **MobileD.ino** : Main logic.
- **ConfigFunction.ino** : Configuration functions which contain AutoConnectAux JSON definitions.
- **HandlerFunction.ino** : AutoConnectAux's handler functions.
- **MqttFunction.ino** : MQTT handler functions.

I tried to understand and correct your intentions as much as possible, but I think there are inadequacies. It hasn't fully tested but will work. Have you try.
If you have some questions, let us continue to consider with each other on the [Gist board](https://gist.github.com/Hieromon/db0b7881a1c3788c1e19b4c7035d2c02).

### IMPORTANT ###
It is necessary newest v110. You need to download or clone [Enhance/v110](https://github.com/Hieromon/AutoConnect/tree/Enhance/v110) branch.

PR #137 (2019-10-16): Enhance/v110
Fixes #132
closed 2019-10-16 by Hieromon