Hieromon/AutoConnect #133: (ESP32)How to get stored wifi credentials?

LaBus3, 2019-09-29:
Here I am trying to get the ssid and password stored in nvs after configuration. Modified the example code as: 

`#include <Arduino.h>
#include <WiFi.h>
#include <WebServer.h>
#include <AutoConnect.h>
#include <AutoConnectCredential.h>

WebServer Server;
AutoConnect Portal(Server);
AutoConnectConfig  config;
void rootPage() {
  char content[] = "Hello, world";
  Server.send(200, "text/plain", content);
}

void setup(){
Serial.begin(115200);
config.apid = "wb.5";
config.psk = "12345678";
//config.autoReconnect = false;
Portal.config(config);
Server.on("/", rootPage);
if (Portal.begin()) {
    Serial.println("WiFi connected: " + WiFi.localIP().toString());
  }
}

AutoConnectCredential creds;
void loop() {
  Portal.handleClient();
  struct station_config  entry;
  creds.load((int8_t)0, &entry);
  Serial.println((char*)entry.ssid);
  p.end();
  delay(1000);
}`

Serial monitor output is: 
`ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:928
ho 0 tail 12 room 4
load:0x40078000,len:8740
load:0x40080400,len:5800
entry 0x4008069c
[E][Preferences.cpp:38] begin(): nvs_open failed: NOT_INITIALIZED
[AC] SoftAP configure 172.217.28.1, 172.217.28.1, 255.255.255.0 
[AC] WiFi.config(IP=0.0.0.0, Gateway=0.0.0.0, Subnetmask=0.0.0.0, DNS1=0.0.0.0, DNS2=0.0.0.0)
[E][WiFiSTA.cpp:219] begin(): connect failed!
[AC] WiFi.begin()
[AC] **Connecting.....................................................................................................time out IP:0.0.0.0**
[AC] SoftAP configure 172.217.28.1, 172.217.28.1, 255.255.255.0 
[AC] SoftAP wb.5/12345678 Ch(1) IP:172.217.28.1 
[AC] http server started
[AC] DNS server started
[AC] Host:,URI:/success.txt,ignored
[E][WebServer.cpp:617] _handleRequest(): request handler not found
[AC] Detected application, detectportal.firefox.com, 0.0.0.0
[AC] Host:detectportal.firefox.com,URI:/success.txt,ignored
[E][Parsing.cpp:275] _parseArguments(): arg missing value: 0
[E][WebServer.cpp:617] _handleRequest(): request handler not found
[AC] Detected application, detectportal.firefox.com, 0.0.0.0
[AC] Host:detectportal.firefox.com,URI:/_ac,generated:/_ac, allocated
[AC] Host:172.217.28.1,URI:/_ac,already allocated
[AC] Host:172.217.28.1,URI:/success.txt,ignored
[E][WebServer.cpp:617] _handleRequest(): request handler not found
[AC] Detected application, detectportal.firefox.com, 0.0.0.0
[AC] Host:detectportal.firefox.com,URI:/_ac,generated:/_ac, allocated
[AC] Host:172.217.28.1,URI:/_ac,already allocated
[AC] Host:172.217.28.1,URI:/success.txt,ignored
[E][Parsing.cpp:275] _parseArguments(): arg missing value: 0
[E][WebServer.cpp:617] _handleRequest(): request handler not found
[AC] Detected application, detectportal.firefox.com, 0.0.0.0
[AC] Host:detectportal.firefox.com,URI:/_ac,generated:/_ac, allocated
[AC] Host:172.217.28.1,URI:/_ac,already allocated
[AC] Host:172.217.28.1,URI:/_ac/config,generated:/_ac/config, allocated
[AC] Host:172.217.28.1,URI:/_ac/config,already allocated
[AC] 1 network(s) found
[AC] Host:172.217.28.1,URI:/success.txt,ignored
[E][WebServer.cpp:617] _handleRequest(): request handler not found
[AC] Detected application, detectportal.firefox.com, 0.0.0.0
[AC] Host:detectportal.firefox.com,URI:/success.txt,ignored
[E][WebServer.cpp:617] _handleRequest(): request handler not found
[AC] Detected application, detectportal.firefox.com, 0.0.0.0
[AC] Host:detectportal.firefox.com,URI:/success.txt,ignored
[E][WebServer.cpp:617] _handleRequest(): request handler not found
[AC] Detected application, detectportal.firefox.com, 0.0.0.0
[AC] Host:detectportal.firefox.com,URI:/_ac/connect,generated:/_ac/connect, allocated
[AC] Host:172.217.28.1,URI:/_ac/connect,already allocated
[AC] Queried SSID:xperio_E3B271
[AC] Attempt:xperio_E3B271 Ch(9)
[AC] Connecting.......established IP:192.168.0.100
[E][Preferences.cpp:38] begin(): nvs_open failed: NOT_FOUND
[AC] Preferences begin failed to import AC_CREDT
[AC]                    xperio_E3B271 credential saved
[AC] Event<17> handler registered
[AC] DNS server stopped
WiFi connected: 192.168.0.100
**0
d
0
d
0
d
0
d**`

creds.entries() seem to return 0. Am I using the API wrong? 
Also why is there a timeout before setting up AP even if there are no credentials saved?

Hieromon, 2019-09-30:
@LaBus3 I had reproduced the issue. The cause is a bug in an AutoConnect. Releases a fix patch, A.S.A.P. 
Thank you for your reports.

Hieromon, 2019-09-30:
@LaBus3 FYI.
The sketch you presented contains a little problem.
It is executing `delay(1000)` unconditionally in the loop. It has nothing to deal with a credential bug, but it will be an obstacle for WebServer working properly.
This sketch will probably break the WebServer session. You may already have the right knowledge, but I informed to you just in case.

closed 2019-09-30 by Hieromon

LaBus3, 2019-09-30:
> @LaBus3 FYI.
> The sketch you presented contains a little problem.
> It is executing `delay(1000)` unconditionally in the loop. It has nothing to deal with a credential bug, but it will be an obstacle [...]

Thank you for pointing out the delay part, I was testing the library in haste and did not notice it. Btw thanks for the great library and the awesome documentation.

LaBus3, 2019-10-20:
@Hieromon creds.entries() return 0, unless a creds.load() is called beforehand. Is this a bug?

Hieromon, 2019-10-21:
Which version of AutoConnect you are using?

LaBus3, 2019-10-21:
Commit 967324b1ff9f0a4a847f28f3bb749ad81ccef269

Test code
```#include <Arduino.h>
#include <WiFi.h>
#include <WebServer.h>
#include <AutoConnect.h>
#include <AutoConnectCredential.h>

WebServer Server;
AutoConnect Portal(Server);
AutoConnectConfig  config;

void setup(){
Serial.begin(115200);
config.apid = "test";
config.psk = "12345678";
Portal.config(config);
if (Portal.begin()) {
    Serial.println("WiFi connected: " + WiFi.localIP().toString());
  }
}

AutoConnectCredential creds;
void loop() {
  Portal.handleClient();
  station_config_t  entry1;
 // creds.load((int8_t)0, &entry1);
  Serial.println(creds.entries());
  for (uint8_t i=0; i<creds.entries(); i++)
  {
    creds.load((int8_t)i, &entry1);
    Serial.println((char*)entry1.ssid);
  }
}
```

OUTPUT
```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
[E][Preferences.cpp:38] begin(): nvs_open failed: NOT_INITIALIZED
[AC] SoftAP configure 172.217.28.1, 172.217.28.1, 255.255.255.0 
[AC] Current:xperio_E3B271
[AC] WiFi.config(IP=0.0.0.0, Gateway=0.0.0.0, Subnetmask=0.0.0.0, DNS1=0.0.0.0, DNS2=0.0.0.0)
[AC] WiFi.begin()
[AC] Connecting..established IP:192.168.0.103
[AC] http server started
WiFi connected: 192.168.0.103
0
0
0
0
0
0
0
0
0
```
OUTPUT with `creds.load((int8_t)0, &entry1); ` uncommented

```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
[E][Preferences.cpp:38] begin(): nvs_open failed: NOT_INITIALIZED
[AC] SoftAP configure 172.217.28.1, 172.217.28.1, 255.255.255.0 
[AC] Current:xperio_E3B271
[AC] WiFi.config(IP=0.0.0.0, Gateway=0.0.0.0, Subnetmask=0.0.0.0, DNS1=0.0.0.0, DNS2=0.0.0.0)
[AC] WiFi.begin()
[AC] Connecting..established IP:192.168.0.103
[AC] http server started
WiFi connected: 192.168.0.103
1
xperio_E3B271
1
xperio_E3B271
1
xperio_E3B271
1
xperio_E3B271
1
```

Hieromon, 2019-10-21:
AutoConnectCredential cannot be declared global.
Since AutoConnect v1.0.3, AutoConnectCredential has migrated to ESP32 Preferences. In the Arduino sketch initialization phase, Preferences are not initialized yet, so AutoConnect cannot get the nvs key for AutoConnectCredential in nvs.
The `creds` declaration must be placed in the setup function or loop function.

LaBus3, 2019-10-22:
Thank you for pointing that out.

PR #134 (2019-09-30): Patches/v103
Fixes #133