Hieromon/AutoConnect #135: OTA Update is not writing on File Systrem

Andreadj, 2019-10-03:
Hi Hieromon,
still here... I'm facing OTA update is not mounting Json files on SPIFFS. It is normal or it's a problem of mine?
2nd: How can I see progress % growing during update (it shoulod be possible?)
3rd: Can I add elements or modify Update page?
Thanks...

Hieromon, 2019-10-07:
@Andreadj 
AutoConnect supports two OTA update methods. One of [Using the web browser](https://hieromon.github.io/AutoConnect/otabrowser.html) and the other one is the [Using the update server](https://hieromon.github.io/AutoConnect/otaserver.html). Based on that...

> I'm facing OTA update is not mounting Json files on SPIFFS.

Are you saying you can't update SPIFFS JSON files using the AutoConnect OTA? If so, that is the correct behavior. OTA is a binary sketch update, not SPIFFS file OTA upload. It is a facility of [AutoConnectFile](https://hieromon.github.io/AutoConnect/acupload.html).

> How can I see progress % growing during update (it shoulod be possible?)

It is only possible with OTA using the update server. The progress bar is not displayed in the method using the Web browser as in the WebUpdate.ino example.

> Can I add elements or modify Update page?

It cannot be possible by each OTA way now and needs to enhance the library.

Andreadj, 2019-10-07:
Ok, thanks a lot!!!

closed 2019-10-07 by Andreadj