Hieromon/AutoConnect #138: Crash due to invalid use of unique_ptr for WebServerClass

Boldie, 2019-10-16:
The following code will trigger a heap corruption on destruction of class or if the pointer_webServer is reset. It is NOT allowed to store a reference to an unique_ptr without using a custom deleter doing a NOOP.

```AutoConnect::AutoConnect(WebServerClass& webServer) {
  _initialize();
  _webServer.reset(&webServer );
  _dnsServer.reset(nullptr);
  _webServerAlloc = AC_WEBSERVER_PARASITIC;
}
```

```
assertion "next > (intptr_t)block" failed: file "/home/sven/workspace/esp/esp-idf/components/heap/multi_heap.c", line 124, function: get_next_block
abort() was called at PC 0x400d44ef on core 0
0x400d44ef: __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:63 (discriminator 8)

ELF file SHA256: 10e8e360d8dab1a972386111fc94087e97f0a0322219777e479ffa0f83d8b971

Backtrace: 0x4008700b:0x3ffbc1d0 0x40087325:0x3ffbc1f0 0x400d44ef:0x3ffbc210 0x4008dfca:0x3ffbc240 0x4008e267:0x3ffbc260 0x4008e3e1:0x3ffbc280 0x400826b5:0x3ffbc2a0 0x40082a05:0x3ffbc2c0 0x4000bec7:0x3ffbc2e0 0x400e5d6d:0x3ffbc300 0x400d3ed8:0x3ffbc320 0x40158997:0x3ffbc340 0x400e966f:0x3ffbc360 0x400e9e76:0x3ffbc380 0x400d43fa:0x3ffbc3a0 0x400d391d:0x3ffbc850 0x400d1f6f:0x3ffbc870
0x4008700b: invoke_abort at /home/sven/workspace/esp/esp-idf/components/esp32/panic.c:715

0x40087325: abort at /home/sven/workspace/esp/esp-idf/components/esp32/panic.c:715

0x400d44ef: __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:63 (discriminator 8)

0x4008dfca: get_next_block at /home/sven/workspace/esp/esp-idf/components/heap/multi_heap.c:699

0x4008e267: assert_valid_block at /home/sven/workspace/esp/esp-idf/components/heap/multi_heap.c:699

0x4008e3e1: multi_heap_free_impl at /home/sven/workspace/esp/esp-idf/components/heap/multi_heap.c:699

0x400826b5: heap_caps_free at /home/sven/workspace/esp/esp-idf/components/heap/heap_caps.c:232

0x40082a05: _free_r at /home/sven/workspace/esp/esp-idf/components/newlib/syscalls.c:42

0x400e5d6d: operator delete(void*) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/del_op.cc:46

0x400d3ed8: RequestHandler::~RequestHandler() at /home/sven/workspace/esp/water-me/main/portal.cpp:190

0x40158997: std::default_delete<AutoConnectAux>::operator()(AutoConnectAux*) const at /home/sven/workspace/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/bits/unique_ptr.h:76

0x400e966f: std::unique_ptr<AutoConnectAux, std::default_delete<AutoConnectAux> >::~unique_ptr() at /home/sven/workspace/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/functional:1974 (discriminator 1)

0x400e9e76: AutoConnect::~AutoConnect() at /home/sven/workspace/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/5.2.0/functional:1974 (discriminator 1)
```

Boldie, 2019-10-16:
It seems to be related to AutoConnectAux instance handling.  Examples suggests to pass reference to void AutoConnect::join(AutoConnectAux& aux) which will be invalid on delete of uniqe_ptr. From my opinion, the construct is highly buggy here, for such situations it is favourable to be able to set a custom deleter, consider using a shared_ptr with a custom deleter instead or use a unique_ptr with the extra template argument as custom deleter and set them explicitly on every assignment.

Hieromon, 2019-10-17:
@Boldie Thank you for teaching me. I agree with your suggestion.
The _aux pointer of AutoConnect and the _ac pointer of AutoConnectAux are cross-handling each other as unique_ptr.  It is an invalid structure that causes access violations during destruction by default destructor.
I also overlooked It, and the termination process has not tested enough.
I confirm your PR.

Hieromon, 2019-10-17:
Closed due to merged #141

closed 2019-10-17 by Hieromon