Hieromon/AutoConnect #143: AUTOCONNECT_MENUCOLOR not affecting output

LaBus3, 2019-10-17:
```
#define AUTOCONNECT_MENUCOLOR_TEXT "#FFFFFF"
#define AUTOCONNECT_MENUCOLOR_BACKGROUND "#7B1FA2" 
#define AUTOCONNECT_MENUCOLOR_ACTIVE "#E1BEE7"
```
Added these to main.cpp but the color reverts to the default values.

LaBus3, 2019-10-17:
> > Added these to main.cpp
> 
> It is no meaning. You have to modify AutoConnectLabels.h.
> https://hieromon.github.io/AutoConnect/colorized.html#autoconnect-menu-colorizing

Why not have an option to change variables like these within the sketch? And preserve the library.

Hieromon, 2019-10-17:
The color scheme is determined at compile time to reduce the heap consuming for dynamically generating HTML. The HTML template that is the source for dynamic generation contained in the separated source code of the library and the color scheme is determined when its source is compiled.

If the compiler knows the #define directive with a color scheme that you have defined at compile-time, your wishes will come true.
The easiest way is to pass the macro definition as a compiler argument with `-D` at build time like as the below:
```
gcc -DAUTOCONNECT_MENUCOLOR_BACKGROUND=#7B1FA2
```
I think that is possible with PIO build configuration definition, is not possible?

Hieromon, 2019-10-17:
The `-D` option is passed directly to the cc preprocessor. It must be a complete string replacement with `#define` directive and declares
```
"#FFFFFF"
```
Not
```
#FFFFFF
```
Your .ini passes #FFFFFF and is wrong. It is not the AutoConnect library issue and is simply a mistake in the syntax of platformio.ini you coded.

```
build_flags =
    -DAUTOCONNECT_MENUCOLOR_TEXT='"#ffa500"'
    -DAUTOCONNECT_MENUCOLOR_BACKGROUND='"#1a237e"'
    -DAUTOCONNECT_MENUCOLOR_ACTIVE='"#283593"'
```

LaBus3, 2019-10-19:
That worked, thank you for pointing out the error