Hieromon/AutoConnect #145: [ESP32] Device freeze on pressing Disconnect option without configuring AP

LaBus3, 2019-10-19:
Two issues I have seen is :
1. Up until WiFi is configured, portal blocks other code in loop(). Is this the default behaviour?
2. If WiFi is not configured and Disconnect is selected, the code stays blocked, which is not the case if WiFi is configured. Reset on disconnect need to be enabled to prevent blocking. This is not always a favourable situation and also Disconnect option become redundant with Reset option.

Please comment on this.
Thank you

Hieromon, 2019-10-20:
It is an intentional loop, and AutoConnect::begin is intended to replace WiFi.begin, not asynchronous.
Also, Disconnect does not simply disconnect from an access point. It has the effect of resetting the current WiFi mode of the ESP module as well as WiFi.disconnect(). Disconnecting means stopping AP_STA mode and placing the sketch in the stand-alone state or restarting the sketch in STA mode.
Your situation seems to be similar to a past issue, refer to #100.

LaBus3, 2019-10-21:
Yes, that seem to be vaguely similar to my issue.

What configuration need to be done to prevent STA to STA_AP transition on WiFi connection failure, if I am using Autoconnect to manage my connections? I just need to know if I can simply detect if connection failure occurs (`portal.begin()?`) and go on with my sketch until next connection attempt (next time `portal.begin()`is called). `autoReconnect=true`, and `retainPortal=false` is enough?

Hieromon, 2019-10-21:
I don't have any idea what "next portal.begin" and is unclear under what conditions should be executed, but maybe you can do it with the [onDetect](https://hieromon.github.io/AutoConnect/api.html#ondetect) exit.
The onDetect exit routine is called before starting STA_AP if the STA connection fails. If it ’s a situation you don’t expect, the exit routine returns the false. Then the captive portal is canceled and the WebServer will not start.

LaBus3, 2019-10-22:
> I don't have any idea what "next portal.begin" and is unclear under what conditions should be executed, but maybe you can do it with the [onDetect](https://hieromon.github.io/AutoConnect/api.html#ondetect) exit routine or [AutoConnectConfig::autoRise](https://hieromon.github.io/AutoConnect/apiconfig.html#autorise) option.
> 
> If you just want to suppress the transition to WIFI_AP_STA, use the autoRise option.
> 
> The onDetect exit routine can cancel the captive portal while starting the WIFI_AP_STA after the webserver started. If the access point connection fails, the onDetect exit routine is useful if the ESP module needs to interact via the webserver.
> If the WIFI_STA connection fails, the onDetect exit routine will be called after WIFI_AP_STA starts. In unexpected situations, the exit routine returns false. After that, the captive portal is canceled and WebServer starts. At this point, AutoConnectBegin returns to the sketch.

My requirement is that I only want WiFi.begin() to start with the saved credentials. I am not required to start the http server nor the SoftAP. I want to start WiFi.begin() and detect if WiFi.begin() failed or not with all the saved credentials, as well as transfer control back to sketch. What configuration should I use for this application? Right now I am using AutoConnectCredentials::load() coupled with WiFi.begin() to manually connect to WiFi and detect failiure.

Note: The saved credentials I am talking about is the WiFi SSID and PSK saved inside nvs/preferences.

Hieromon, 2019-10-22:
Hmm...I have come to understand what you want to do.
How do you establish a connection with an access point that has never connected? AutoConnectCredential does not have its credentials.

LaBus3, 2019-10-22:
Incidentally if no access point has been configured beforehand, I switch to an LTE modem, for this I am using  AutoConnectCredentials::entries() to detect if any AP has been configured beforehand.
At this point I do not require the AutoConnect::begin() to go past WiFi.begin(). As such I as inquiring if any of the configuration need to be set can be set for that.
I am evaluating the use of this library for applications not limiting to only WiFi configuration. The custom page builder is quite flexible which is an advantage.

I hope this is not getting too OT

Hieromon, 2019-10-22:
How do you save the AutoConnectCredential?

LaBus3, 2019-10-22:
> How do you save the established credential as AutoConnectCredential data to the nvs?

I raise the portal using a hardware button, configure device parameters in a custom page and then Configure the AP,  if required. If not Disconnect and let the sketch handle the rest.

Hieromon, 2019-10-22:
Ok, I will bulletin some suggestions to implement your requirements as follows:

1. Promote _loadAvailCredentials to the public method
AutoConnect::begin failure can be predicted by using loadAvailCredentials prior to AutoConnect::begin. However, this is only a prediction and No SSID is not the only factor that causes WiFi.begin to fail.
This implementation is simple and you can do it yourself with AuutoConnect.h modification.
https://github.com/Hieromon/AutoConnect/blob/967324b1ff9f0a4a847f28f3bb749ad81ccef269/src/AutoConnect.h#L231
https://github.com/Hieromon/AutoConnect/blob/967324b1ff9f0a4a847f28f3bb749ad81ccef269/src/AutoConnect.cpp#L617
    ```cpp
    AutoConnect portal;
    void setup() {
      if (portal._loadCredentail(nullPtr)) {
        AutoConnectConfig config;
        config.autoReconnect = true;
        portal.config(config);
        portal.begin();
      }
    }
   ```

2. Improve the new function which can call exit routine when WiFi.beign fails like as follows:
    It is the `onFail` exit routine, and its implementation requires AutoConnect enhancements but allows you to create flexible sketches that fit your situation.
    ```cpp
    AutoConnect portal;

    // Determine whether to continue the Autoconnect::begin process.
    bool determine(const char* ssid) {
      if (...)
        // Continues the AutoConnect::begin
        return true;
      else
        // Exit from the AutoConnect::begin
        return false;
    }
    
    void setup() {
      portal.onFail(determine);	// Register an exit routine.
      portal.begin();
    }
    ```
Both are specification changes. Which is useful as a necessity?

Hieromon, 2019-10-22:
Your requirements are fragmented and not clearly communicated to me yet.
Please refer to the documentation of [AutoConnect::begin](https://hieromon.github.io/AutoConnect/lsbegin.html) the logic sequence and consider how well your requirements fit the logic sequence.
AutoConnect may not be suitable for your requirements If you only want a dynamic HTML generation utility.

LaBus3, 2019-10-22:
This is exactly what I am doing right now, thank you for your help. I shall proceed this way.
NB: I was inquiring if there was  any configuration that limits portal.begin() to what the connectWiFi() method does, I  understand there isn't any. Sorry  your wasting your time.

closed 2019-10-22 by LaBus3