Hieromon/AutoConnect #146: Error with new version

LorenzoCerda, 2019-11-01:
Hi, I am using AutoConnect in Platformio (VSCode) and until version 1.0.3 it works ok, but when I upgraded it to version 1.1.2 I am getting the following error. I am also including the lines of code where the error is present. Any idea? Thank you in advance (I never touch these lines, btw).

Thank you in advance!

src/cerco-dev-01.cpp: In function 'String viewCredential(PageArgument&)':
src/cerco-dev-01.cpp:263:26: error: aggregate 'viewCredential(PageArgument&)::station_config entry' has incomplete type and cannot be defined
   struct station_config  entry;
                          ^
src/cerco-dev-01.cpp: In function 'String delCredential(PageArgument&)':
src/cerco-dev-01.cpp:284:30: error: aggregate 'delCredential(PageArgument&)::station_config entry' has incomplete type and cannot be defined
       struct  station_config entry;
                              ^

**CODE (I couldn't add it correctly with the "insert code" correctly... the bullets don't exist, for example)**

// Retrieve the credential entries from EEPROM, Build the SSID line
// with the <li> tag.
String viewCredential(PageArgument& args) {
  AutoConnectCredential  ac(CREDENTIAL_OFFSET);
  struct station_config  entry;
  String content = "";
  uint8_t  count = ac.entries();          // Get number of entries.

  for (int8_t i = 0; i < count; i++) {    // Loads all entries.
    ac.load(i, &entry);
    // Build a SSID line of an HTML.
    content += String("<li>") + String((char *)entry.ssid) + String("</li>");
  }

  // Returns the <li>SSID</li> container.
  return content;
}

// Delete a credential entry, the entry to be deleted is passed in the
// request parameter 'num'.
String delCredential(PageArgument& args) {
  AutoConnectCredential  ac(CREDENTIAL_OFFSET);
  if (args.hasArg("num")) {
    int8_t  e = args.arg("num").toInt();
    if (e > 0) {
      struct  station_config entry;

      // If the input number is valid, delete that entry.
      int8_t  de = ac.load(e - 1, &entry);  // A base of entry num is 0.
      if (de > 0) {
        ac.del((char *)entry.ssid);

        // Returns the redirect response. The page is reloaded and its contents
        // are updated to the state after deletion. It returns 302 response
        // from inside this token handler.
        Server.sendHeader("Location", String("http://") + Server.client().localIP().toString() + String("/"));
        Server.send(302, "text/plain", "");
        Server.client().flush();
        Server.client().stop();

        // Cancel automatic submission by PageBuilder.
        delPage.cancel();
      }
    }
  }
  return "";
}

Hieromon, 2019-11-05:
A `station_config` structure has already [deprecated](https://hieromon.github.io/AutoConnect/faq.html#saved-credentials-are-wrong-or-lost).
The replaced structure is [**station_config_t**](https://hieromon.github.io/AutoConnect/credit.html#station_config_t).

Hieromon, 2019-12-12:
Closed due to no activity for one month. If a new report occurs, open it again.

closed 2019-12-12 by Hieromon