Hieromon/AutoConnect #149: Is there a way to make it non blocking when calling the portal.begin()

sap-auto, 2019-11-04:
After calling portal.begin() in the setup() the function blink() in main Loop() is never serviced till a wifi connection is made in case the known network is not available and Softap is started but the loop is not serviced. Is it possible to make this non blocking.

void loop() {

  server.handleClient();
  portal.handleRequest();   // Need to handle AutoConnect menu.
  if (WiFi.status() == WL_IDLE_STATUS) {
#if defined(ARDUINO_ARCH_ESP8266)
    ESP.reset();
#elif defined(ARDUINO_ARCH_ESP32)
    ESP.restart();
#endif
    delay(1000);
  }
    Blink();
}

void Blink(){
  unsigned long currentMillis = millis();

  if (currentMillis - previousMillis >= interval) {
        previousMillis = currentMillis;
    if (ledState == LOW) {
      ledState = HIGH;
    } else {
      ledState = LOW;
    }
    digitalWrite(led, ledState);
  }
}

Hieromon, 2019-11-06:
For now, you can avoid it with the following combination of configuration options:
- AutoConnectConfig::portalTimeout
- AutoConnectConfig::retainPortal

By setting the `portalTimeout` to 0 and setting the `retainPortal` to true, you can return to the sketch from AutoConnect::begin while maintaining the AP_STA mode for the captive portal.

Hieromon, 2019-12-12:
@sap-auto I made the mistake the portalTimeout threshold to escape from the AutoConnect:: begin loop I mentioned above. The `portalTimeout=0` means no timeout limitation. Specify `portTimeout = 1` instead. It will escape from the Captive portal loop after 1[ms].

Hieromon, 2019-12-31:
Close once due to no activity record for one month.
Reopen when new reports are posted.

soemarko, 2020-06-26:
Apologies for reviving closed issue. Setting a portal timeout and retain portal seems like an acceptable answer, but it seems like `portalTimeout = 1` is waaay longer than 1ms. I timed it, it's about 30 secs. Felt like forever though.

I think making portal.begin() non blocking is the ideal solution here. Any devices would be better if WiFi is just optional.

Hieromon, 2020-09-24:
@soemarko 
>  it seems like portalTimeout = 1 is waaay longer than 1ms.

A portalTimeout will not be measured unless the ESP32 SoftAP is completely standalone. SoftAP owns the station as long as someone keeps looking into the esp32ap from their smartphone.
So, can you post the log output by AutoConnect, the sketch code (Reduce to the reproducible minimum code for the problem) and explain the situation where the phenomenon occurs? So, can you post the log output by AutoConnect, the sketch code (Reduce to the reproducible minimum code for the problem) and explain the situation where the phenomenon occurs?

> I think making portal.begin() non blocking is the ideal solution here. Any devices would be better if WiFi is just optional.

This advice is also ambiguous. AutoConnect::begin is not asynchronous in the first place. Please refer to the [Inside AutoConnect::begin diagram](https://hieromon.github.io/AutoConnect/lsbegin.html) to suggest your changes to clarify your desired specifications. Of course, we need to envision various use cases.

Hieromon, 2020-10-20:
In v1.20, you can prevent the blocking of the Sketch's process by AutoConnect and can avoid in the following two ways:

1. AutoConnect::whileCaptivePortal exit https://github.com/Hieromon/AutoConnect/commit/b26e4f90c28b412994211f04018648dcf52c05ff
Prevents Sketch process from stalling due to waiting for a connection from a client after AutoConnect::begin enters the captive portal state. See #244

2. AutoConnectConfig::autoReconnect with reconnectInterval https://github.com/Hieromon/AutoConnect/commit/093bb22f34b2271b34622903820b7c8328dc4acc
AutoConnectConfig gets a new option for auto-reconnect working behind AutoConnect::handleClient.  
Prior to v1.20, autoReconnect was only valid while AutoConnect::begin was running. In v120 or later, Sketch can attempt to reconnect while handleClient is running. AutoConnectConfig::reconnectInterval specifies the interval time to try automatic reconnection.

Hieromon, 2020-12-02:
Merged #280

closed 2020-12-02 by Hieromon