Hieromon/AutoConnect #150: not connecting after pressing "enter" on the phone keyboard after typing in the password

ernfel, 2019-11-07:
I came across the following, it would be nice if this could be improved:
After choosing a WIFI SSID for initial connection, i enter the password in the field (on my smartphone). Pressing the enter button on the phone keyboard triggers to triggers to go to the next page of found wifi ssids, instead of trigger the action of connecting. 
This was confusing, because at first i didnt unterstand, why it is not connecting and why my Wifi is not shown anymore (since it was on page 1 of found wifi ssids, and after pressing "enter", page 2 is shown)

Hieromon, 2019-11-08:
Its behaviors not my intention, and I reproduced the issue on Safari with iOS.
It seems that degradation has occurred due to the support of static IP configuration.
I will fix it.

Hieromon, 2019-11-08:
@ernfel I improved the button behavior with Config New.
Even if the focus is on the password field, the connection attempting starts immediately by tapping ENTER on the smartphone soft-keyboard.
You can try the improved version with the [patches/v113](https://github.com/Hieromon/AutoConnect/tree/patches/v113) branch.

ernfel, 2019-11-09:
@Hieromon i tested the improved version with my android phone, it worked as desired. Thank you for the quick fix!

commit 231e2948 (2019-12-31): Merge pull request #172 from Hieromon/patches/v113

Fixes #150
closed 2019-12-31 by Hieromon