Hieromon/AutoConnect #152: Choosen WiFi does't connect with right key

kkberg, 2019-11-16:
Hello,
last week I flashed a NodeMCU LOLIN with a sketch and connecting to my Wifi worked perfectly. I'm now working on another sketch and an equal NodeMCU My wifi is shon in Autoconnect, but after havong used my wifi key it doesn't connect. SSID is shown and the "running" ball around it is shown very long and then connection is lost.  I included my sketch, may be I defined something wrong.
Sketch is compiled without any error. Rename file to .ino .
[Code-Wettermonster_Autologon.ino.txt](https://github.com/Hieromon/AutoConnect/files/3854597/Code-Wettermonster_Autologon.ino.txt)

Hieromon, 2019-11-17:
Long-time delay in the loop() your sketch has. ESP8266WebServer is not asynchronous execution and cannot respond with this logic.
If you want to handle with AutoConnect real-timely and take temperature measurements every few minutes, you only need to delay the readBME interval. Cannot delay portal.handleClient.
```cpp
unsigned long tm = millis();

void loop() {
  if (millis() - tm > (interval * 60000)) {
    tm = millis();
    readBME();
    sendToWettermonster();
  }
  Portal.handleClient();
}
```

kkberg, 2019-11-17:
Thank you very much. I'll change it to the following and try it.

unsigned long tm = millis();

void loop() {
 Portal.handleClient();
  if (millis() - tm > (interval * 60000)) {
    tm = millis();
    readBME();
    sendToWettermonster();
  }
 
}

Hieromon, 2019-12-31:
Close once due to no activity record for one month.
Reopen when new reports are posted.

closed 2019-12-31 by Hieromon

kkberg, 2019-12-31:
sorrry, forgot to close