Hieromon/AutoConnect #154: Problems with AutoConnectCredential.entries() always returns zero

b0153n, 2019-11-21:
Hi Hieromon,

I am having some difficulty with your AutoConnectCredential class.

Chip: ESP32
Target hardware: LILYGO TTGO T-Journal module
IDE: VS Code, PlatformIO
Developer level: Beginner

I have implemented your AutoConnect library.
I have successfully compiled and run the code.
It did exactly as expected and entered into SoftAP mode and created the captive portal.
I have entered the SSID and PSK for my home access point into my phone and the credentials are saved successfully by your code.
If I reboot the target hardware it automatically connects to my access point exactly as expected.

However, I want to implement a method of clearing all the credentials and resetting the device so that it will create the captive portal on the next boot (reset).
I followed your code covering “Saved credentials access” at https://hieromon.github.io/AutoConnect/credit.html
You give an example ‘deleteAllCredentials’ about half way down the page.
The problem I am having is that ‘credential.entries()’ always returns zero. However there must be credentials as I set them up through the captive portal and the device can connect to my access point.
I also tried to brute force the delete by simply calling the ‘credential.delete(<SSID>)’ method while passing in my access point SSID.
The delete did not appear to work as the device still connects to my access point.

In summary:
I must be doing something wrong as I cannot get ‘credential.entries()’ or ‘credential.delete(<SSID>)’ to work at all.

My full code here: https://github.com/b0153n/WifiCam1/blob/master/src/main.cpp

I hope you can offer some advice.

Many thanks.

Hieromon, 2019-11-21:
A problem you have is due to the ambiguity of the AutoConnectCredential specification. 
AutoConnectCredential instance works independently of AutoConnect. AutoConnect does not consider the AutoConnectCredential global variable (https://github.com/b0153n/WifiCam1/blob/d1df8b3b17142d96dc282f7725a4825256abc1fe/src/main.cpp#L54 in your sketch).
If you use AutoConnectCredential in a sketch, you should always capture the "physical state of the moment" with nvs updated by AutoConnect. In particular, instantiate it when you use it.
By declaring AutoConnectCredential as a local variable of the deleteAllCredentials function, you can get the latest number of entries at that time.
See also [same mention](https://github.com/Hieromon/AutoConnect/issues/133#issuecomment-544669601).

b0153n, 2019-11-22:
Hi Hieromon,
Thank you for your prompt reply.

I have moved the declaration for both:
AutoConnectCredential credential(CREDENTIAL_OFFSET);
station_config_t config;
to the deleteAllCredentials function.

There is no change in behaviour. credential.entries() still returns zero.

I have compiled and uploaded your example code for displaying and deleting credentials:
https://github.com/Hieromon/AutoConnect/blob/master/examples/Credential/Credential.ino
It does successfully display a web page when I browse to it, however there are no entries in the list on that page.
In that code ‘ac.entries()’ returns zero for me also.
There must be one set of credentials – the device connects to my home access point.

I don’t think it could be faulty hardware as AutoConnect originally correctly stored one set of credentials and is able to load them to connect to my home access point.

I must still be doing something wrong…

Hieromon, 2019-11-22:
> https://github.com/Hieromon/AutoConnect/blob/master/examples/Credential/Credential.ino
> however there are no entries in the list on that page.

I tested on my ESP32 with your showed up code, but cannot reproduce the issue. Stored credentials are displayed correctly in the browser.
If you increase the version of the arduino-esp32 core, you will need to erase the flash once by the esptool.

[image]

b0153n, 2019-11-23:
Hi Hieromon,
Thank you for your prompt reply.

I have uploaded the program to a brand new module.
After boot up and the 30 second delay it entered Captive Portal mode as expected.
I connected to it and entered my WiFi credentials from my phone.
It saves the credentials and then correctly connects to my WiFi network.
I then reboot the device and it again connects to my WiFi network as expected.
I then hold down my user defined button that invokes the deleteAllCredentials() function and press the reset button.
From deleteAllCredentials() my Serial.println statement confirms that credential.entries() returns the expected number of 1.
The device then connects to my WiFi network, which is not what was expected.
I then reset the device again and it connects to my WiFi network.
I then hold down my user defined button again and press the reset button.
This time credential.entries() returns zero.

Thank you very much for your assistance.

Below is the monitor terminal output.
I have inserted my annotations between groups of ****** characters.

`> Executing task: C:\Users\mp3\.platformio\penv\Scripts\platformio.exe device monitor <

--- Available ports:
---  1: COM3                 'USB Serial Port (COM3)'
---  2: COM7                 'Silicon Labs CP210x USB to UART Bridge (COM7)'
--- Enter port index or full name: 2
--- Miniterm on COM7  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac

CapPortal SSID = F03E8C-CAM  
CapPortal PSK = forkcam2019
Display Initalised. ******First time boot on new hardware. Captive Portal is activated as expected.******
[E][Preferences.cpp:38] begin(): nvs_open failed: NOT_FOUND
[E][WiFiSTA.cpp:219] begin(): connect failed!
dhcps: send_offer>>udp_sendto result 0
[E][WebServer.cpp:617] _handleRequest(): request handler not found
[E][WebServer.cpp:617] _handleRequest(): request handler not found
[E][WebServer.cpp:617] _handleRequest(): request handler not found
[E][WebServer.cpp:617] _handleRequest(): request handler not found
[E][WebServer.cpp:617] _handleRequest(): request handler not found
[E][WebServer.cpp:617] _handleRequest(): request handler not found
[E][WebServer.cpp:617] _handleRequest(): request handler not found
[E][WebServer.cpp:617] _handleRequest(): request handler not found
[E][WebServer.cpp:617] _handleRequest(): request handler not found
[E][WebServer.cpp:617] _handleRequest(): request handler not found
[E][WebServer.cpp:617] _handleRequest(): request handler not found
[E][WebServer.cpp:617] _handleRequest(): request handler not found
[E][WebServer.cpp:617] _handleRequest(): request handler not found
[E][WebServer.cpp:617] _handleRequest(): request handler not found
[E][Preferences.cpp:38] begin(): nvs_open failed: NOT_FOUND
WiFi connected: 192.168.1.119  ******Finished entering WiFi details on my phone. The device has connected to my WiFi network.******

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac

CapPortal SSID = F03E8C-CAM
CapPortal PSK = forkcam2019
Display Initalised.
WiFi connected: 192.168.1.119  ******Reset button re-boot. Connects to my WiFi network as expected.******

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac

CapPortal SSID = F03E8C-CAM  ******Reset button re-boot while holding the user defined button down that envokes deleteAllCredentials().******
CapPortal PSK = forkcam2019
Display Initalised.
Delete stored credentials: 1   ******YES! Found the credentials - but they do not appear to have been deleted as you can see on the next line.******
WiFi connected: 192.168.1.119

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac

CapPortal SSID = F03E8C-CAM
CapPortal PSK = forkcam2019
Display Initalised.
WiFi connected: 192.168.1.119  ******Just a simple reset, but the credentials still exist.******

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac

CapPortal SSID = F03E8C-CAM
CapPortal PSK = forkcam2019
Display Initalised.
Delete stored credentials: 0   ******Attempted to delete the credentials again. None found, however they must be there as you can see on the next line.******
WiFi connected: 192.168.1.119
`

b0153n, 2019-11-23:
Note: i am using VS Code, PlatformIO, Espressif32 ver 1.11.1
Thank you.

Hieromon, 2019-11-23:
Please turn on AC_DEBUG with platformio.ini and post an outputted debug log.

b0153n, 2019-11-24:
Hi Hieromon,
Thank you for your prompt reply.

Two logs follow:
The first one is with build_flags = -DAC_DEBUG as requested.
The second one is where I have replaced the deleteAllCredentials() code with an attempt to directly manipulate the Preferences object using the AC_CREDT namespace.

Log one:
`> Executing task: C:\Users\mp3\.platformio\penv\Scripts\platformio.exe device monitor <

--- Available ports:
---  1: COM3                 'USB Serial Port (COM3)'
---  2: COM7                 'Silicon Labs CP210x USB to UART Bridge (COM7)'
--- Enter port index or full name: 2
--- Miniterm on COM7  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---

*Plain boot up*
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac

[AC] SoftAP configure 172.217.28.1, 172.217.28.1, 255.255.255.0 
CapPortal SSID = F03E8C-CAM
CapPortal PSK = forkcam2019
Display Initalised.
[AC] Current:b0153n
[E][Preferences.cpp:457] getBytesLength(): nvs_get_blob len fail: AC_CREDT NOT_FOUND
[AC] WiFi.config(IP=0.0.0.0, Gateway=0.0.0.0, Subnetmask=0.0.0.0, DNS1=0.0.0.0, DNS2=0.0.0.0)
[AC] WiFi.begin()
[AC] Connecting.established IP:192.168.1.119
[AC] http server started
WiFi connected: 192.168.1.119

*Boot up with the user button pressed to invoke deleteAllCredentials()*
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac

[AC] SoftAP configure 172.217.28.1, 172.217.28.1, 255.255.255.0 
CapPortal SSID = F03E8C-CAM
CapPortal PSK = forkcam2019
Display Initalised.
[E][Preferences.cpp:457] getBytesLength(): nvs_get_blob len fail: AC_CREDT NOT_FOUND
Delete stored credentials: 0
[AC] Current:b0153n
[E][Preferences.cpp:457] getBytesLength(): nvs_get_blob len fail: AC_CREDT NOT_FOUND
[AC] WiFi.config(IP=0.0.0.0, Gateway=0.0.0.0, Subnetmask=0.0.0.0, DNS1=0.0.0.0, DNS2=0.0.0.0)
[AC] WiFi.begin()
[AC] Connecting.established IP:192.168.1.119
[AC] http server started
WiFi connected: 192.168.1.119
`

Log two:
`> Executing task: C:\Users\mp3\.platformio\penv\Scripts\platformio.exe device monitor <

--- Available ports:
---  1: COM3                 'USB Serial Port (COM3)'
---  2: COM7                 'Silicon Labs CP210x USB to UART Bridge (COM7)'
--- Enter port index or full name: 2
--- Miniterm on COM7  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac

[AC] SoftAP configure 172.217.28.1, 172.217.28.1, 255.255.255.0 
CapPortal SSID = F03E8C-CAM
CapPortal PSK = forkcam2019
Display Initalised.
The user button is pressed - delete all credentials so that the Soft AP and captive portal are started.
The sledgehammer approach:
Instanciate a 'Preferences' object
Call .begin() with AC_CREDIT as the namespace
prefs.begin('AC_CREDT', false) returned 1
Call the .clear() method of the 'Preferences' object
prefs.clear() returned 1
Call the .end() method of the 'Preferences' object
[AC] Current:b0153n
[E][Preferences.cpp:457] getBytesLength(): nvs_get_blob len fail: AC_CREDT NOT_FOUND
[AC] WiFi.config(IP=0.0.0.0, Gateway=0.0.0.0, Subnetmask=0.0.0.0, DNS1=0.0.0.0, DNS2=0.0.0.0)
[AC] WiFi.begin()
[AC] Connecting.established IP:192.168.1.119
[AC] http server started
WiFi connected: 192.168.1.119
`

Many thanks for your help.

Hieromon, 2019-11-24:
@b0153n Thank you for taking the log.

> This looks to be an issue, but I don't know how to interpret it:
> [E][Preferences.cpp:457] getBytesLength(): nvs_get_blob len fail: AC_CREDT NOT_FOUND

I look like it works fine.  AC_CREDT namespace exists on NVS but preferences key as AC_CREDT does not exist. The Preferences namespace and key name is a different object.
AutoConnectCredential stores credentials with the key AC_CREDT in the namespace AC_CREDT.
This state indicates that the credentials were not written correctly or that only the Preferreds key was deleted. This is not the behavior of AutoConnectCredential.
If AutoConnect cannot write the credentials correctly, it will output an error log based on the return value from AutoConnectCredential.

b0153n, 2019-12-01:
Hi Hieromon,
I have found a work around.
I can force 'immediateStart' to get back into the captive portal.
Thank you for your help.

closed 2019-12-01 by b0153n