Hieromon/AutoConnect #155: override #define from AutoConnectDefs.h

adynis, 2019-11-23:
Hi,
First: AutoConnect it's an Excelent package (library+documentation) !!! Thank you !!

I'm quite new to Arduino environment, so ... here is my **question**:
Which is the best method to override one ore more #define statements from AutoConnectDefs.h ? (e.g.: AUTOCONNECT_URI)
- I just read that I am unable to #define it before including the AutoConnect.h or Undefining it and then re-defining it, because of the way Arduino is internally compiling stuf ... (And I also tested, foudn no working solution)
- If I edit in place, in the directory where the library was downloaded, I suppose I will have issues if I will ever update the library (and I might also forget about those changes)

Which is the recommended solution?

Thanks,

Hieromon, 2019-11-23:
A good way for you to override define directives per sketch without making temporary modifications to AutoConnectDefs.h is to use PlatoformIO.
By using [VSCode](https://code.visualstudio.com/) combined [PlatformIO IDE](https://platformio.org/platformio-ide), you will be able to develop comfortably without using the Arduino IDE.

You can have platformio.ini that can temporarily define the `#define` directives with `build_flags` for each sketch  as like:
```ini
[env:nodemcuv2]
platform = espressif8266
board = nodemcuv2
framework = arduino
lib_ldf_mode = chain+
build_flags =
  -DAC_DEBUG
  -DAUTOCONNECT_PSK='"newpassword"'
  -DAUTOCONNECT_AP_IP=0x01F4A8C0
  -DAUTOCONNECT_AP_GW=0x01F4A8C0
  -DAUTOCONNECT_URI='"/_portal"'
upload_speed = 921600
monitor_speed = 115200
```

adynis, 2019-11-23:
Excellent ! thank you, VSCode + PlatformIO looks much better ! And also the #define re-definition works nice !

Hieromon, 2019-11-24:
Ok, can you close the topic?
If you cannot override the `#define` directives using the PlatformIO, please reopen this issue.
Thank you.

closed 2019-11-24 by adynis