Hieromon/AutoConnect #157: How to change url of BUILT-IN LED page in esp32_HandlePortalEX sketch.

mellowmonk, 2019-11-26:
I am totally in love with this library and cant stop customizing it. （ΦωΦ）

**Sketch used:** esp32_HandlePortalEX.ino
**Device:** ESP32 Development board

In the AutoConnect Menu;
I wanted to change to the location of the **HOME** page.
(whenever i click on home, it should take me to another page, instead of the _BUILT-IN LED_ page.)

So i modified a few lines in the _"AutoConnectPage.cpp"_ file:
`#define AUTOCONNECT_HOMEURI     "/_ac"`

**This worked,** and I am able to navigate to another page whenever i click on HOME.

Now, I created another menu item, _"TEST"_. 
(I would like this page to go to the BUILT-IN LED page. Ideally, (ip address)/test)

However, whenever i click on, "TEST" it takes me back to the, "/_ac" page.

My problem now is that i cannot find a way to call the BUILT-IN LED page.

**Another way to ask the question:**
How do i change the url of **_root_** page to something like, "(ip address)/BUILT-IN LED" page.

Thanks.

[image]

Hieromon, 2019-11-28:
@mellowmonk Thank you for your interest in AutoConnect.
First, To change the link of the HOME item in the menu, using the [AutoConnect::home](https://hieromon.github.io/AutoConnect/api.html#home) function can change the URI without AutoConnectDefs.h modification.

> I created another menu item, "TEST".
Is it an AutoConnectAux menu item? Or is it a menu item added by directly modifying AutoConnectPage.cpp?

The location of a URI of the BUILTIN_LED page is defined in HandlePortalEX.ino is as follows:

https://github.com/Hieromon/AutoConnect/blob/ac8687939f868267383adbb543943dce4b47e838/examples/HandlePortalEX/HandlePortalEX.ino#L98

https://github.com/Hieromon/AutoConnect/blob/ac8687939f868267383adbb543943dce4b47e838/examples/HandlePortalEX/HandlePortalEX.ino#L106

You can change these two places to `/builtin_led` and have a link to `/builtin_led` with the TEST item, I think it will work.

mellowmonk, 2019-11-29:
_Thanks, that worked._

**updated lines,**
_PageBuilder root("/led", { elm_gpio });_
and
_sendRedirect("/led");_

[image]

And updated the link in the _AUTOCONNECT_HOMEURI_ menu too in _AutoConnectDefs.h_

[image]

**Result**
[image]

mellowmonk, 2019-11-29:
> Is it an AutoConnectAux menu item? Or is it a menu item added by directly modifying AutoConnectPage.cpp?

I made the page by modifying the _AutoConnectPage.cpp_ file

Now, the _root_ page shows **404**
[image]

_To solve this:_
Would try to make another page in the **_HandlePortalEX.ino_** using AutoConnectAux so it doesnt show 404 whenever anyone enters the root ip address.
Lets hope that works.

Hieromon, 2019-12-02:
Preparing a page for uri `/` is one way that's right as you said. Related to it, I have two things that can advise you:

1. Override **AUTOCONNECT_HOMEURI** without modifying `AutoConnectDefs.h`.
You can override the **AUTOCONNECT_HOMEURI** in the sketch by using [AutoConnect::config](https://hieromon.github.io/AutoConnect/api.html#config) function with specifying [AutoConnectConfig::homeUri](https://hieromon.github.io/AutoConnect/apiconfig.html#homeuri).
https://hieromon.github.io/AutoConnect/advancedusage.html#assign-user-sketchs-home-path
```cpp
#define MY_HOME  "/led"  // an home uri declaration

AutoConnect Portal;
AutoConnectConfig Config;  // Declare a AutoConnectConfig to configure AutoConnect

PageElement elm_gpio(mold_page, {
...
});
PageBuilder root(MY_HOME, { elm_gpio });

String gpio(PageArgument& args) {
...
  sendRedirect(MY_HOME);  // Redirect to the path of home when after blinking the LED.
  return "";
}

void setup() {
  Config.homeUri = String(MY_HOME);  // Specify the home uri.
  Portal.config(Config);  // Override the home uri.
  Portal.begin();
}
```

2. Alternate a `/` page with an existing page.
If you need a dedicated page for `/`, this item is unnecessary. The following is a method for sharing the root page by redirecting the `/` access.  
Use the `sendRedirect` function of **HandlePortalEX.ino** for redirection.
```cpp
// Execute page redirection as a handle of "reroot" token declared with reRoot of PageBuilder.
// This is a trick using PageBuider.
String redirect (PageArgument& args) {
  sendRedirect("/led");  // Specify the destination of the redirection.
  return "";
}

// PageBuilder declaration for page redirection. This page is not actually generated.
PageElement elm_reRoot("{{reroot}}", { {"reroot", redirect} });
PageBuilder reRoot("/", elm_reRoot);

void setup() {
  portal.begin();
  reRoot.insert(server);  // Register the "/" page as the redirector.
}
```

Have fun.

mellowmonk, 2019-12-03:
Thank you Hieromon. 

Will experiment with it over the next few days🔢

closed 2019-12-03 by mellowmonk