Hieromon/AutoConnect #158: IOT standalone network

happytm, 2019-12-02:
This is not an issue but rather feedback. Thank you for creating such a painless wifi management and webserver library. I have just started digging deeper and it seems very powerful. I just implemented your library to my  IOT network code and it is very helpful to update remote devices using OTA web update feature of you library. My code is described as below:

I have been working on a simple very low power IOT application using uMQTT broker. It is a template for a standalone star network of few battery powered sensors which can be customized easily. There are 2 sketches ProbeSender and ProbeReceiver. ProbeReceiver is using uMQTT broker to connect to outside world and it is up all the time using mains power. All ProbeSendor devices are not directly connected to internet. I have implemented basic 2 way communication between ProbeReceiver and all ProbeSender device on the network. For issuing commands to ProbeSender devices on network I use android app called MQTT Dash. Only requirement for this code to work efficiently (less power consumption) is to have same access point channel as home's main access point across all devices. The link to code is below:

https://github.com/happytm/BatteryNode

Thanks again.

Hieromon, 2019-12-05:
@happytm Thank you for your interest in AutoConnect and for your feedback.
I was hoping that users make practical and effective use of AutoConnect like your project. Such feedback is very encouraging for library developers.

happytm, 2019-12-05:
Thank you.I know I scratched the surface with your code but I will be trying to use elements to create some kind of dashboard to display data from my  device next.

Thanks.

Hieromon, 2020-04-01:
Close once due to no activity record for one month.
Reopen when new reports are posted.

closed 2020-04-01 by Hieromon