Hieromon/AutoConnect #159: Disconnect Push Button

mellowmonk, 2019-12-05:
With a disconnect button available in the webserver, i wanted to add a disconnect button too.
Whenever the button is pushed, the esp32 should forget the saved SSID's.

**Logic:**

```
void loop(){
...
      if (buttonState == HIGH) {
          if (WiFi.waitForConnectResult() == WL_CONNECTED) {
                 WiFi.disconnect();               
          }
        Serial.println("BUTTON DISCONNECT");
...
}
```

However, It doesn't seem to work.

Hieromon, 2019-12-12:
Yes, `WiFi.disconnect` has the effect of clearing the recent SSID stored by ESP32.
Can you use the [AutoConnectConfig::autoReconnect](https://hieromon.github.io/AutoConnect/advancedusage.html#automatic-reconnect) option as a workaround this?

Hieromon, 2019-12-31:
Close once due to no activity record for one month.
Reopen when new reports are posted.

closed 2019-12-31 by Hieromon