Hieromon/AutoConnect #160: Feature request : Station_AP mode

happytm, 2019-12-10:
I require Station_AP wifi mode to use with ESP8266 in my code. Is it possible to activate this mode so ESP8266 device can have station to connect to internet and Access point with predefined APSSID so other devices on local network can connect to this APSSID. Right now I can set Station_AP mode successfully  without using AutoConnect library by using following functons:

void startWiFiClient()
{
  Serial.println("Connecting to " + (String)ssid);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, pass);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");

  Serial.println("WiFi connected");
  Serial.println("IP address: " + WiFi.localIP().toString());
}

void startWiFiAP()
{

  WiFi.softAP(apssid, "<notused>", apChannel, 0, 0); 
  Serial.println("AP started");
  Serial.println("IP address: " + WiFi.softAPIP().toString());
}

void setup()
{
  Serial.begin(115200);
  Serial.println();
  Serial.println();

 startWiFiClient();
  startWiFiAP();
}

The whole code is to big to post here but it is linked below:

https://github.com/happytm/BatteryNode/blob/master/ProbeReceiver.ino

Thanks.

Hieromon, 2019-12-12:
AutoConnect is in AP_STA mode when it is a captive portal.
Please focus specifically on the problems you have. I don't walk the through for your whole code and am not going to understand your whole code.

happytm, 2019-12-12:
Does that mean the device publishes it's own ssid and I can connect to that ssid from my laptop ? if so what would be the IP I have to use to access the GUI  ? Is it the usual 192.168.4.1 or some other IP ?

Thanks

happytm, 2019-12-12:
Also I wanted to have AP_STA mode all the time even if device is successfully connected to internet.

Thanks.

mitchsf, 2019-12-19:
I too would like to have the ability to access the web server through an open port at any time after credentials are saved and AP mode is closed. I think that would be a valuable addition to the library.

Hieromon, 2019-12-23:
@happytm 
>  if so what would be the IP I have to use to access the GUI ?

Please refer to the documentation. In AP_STA, AutoConnect menu IP is decided by [AutoConnectConfig::apip](https://hieromon.github.io/AutoConnect/apiconfig.html#apip), and A home path of the menu is /_ac by default (https://hieromon.github.io/AutoConnect/menu.html#where-the-from).

> Also I wanted to have AP_STA mode all the time even if device is successfully connected to internet.

[AutoConnectConfig::retainPortal](https://hieromon.github.io/AutoConnect/apiconfig.html#retainportal).

Hieromon, 2020-02-15:
Close once due to no activity record for one month.
Reopen when new reports are posted.

closed 2020-02-15 by Hieromon