Hieromon/AutoConnect #161: Problem connecting to AutoConnect GUI

happytm, 2019-12-11:
I am using your code in my larger home automation application. I tested your library alone and it worked fine but when I add it to my code it appears to be working fine and it prints IP address where I   
can connect to GUI but when I try to connect to it does not show any GUI. I tried for couple of days to solve the problem but can not find anything obvious. I have link to the code below:

https://github.com/happytm/BatteryNode/blob/master/ProbeSender.ino

Please help.
Thanks.

Hieromon, 2019-12-12:
At first glance, I think the sketch structure is strange. I can't afford to scrutinize the code you presented, but it is wrong that portal.handleClient is restricted when the following conditions.

https://github.com/happytm/BatteryNode/blob/acf77a9e847c49017c59b4a0e576b96032a47586/ProbeSender.ino#L465

Your code also has many #directives that are inappropriate for review by the third person like as me.
Please refer to [here](https://hieromon.github.io/AutoConnect/faq.html#4-reports-the-issue-to-autoconnect-github-repository) and reduce the code to reproduce the problem.

> The sketch code with formatted by the code block tag (Reduce to the reproducible minimum code for the problem)

Hieromon, 2020-02-15:
Close once due to no activity record for one month.
Reopen when new reports are posted.

closed 2020-02-15 by Hieromon