Hieromon/AutoConnect #166: Buttons are not appearing

gcormier, 2019-12-23:
I've got a custom parameter page defined. The elements are showing up, but there's no submit/cancel button.

json:
````
{
  "title": "Fumebuddy Settings",
  "uri": "/fumebuddy_setting",
  "menu": true,
  "element": [
    {
      "name": "style",
      "type": "ACStyle",
      "value": "label+input,label+select{position:sticky;left:120px;width:230px!important;box-sizing:border-box;}"
    },
    {
      "name": "caption",
      "type": "ACText",
      "value": "Enter URL's that can be called from this device",
      "style": "font-family:serif;color:#4682b4;"
    },
    {
      "name": "urlon",
      "type": "ACInput",
      "value": "",
      "placeholder": "on URL",
      "label": "on URL"
    },
		{
      "name": "urloff",
      "type": "ACInput",
      "value": "",
      "placeholder": "off URL",
      "label": "off URL"
    },
    {
      "name": "urltoggle",
      "type": "ACInput",
      "value": "",
      "placeholder": "toggle URL",
      "label": "toggle URL"
    },
    {
      "name": "save",
      "type": "ACSubmit",
      "value": "Save&Start",
      "uri": "/fumebuddy_save"
    },
    {
      "name": "discard",
      "type": "ACSubmit",
      "value": "Discard",
      "uri": "/"
    }
  ]
}
````

[image]

gcormier, 2019-12-23:
Hmm, not sure what is going on - it is loading those 3 from params.json, which is a bit odd.

adynis, 2019-12-23:
My impression is that the json pasted and the screenshot do not match:
In the json you have "label": "on URL" , but in the screenshot I see: "url on"; The same discrepancy for the others... 
Maybe you use something different for this page.

gcormier, 2019-12-23:
So yeah, I made some changes to param.json to confim this is what I'm seeing the values from.

param.json
````
[
  {
    "name": "urlon",
    "type": "ACInput",
    "value": "",
    "label": "url on"
  },
  {
    "name": "urloff",
    "type": "ACInput",
    "value": "",
    "label": "url off"
  },
  {
    "name": "urltoggle",
    "type": "ACInput",
    "value": "",
    "label": "url togglePARAM"
  }
] 
````

[image]

But I have no idea why that's happening. Here's my setup

````
// Setup AutoConfig
  config.bootUri = AC_ONBOOTURI_HOME;
  config.homeUri = "/";
  config.hostName = String("fumebuddy") + GET_CHIPID();
  portal.config(config);

  portal.join({fumebuddySettingsAux, fumebuddySavedAux});
  PageArgument args;
  loadParams(fumebuddySettingsAux, args);
  AutoConnectInput &e_on = fumebuddySettingsAux["urlon"].as<AutoConnectInput>();
  AutoConnectInput &e_off = fumebuddySettingsAux["urloff"].as<AutoConnectInput>();
  AutoConnectInput &e_toggle = fumebuddySettingsAux["urltoggle"].as<AutoConnectInput>();

  fumebuddyOn = e_on.value;
  fumebuddyOff = e_off.value;
  fumebuddyToggle = e_toggle.value;

  Serial.println("fumebuddyOn set to " + fumebuddyOn);
  Serial.println("fumebuddyOff set to " + fumebuddyOff);
  Serial.println("fumebuddyToggle set to " + fumebuddyToggle);
  Serial.println("hostname set to " + config.hostName);

  portal.on(AUX_SETTINGS, loadParams);
  portal.on(AUX_SAVED, saveParams);

  Serial.print("WiFi\n");
  if (portal.begin())
  {
    Serial.println("connected:" + WiFi.SSID());
    Serial.println("IP:" + WiFi.localIP().toString());
  }
  else
  {
    Serial.println("connection failed:" + String(WiFi.status()));
    while (1)
    {
      delay(100);
      yield();
    }
  }
````

adynis, 2019-12-23:
I'm slightly confused.
I see that you are using portal.join(<aux_pages>) , and not portal.load(<json>).
So I suppose you are not using the content defined by the JSON, and you are showing what you've constructed with fumebuddySettingsAux and fumebuddySavedAux.

Hieromon, 2019-12-23:
@gcormier Maybe, `&` symbol in ACSubmit `save` is the cause.
Please try it:
```json
    {
      "name": "save",
      "type": "ACSubmit",
      "value": "Save&amp;Start",
      "uri": "/fumebuddy_save"
    },
```

gcormier, 2019-12-24:
In the constructor of the AutoConnectAux, is that not loading from the json file based on the parameter I pass it?

gcormier, 2019-12-24:
I followed the example using the MQTT file system code and managed to cobble things back together!

closed 2019-12-24 by gcormier