Hieromon/AutoConnect #167: detaching ArduinoJson (when using custom web pages)

adynis, 2019-12-23:
Hello,
**Introduction:**
I was using JSON webpages, and I've seen that after a couple of days, my ESP32 was freezing ... analyzing, it seems that the ESP.getFreeHeap() (and getMinFreeHeap) were decreasing approx 20KB each 6h ... And I think/hope this is the issue. (_otherwise I still to digg more_)
Hoping that JSON pages were the ones slowly filling-up my HEAP , I changed to the other method of defining custom webpages.
I temporary removed ArduinoJson library to test if it is used.

**Problem(s):**
A.) I tried to UNDEF the AUTOCONNECT_USE_JSON:
in VSCode with Platformio, I've tried adding: build_flags  "-UAUTOCONNECT_USE_JSON" or "-U AUTOCONNECT_USE_JSON" but no luck: I got errors saying that AutoConnect is requiring ArduinoJson.h ....
B.) I commented (_despite the fact that I was hoping for a solution without changing the code from an imported library_) in AutoConnectDefs.h  the line: #define AUTOCONNECT_USE_JSON 
I got another error:
`.platformio/lib/AutoConnect_ID2678/src/AutoConnect.h:240:15: error: 'JsonVariant' has not been declared`
I had to comment out from: AutoConnectUpdate.h the lines:
`#ifndef AUTOCONNECT_USE_JSON
#define AUTOCONNECT_USE_JSON
#endif`
_(or AUTOCONNECT_USE_UPDATE from AutoConnectDefs.h)_
to get a successful build.

**Questions:**
1.) How to properly detach ArduinoJson from AutoConnect  ?
2.) If I do not comment AUTOCONNECT_USE_JSON, but I do Not use JSON for defining custom webpages, do I benefit from the memory usage reduction ?

Hieromon, 2019-12-23:
Due to the presence of `#define` code in the include file, the -U build flag will have no effect.
https://github.com/Hieromon/AutoConnect/blob/ac8687939f868267383adbb543943dce4b47e838/src/AutoConnectUpdate.h#L32-L36

> 1.) How to properly detach ArduinoJson from AutoConnect ?

If you don't use AutoConnectUpdate, you can detach ArduinoJson with (at least my intention is).
But that's the way you commented both AUTOCONNECT_USE_UPDATE and AUTOCONNECT_USE_JSON, as you mentioned. I don't like this method either, but I can't come up with any other efficient (compile-time determinable) method.

>  If I do not comment AUTOCONNECT_USE_JSON, but I do Not use JSON for defining custom webpages, do I benefit from the memory usage reduction ?

I think there is no much difference.
I think the primordial matter is that the heap continues to be consumed. Can you consider another cause other than ArduinoJson?

adynis, 2019-12-25:
I replaced the custom webpages from json to ... normal code; 
Iv'e tested a couple of days and ... the same: The heap memory decreases in time ... I'm still including ArduinoJson for other library .... so .. i need to digg more to identify what is the entity which consumes the heap ...

I will close this, is will ever have any idea to improve topic no. #1 (regarding detaching ArduinoJson) you will probably do it or create another ticket :)

closed 2019-12-25 by adynis

Hieromon, 2019-12-26:
I got it. AutoConnect makes heavy use of String classes with PageBuilder. I recently discovered a memory leak in PageBuilder and released a fix as v1.3.6. Please use the latest PageBuilder.
Also, some AutoConnectElements store element properties in std::vector. Since the memory release depends on the release of AutoConnectElement, heap consumption may continue to increase depending on the API sequence.
Please let me know if your verification reveals anything new.
Thank you for your contribution

adynis, 2019-12-26:
Very interesting to read your observation !!
Last night I did 2 things: 
Updated PageBuilder to v1.3.6 and commented out "ESP32Ping" library, which I was using, reading somewhere on internet that someone had a similar issue with "ping" .... I said .. let's try.
Now, after ~20hours, the chart is looking MUCH better ... but I want to investigate more to see if is decreasing on a lower rate, or it's stable; After that I need to check if it was a "Ping" issue, or the fix from you resolved it :)

[image]

adynis, 2019-12-28:
After more investigations I conclude:
Clearly my issue Heap memory leak was coming from the ESP32Ping module.
After I've commented out, the memory chart was almost perfectly horizontal, and after I included back, it again started to show the problem 
I didn't tested again with PageBuilder/v1.3.5 and no Ping, because the influence of the Ping module was so big, that it was obvious for me "who was the guilty"...

[image]

Hieromon, 2019-12-30:
Thanks for the detailed report. Your report is very useful.
AutoConnect was acquitted and I was relieved for now. Although, it is clear that PageBuilder 1.3.5 has a memory leak, so just use 1.3.6.

BTW, I rethink the directive that detaches ArduinoJson. As you pointed out, it's not a good idea for users to modify the original state of the library individually to detach ArduinoJson.
I try to fix that be able to detach it (also AutoConnectUpdate) by only one external definition.

Hieromon, 2019-12-30:
Detaching the ArduinoJson by `#define AUTOCONNECT_NOUSE_JSON`, it is done.
Specify `build_flags = -DAUTOCONNECT_NOUSE_JSON` of the platformio.ini file. Please try with the [patches/v113](https://github.com/Hieromon/AutoConnect/tree/patches/v113) branch.