Hieromon/AutoConnect #170: Can this be used manually?

steveheinsch, 2019-12-28:
My product has several webpages already. They are not required to use the product. The product works standalone, but can be enhanced via the webpages (alter parameters, etc).

So, I don't want to require the user to connect to wifi to use the device. They may never want/need to. What I'd like to do is manually put it in autoconnect mode so that it creates the AP using mDNS like at the press of a button, and then the user can go there in their browser to enter the credentials and once it can connect, switch back to STA and present my regular web pages.

Is it possible to do this with this library, or is it coded to require wifi credentials to work?

Hieromon, 2019-12-31:
Have you considered applying WiFiManager?
Also, I don't clearly understand what you want to do, but I hope you find the following comment helpful.
https://github.com/Hieromon/AutoConnect/issues/145#issuecomment-544916950

Hieromon, 2020-04-01:
Close once due to no activity record for one month.
Reopen when new reports are posted.

closed 2020-04-01 by Hieromon