Hieromon/AutoConnect #175: deleteAllCredentials() doesn't delete credentials ...

adynis, 2020-01-11:
Hi,
After many trials, I reduced my test-case to a small example:
-> ESP32-WROOM-32D
-> Visual Studio Code + PlatformIO

1.) I used the Flash Download Tools (https://www.espressif.com/en/support/download/overview) to erase the chip 
2.) I upload the sketch listed below
3.) I wait ~30 seconds and connect my mobile phone to: Test_AC
4.) In the captive portal, I configure a new AP, by entering the corresponding password for my router's WiFi.
5.) From my computer, I connect to the IP (IP is shown also on serial) -> http://192.168.1.189/_ac
6.) I enter the "Test_AC" menu option
7.) I press the button: "Action please!" . This will deleteCredentials and ESP.restart(). I watch the serial output.
8.) I see on Serial Deleting credentials... \n credentials.entries = 1 , **which is good**, it's what I was expecting.
9.) **PROBLEM**: immediately after point 8.), after the ESP restart ... **ESP is again connected**.... (I thought credentials are deleted) ... : "WiFi connected: 192.168.1.189"
10.) I try again. I reopen the page, go to Test_AC, I press again "Action please!"
11.) I see on serial: Deleting credentials... \n credentials.entries = 0 , and immediately after restart it's connected back to the same IP :(
12.) From this moment, no matter how many times I try, I get only: credentials.entries() == 0 , but the ESP remembers credentials and reconnects to the router's WiFi ...

The sketch:
```c
#include <Arduino.h>

#include <WiFi.h>      // Replace with WiFi.h for ESP32
#include <WebServer.h> // Replace with WebServer.h for ESP32
#include <AutoConnect.h>

WebServer webServer; // Replace with WebServer for ESP32
AutoConnect portal(webServer);
AutoConnectConfig acConfig("Test_AC", ""); //no password

ACText(mp_header, "Test_AC", "color:red;font-family:verdana;font-size:300%;");
ACSubmit(action, "Action please !", "/action");
AutoConnectAux mainPage("/", "Test_AC", true, {mp_header, action});

void deleteAllCredentials(void) {
    AutoConnectCredential credential;
    station_config_t config;
    uint8_t ent = credential.entries();

    
    Serial.printf("credentials.entries = %d\n", ent);

    while (ent--) {
        credential.load((int8_t)0, &config);
        credential.del((const char*)&config.ssid[0]);
    }
}

void onAction()
{
    Serial.println("Action button was pressed");

    Serial.println("Deleting credentials...:");
    deleteAllCredentials();

    char content[] = "Action button was pressed; ESP32 is restarting...";
    webServer.send(200, "text/plain", content);

    ESP.restart();
}

void setup()
{
    // put your setup code here, to run once:

    Serial.begin(115200);
    Serial.println("Started ....");

    acConfig.title = "My menu";
    acConfig.autoReconnect = true; //Tries to connect to any of the saved SSIDs

    portal.config(acConfig);

    portal.join({mainPage});

    webServer.on("/action", onAction);

    if (portal.begin())
    {
        Serial.println("WiFi connected: " + WiFi.localIP().toString());
    }
}

void loop()
{
    // put your main code here, to run repeatedly:

    portal.handleClient();

}
```

Serial output:

```
--- Miniterm on COM8  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
Started ....
[AC] SoftAP configure 172.217.28.1, 172.217.28.1, 255.255.255.0 
[AC] / on hands
[AC] Current:
[E][Preferences.cpp:38] begin(): nvs_open failed: NOT_FOUND
[AC] Preferences begin failed to import AC_CREDT
[AC] WiFi.config(IP=0.0.0.0, Gateway=0.0.0.0, Subnetmask=0.0.0.0, DNS1=0.0.0.0, DNS2=0.0.0.0)
[E][WiFiSTA.cpp:219] begin(): connect failed!
[AC] WiFi.begin()
[AC] Connecting.....................................................................................................time out IP:0.0.0.0
[E][Preferences.cpp:38] begin(): nvs_open failed: NOT_FOUND
[AC] Preferences begin failed to import AC_CREDT
[AC] SoftAP configure 172.217.28.1, 172.217.28.1, 255.255.255.0 
[AC] SoftAP Test_AC/ Ch(1) IP:172.217.28.1 
[AC] http server started
[AC] DNS server started
dhcps: send_offer>>udp_sendto result 0
[AC] Host:,URI:/chat,ignored
[E][WebServer.cpp:617] _handleRequest(): request handler not found
[AC] Detected application, c.whatsapp.net, 0.0.0.0
[AC] Host:c.whatsapp.net,URI:/generate_204,ignored
[E][WebServer.cpp:617] _handleRequest(): request handler not found
[AC] Detected application, connectivitycheck.gstatic.com, 0.0.0.0 
[AC] Host:connectivitycheck.gstatic.com,URI:/generate_204,ignored
[E][WebServer.cpp:617] _handleRequest(): request handler not found
[AC] Detected application, connectivitycheck.gstatic.com, 0.0.0.0
[AC] Host:connectivitycheck.gstatic.com,URI:/_ac,generated:/_ac, allocated
[AC] Host:172.217.28.1,URI:/_ac,already allocated
[AC] Host:172.217.28.1,URI:/favicon.ico,ignored
[E][WebServer.cpp:617] _handleRequest(): request handler not found
[AC] Host:172.217.28.1,URI:/generate_204,ignored
[E][WebServer.cpp:617] _handleRequest(): request handler not found
[AC] Detected application, connectivitycheck.gstatic.com, 0.0.0.0
[AC] Host:connectivitycheck.gstatic.com,URI:/generate_204,ignored
[E][WebServer.cpp:617] _handleRequest(): request handler not found
[AC] Detected application, connectivitycheck.gstatic.com, 0.0.0.0
[AC] Host:connectivitycheck.gstatic.com,URI:/_ac/config,generated:/_ac/config, allocated
[AC] Host:172.217.28.1,URI:/_ac/config,already allocated
[AC] 19 network(s) found

[AC] Host:172.217.28.1,URI:/generate_204,ignored
[E][WebServer.cpp:617] _handleRequest(): request handler not found
[AC] Detected application, connectivitycheck.gstatic.com, 0.0.0.0
[AC] Host:connectivitycheck.gstatic.com,URI:/generate_204,ignored
[E][WebServer.cpp:617] _handleRequest(): request handler not found
[AC] Detected application, connectivitycheck.gstatic.com, 0.0.0.0
[AC] Host:connectivitycheck.gstatic.com,URI:/_ac/connect,generated:/_ac/connect, allocated
[AC] Host:172.217.28.1,URI:/_ac/connect,already allocated
[AC] Queried SSID:SuperNet
[AC] WiFi.config(IP=0.0.0.0, Gateway=0.0.0.0, Subnetmask=0.0.0.0, DNS1=0.0.0.0, DNS2=0.0.0.0)
[AC] Attempt:SuperNet Ch(10)
[AC] Connecting......established IP:192.168.1.189
[E][Preferences.cpp:38] begin(): nvs_open failed: NOT_FOUND
[AC] Preferences begin failed to import AC_CREDT
[AC] SuperNet credential saved
[AC] Event<17> handler registered
[AC] DNS server stopped
WiFi connected: 192.168.1.189
[AC] Host:172.217.28.1,URI:/generate_204,ignored
[E][WebServer.cpp:617] _handleRequest(): request handler not found
[AC] Detected application, connectivitycheck.gstatic.com, 192.168.1.189
[AC] Host:connectivitycheck.gstatic.com,URI:/_ac,generated:/_ac, allocated
[AC] Host:192.168.1.189,URI:/_ac,already allocated
[AC] Host:192.168.1.189,URI:/_ac,already allocated
[AC] Host:192.168.1.189,URI:/_ac,already allocated
[AC] Host:192.168.1.189,URI:/,generated:/, allocated
[AC] Host:192.168.1.189,URI:/,already allocated
Action button was pressed
Deleting credentials...:
credentials.entries = 1
Started ....
[AC] SoftAP configure 172.217.28.1, 172.217.28.1, 255.255.255.0 
[AC] / on hands
[AC] Current:SuperNet
[AC] WiFi.config(IP=0.0.0.0, Gateway=0.0.0.0, Subnetmask=0.0.0.0, DNS1=0.0.0.0, DNS2=0.0.0.0)
[AC] WiFi.begin()
[AC] Connecting.established IP:192.168.1.189
[AC] http server started
WiFi connected: 192.168.1.189
[AC] Host:,URI:/,generated:/, allocated
[AC] Host:192.168.1.189,URI:/,already allocated
[AC] Host:192.168.1.189,URI:/,already allocated
[AC] Host:192.168.1.189,URI:/,already allocated
Action button was pressed
Deleting credentials...:
credentials.entries = 0
Started ....
[AC] SoftAP configure 172.217.28.1, 172.217.28.1, 255.255.255.0 
[AC] / on hands
[AC] Current:SuperNet
[AC] WiFi.config(IP=0.0.0.0, Gateway=0.0.0.0, Subnetmask=0.0.0.0, DNS1=0.0.0.0, DNS2=0.0.0.0)
[AC] WiFi.begin()
[AC] Connecting.established IP:192.168.1.189
[AC] http server started
WiFi connected: 192.168.1.189
```

It might be related to: https://github.com/Hieromon/AutoConnect/issues/154 or https://github.com/Hieromon/AutoConnect/issues/133

Hieromon, 2020-01-14:
> PROBLEM: immediately after point 8.), after the ESP restart ... ESP is again connected.... (I thought credentials are deleted) ... : "WiFi connected: 192.168.1.189"

The AutoConnectCredential removal logic in your sketch looks fine. The problem your pointed is due to the behavior of WiFi.begin().
The ESP32 arduino core saves the SSID and password established just before to the flash and reuses it with WiFi.begin() no parameters accompanied. Its store is data independent from AutoConnectCredential and cannot be manipulated with the AutoConnectCredential API. (It is same as ESP8266 arduino core)
If you want to delete both the AutoConnect saved credentials and the ESP32 arduino saved configuration data in the `onAction` function (that is, clear all WiFi connection configurations), You need to call `WiFi.disconnect` in addition to calling `AutoConnectCredentail::del`.
https://github.com/Hieromon/AutoConnect/issues/59#issuecomment-485796741

```cpp
void onAction()
{
    Serial.println("Action button was pressed");

    Serial.println("Deleting credentials...:");
    deleteAllCredentials();

    char content[] = "Action button was pressed; ESP32 is restarting...";
    webServer.send(200, "text/plain", content);

    WiFi.disconnect();    // Clear WiFi connection, not AutoConnectCredential
    ESP.restart();
}
```

adynis, 2020-01-14:
Hi Hieromon,
Thanks a lot for the explanation. It helped a lot !! I understood now the behavior I had. 
There was only one change I had to make in order to properly work:

Add "true" as second parameter for WiFi.disconnect(....):
```c
// Clear WiFi connection, not AutoConnectCredential
    WiFi.disconnect(false, true); //WiFi.disconnect(bool wifioff, bool eraseap)
```

So for anyone trying, here is **my full working test-example**:

```c
#include <Arduino.h>

#include <WiFi.h>      // Replace with WiFi.h for ESP32
#include <WebServer.h> // Replace with WebServer.h for ESP32
#include <AutoConnect.h>

WebServer webServer; // Replace with WebServer for ESP32
AutoConnect portal(webServer);
AutoConnectConfig acConfig("Test_AC", ""); //no pasword

ACText(mp_header, "Test_AC", "color:red;font-family:verdana;font-size:300%;");
ACSubmit(action, "Action please !", "/action");
AutoConnectAux mainPage("/", "Test_AC", true, {mp_header, action});

void deleteAllCredentials(void) {
    AutoConnectCredential credential;
    station_config_t config;
    uint8_t ent = credential.entries();

    
    Serial.printf("credentials.entries = %d\n", ent);

    while (ent--) {
        credential.load((int8_t)0, &config);
        credential.del((const char*)&config.ssid[0]);
    }
}

void onAction()
{
    Serial.println("Action button was pressed");

    Serial.println("Deleting credentials...:");
    deleteAllCredentials();

    char content[] = "Action button was pressed; ESP32 is restarting...";
    webServer.send(200, "text/plain", content);

    // Clear WiFi connection, not AutoConnectCredential
    WiFi.disconnect(false, true); //WiFi.disconnect(bool wifioff, bool eraseap)

    ESP.restart();
}

void setup()
{
    // put your setup code here, to run once:

    Serial.begin(115200);
    Serial.println("Started ....");

    acConfig.title = "My menu";
    acConfig.autoReconnect = true; //Tries to connect to any of the saved SSIDs

    portal.config(acConfig);

    portal.join({mainPage});

    webServer.on("/action", onAction);

    if (portal.begin())
    {
        Serial.println("WiFi connected: " + WiFi.localIP().toString());
    }
}

void loop()
{
    // put your main code here, to run repeatedly:

    portal.handleClient();

}
```

closed 2020-01-14 by adynis