Hieromon/AutoConnect #177: Support for non-ASCII chars in the webpages

JanHBade, 2020-01-19:
Hello,

great work, nice lib!

What about german üöäß? an other non-ascii chars?
[image]

Hieromon, 2020-01-20:
The character set for AutoConnectAux pages is not ASCII. HTML charset of the pages has already specified to UTF-8.
https://github.com/Hieromon/AutoConnect/blob/c70275d5f48749782b9b7a21517a5e5187432673/src/AutoConnectPage.cpp#L517

And, the default fonts assign to `Helvetica,Arial,sans-serif` for safety.
https://github.com/Hieromon/AutoConnect/blob/c70275d5f48749782b9b7a21517a5e5187432673/src/AutoConnectPage.cpp#L27

We can use AutoConnectStyle to specify the font for a particular element with CSS selector by the name.
However, AutoConnectAux does not activate the decoding with [Unicode in ArduinoJson](https://arduinojson.org/v6/api/config/decode_unicode/) for parsing JSON of the custom Web pages. As a result, the page content will probably be broken with the character as `??` etc.
There is no plan to support it now, as it may not always work properly.

JanHBade, 2020-01-22:
`ACSubmit(save, "Übernehmen", AUX_SETTING_SAVE);
`
this code in a custom web page generates the error above, so there is no json there... or use the lib json to generate the page?

have i another chance to display it correct?

Hieromon, 2020-01-23:
In the first place, does the ESP8266 / ESP32 arduino-core's String class, WebServer class support multibyte languages? I don't think so.
Below is an example of the simplest web server. It breaks the content even for the simple page.
Even if the page can be displayed correctly, the Arduino core WString class does not support multi-byte (it handles it with a byte offset instead of character offset), so if it involves complicated content processing like the AutoConnect, it does not work correctly unless enumerates the byte string, I think.

```cpp
#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>

#define WIFI_SSID     "YOURSSID"
#define WIFI_PASSWORD "PASSWORD"

ESP8266WebServer  server(80);

void setup() {
  delay(1000);
  Serial.begin(115200);
  Serial.println();
  WiFi.mode(WIFI_STA);
  delay(100);
  WiFi.begin(WIFI_SSID, WIFI_PASSWORD);
  if (WiFi.waitForConnectResult(30000) != WL_CONNECTED) {
    Serial.println("Connect failed");
    return;
  }
  server.on("/", [](){
    server.send(200, "text/html", 
    "<html lang=\"de\">"
    "<head><meta charset=\"ISO-8859-1\"/></head>"
    "<body>Übernehmen</body>"
    "</body>"
    "</html>"
    );
  });
  server.begin();
  Serial.print("Server started:");
  Serial.println(WiFi.localIP().toString());
}

void loop() {
  server.handleClient();
}
```
It cannot handle the Unicode literal.
```cpp
"<body>Übernehmen</body>"
```

> `ACSubmit(save, "Übernehmen", AUX_SETTING_SAVE);`
> have i another chance to display it correct?

Point directly in Unicode.
```cpp
ACText(text1, "&#x304a;&#x306f;&#x3088;&#x3046;");
ACSubmit(sub1, "&#x4e86;&#x89e3;", "/");
AutoConnectAux page("/", "Hello", true, { text1, sub1 });
```
[image]

JanHBade, 2020-01-24:
thank you for the infos

closed 2020-01-24 by JanHBade