Hieromon/AutoConnect #179: Portal.begin() often fails with valid credentials

klkl140, 2020-02-02:
Hello

Thanks for this nice project. It solves one of my problems.
After playing around for some days I localized a problem which is a showstopper for me (and a not so nice workaround).
I have a device which should automatically reconnect to a given wifi (entered once in AutoConnect) and show an AP when the wifi connection could not establish. Powerup could happen in normal work when no one is there to do anything. The given wifi might be unreliable, but when its available it should be used.

In some of my wifis the powerup reconnection works reliable, but sometimes (related to the used wifi) it fails repeatable with the valid credential. The AP is opened. After AP timed out, it returns with false. 

I solved this problem by reading the save credentials and call WiFi.begin() directly. This is not as nice as it can be, because the AP is opened and has to time out, but thats ok in my case.

  if(!Portal.begin()){
    AutoConnectCredential  ac(0);
    station_config_t  entry;
    mySerial.print("try to connect directly, ");
    if(ac.entries()){
      uint8_t idx = 0;
      ac.load(idx, &entry); // get first setting
      mySerial.print("WiFi.begin(" + String((char *)entry.ssid) + ")");
      WiFi.begin((char*)entry.ssid, (char*)entry.password);
    } else {
      mySerial.print("no predefined settings");
    }
  }

This is working reliable , so the problem seems to be in Portal.begin().
Why does it fail to establish the valid wifi connection when WiFi.begin() is able to establish the wifi?
Could I do something in advance? Waiting at PowerUp does not help.
Could I increase the wifi connection tries?

Thanks in advance....
Greetings Klaus

Hieromon, 2020-02-05:
I think the cause is the logic sequence for WiFi.disconnect or WiFi.mode(WIFI_STA) preceding WiFi.begin which is inside AutoConnect::begin.
https://github.com/Hieromon/AutoConnect/blob/c70275d5f48749782b9b7a21517a5e5187432673/src/AutoConnect.cpp#L98-L124
Which module do you use? I will try to reproduce the case.

klkl140, 2020-02-12:
Thanks for your fast answer.
Im not an Arduino specialist, but ill do my best....

Im on an Heltec Wifi/Lora 32 using DHCP

How can I activate your debug output?

```
#define AC_DEBUG
#define AC_DEBUG_PORT mySerial
#include "AutoConnect.h"
```` 
does not help

What is this line for?
```
if (!_configSTA(_apConfig.staip, _apConfig.staGateway, _apConfig.staNetmask, _apConfig.dns1, _apConfig.dns2)) 
   return false; 
```
Does it fail when I use DHCP?

klkl140, 2020-02-12:
a short test not using DHCP seems to be much more stable. I could not reproduce a failure once.

Hieromon, 2020-02-13:
> How can I activate your debug output?

If you are using the Arduino IDE to build, you will need to directly rewrite AutoConnectDefs.h to output other than **Serial**.

https://github.com/Hieromon/AutoConnect/blob/c70275d5f48749782b9b7a21517a5e5187432673/src/AutoConnectDefs.h#L13-L19

If it is not Arduino IDE such as PlatformIO, Sloeber or Arduino CLI, rewriting is not required and it can be controlled by build options at compile time.
(e.g. as using PlatformIO with platformio.ini)
```ini
build_flags =
  -DAC_DEBUG
  -DAC_DEBUG_PORT=mySerial
```
> What is this line for?
> ```cpp
> if (!_configSTA(_apConfig.staip, _apConfig.staGateway, _apConfig.staNetmask, _apConfig.dns1, _apConfig.dns2)) 
>  return false; 
> ```
> Does it fail when I use DHCP?

No, it's a legitimate procedure.
I'm concerned about your presented test code as above. (it is a part of your sketch, right?) When [AutoConnect::begin](https://hieromon.github.io/AutoConnect/lsbegin.html) returned with the false, the default behavior is that the module remains AP_STA. Also, since the IP address of the STA is (unset), it will be the IP address leased from the connected access point.
If this works, the STA mode DHCP procedure is correct. But you had mentioned.

> sometimes (related to the used wifi) it fails repeatable with the valid credential.

AutoConnect does not rely on the SSID string of stored credentials. Whether to try Wifi.begin to the detected AP depends on whether the BSSID matches. So, Is there something in your environment that BSSID can cause this problem?

klkl140, 2020-02-14:
Yes, the test code is part of my sketch. Im on Arduino.

> AutoConnect does not rely on the SSID string of stored credentials. Whether to try Wifi.begin to the detected AP depends on whether the BSSID matches. So, Is there something in your environment that BSSID can cause this problem?

BSSID is new for me and I researched a bit without understanding too much...
How can I get the necessary information about BSSID?
It seems possible to me that there is some old wifi information in the ESP (maybe the BSSID) which is used by Portal.begin() (and not using WiFi.begin()) that im not able to get rid of.....

I have three places, all connected to the internet using an additional wifi-router. The workplace (which is working most reliable) has an apple timecapsule, the others are a Fritzbox and a Devolo. All these boxes are connected by wire to my router and use the same 192.168.2.x. Using all these Wifis which my laptop works without problems.

The problem: At powerup the ESP boots in the captive portal all the time. Using reset to boot the ESP, it finds the right Wifi more often, maybe 30%.
Even when I delete all credentials and create one new valid credential, this behavior seems unchanged.
Not using DHCP is working much better, the connection is build up all the time without falling in the captive portal, but then the following mqtt connection is not build up..(?)

At the moment im using autoRise = true and retainPortal = false, Ive tried many different ways without finding a solution

I will make some more test to get a more validated error description. If I can do something to make it clearer for you I will be happy do do it :-)

klkl140, 2020-02-15:
Here a logfile where the autoconnect fails with a valid credential (klStall)
and the WiFi.begin() working soon afterwards.

```
[AC] SoftAP configure 172.217.28.1, 172.217.28.1, 255.255.255.0 
Portal begin
[AC] Current:klStall
[AC] WiFi.config(IP=0.0.0.0, Gateway=0.0.0.0, Subnetmask=0.0.0.0, DNS1=0.0.0.0, DNS2=0.0.0.0)
[AC] WiFi.begin()
[AC] Connecting.....................................................................................................time out IP:0.0.0.0
[AC] SoftAP configure 172.217.28.1, 172.217.28.1, 255.255.255.0 
[AC] SoftAP UNET_Wifi_246F2896DB28_AP/ Ch(3) IP:172.217.28.1 
Captive Portal started, IP:0.0.0.0
[AC] http server started
[AC] DNS server started
[AC] CP timeout exceeded:60001
[AC] Event<17> handler released
[AC] Portal stopped
Portal failed
try to connect directly, WiFi.begin(klStall)
```

'[AC] Current' is showing the last successful connection, even if it has been selected by using the autoconnect website in advance. This is working as expected.
But WiFi.begin() fails

Here the successful connection to the reliable working SSID klTC
```
[AC] SoftAP configure 172.217.28.1, 172.217.28.1, 255.255.255.0 
Portal begin
[AC] Current:klTC
[AC] WiFi.config(IP=0.0.0.0, Gateway=0.0.0.0, Subnetmask=0.0.0.0, DNS1=0.0.0.0, DNS2=0.0.0.0)
[AC] WiFi.begin()
[AC] Connecting.established IP:192.168.2.199

```

Where can I search ?

Hieromon, 2020-02-15:
Your posted log suggests that the default configuration has been changed by AutoConnectConfig.
For me to conduct an accurate survey, you may provide accurate and sufficient information.
I need all the information I mentioned in the documentation below:
https://hieromon.github.io/AutoConnect/faq.html#4-reports-the-issue-to-autoconnect-github-repository
Also, reduce the sketch to the minimum reproducible code for the problem.

Hieromon, 2020-04-01:
Close once due to no activity record for one month.
Reopen when new reports are posted.

closed 2020-04-01 by Hieromon

ark02, 2020-07-20:
Hi There, 

I think I'm having a similar issue on ESP32. 
Credentials are stored successfully and I can view them via "Open SSIDs" 
On most reboots, power cycle, or firmware flash Autoconnect does not automatically connect to the available WiFi network. It times out and then goes to SoftAP mode. Must be re-connected via web portal every time. 

- Hardware module
Espressif ESP32-DevkitC-32D 
Using ESP32-WROOM-32D module
- Arduino core version Including the upstream commit ID if necessary
1.0.4
- Operating System which you use
Windows 10, programming via platformio
- Your smartphone OS and version (Especially for Android)
Android 9.0
- Your AP information (IP, channel) if related
- lwIP variant
- Problem description
- If you have a STACK DUMP decoded result with formatted by the code block tag
- the Sketch code with formatted by the code block tag (Reduce to the reproducible minimum code for the problem)
`
static WebServer Server;          
static AutoConnect      Portal(Server);
static AutoConnectConfig acConfig;

void rootPage() {
  char content[] = "Hello, world";
  Server.send(200, "text/plain", content);
}

void WIFI_Setup() {

  Serial.println();
  
  acConfig.apid = "AP NAME";
  acConfig.psk = "AP_PASS";
  Portal.config(acConfig);

  Server.on("/", rootPage);
  if (Portal.begin()) {
    Serial.println("WiFi connected: " + WiFi.localIP().toString());
  } else {
    Serial.println("WiFI Connect failed!");
  }
  WiFi.onEvent(WiFi_events);
}

void Wifi_client() {
    Portal.handleClient();
}

void setup() {
  
  Serial.println("Booting");
  Serial.println("Connecting to WIFI:");
  WIFI_Setup();  
}

void loop() {
  Wifi_client();
}
`
- Debug messages output (Including arduino core)
Connecting to WIFI:

[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 14 - AP_STOP
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 3 - STA_STOP
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 3 - STA_STOP
[AC] SoftAP configure 172.217.28.1, 172.217.28.1, 255.255.255.0 
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 14 - AP_STOP
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 3 - STA_STOP
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
[AC] Current:
[AC] WiFi.config(IP=0.0.0.0, Gateway=0.0.0.0, Subnetmask=0.0.0.0, DNS1=0.0.0.0, DNS2=0.0.0.0)
[E][WiFiSTA.cpp:219] begin(): connect failed!
[AC] WiFi.begin()
[AC] Connecting.....................................................................................................time out IP:0.0.0.0
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 15 - AP_STACONNECTED
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 14 - AP_STOP
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 14 - AP_STOP
[AC] SoftAP configure 172.217.28.1, 172.217.28.1, 255.255.255.0 
[AC] SoftAP Hi! Im Mona/washless Ch(1) IP:172.217.28.1 
[AC] http server started
[AC] DNS server started

klkl140, 2020-07-27:
Hello

I have this problem sometimes, but most of the time it is working as it should.
If the problem occurs I have to reset the device. The second connection does always work.

Greetings Klaus