Hieromon/AutoConnect #183: Custom Parameters - To Preferences instead of SPIFFS?

mriksman, 2020-02-23:
Hi,
Can custom parameters be stored in Preferences/EEPROM instead of SPIFFS?

Hieromon, 2020-02-24:
It is not clear what you mean by custom parameters, but the functions included in the AutoConnect library do not support the data IO with EEPROM / Preferences.
If you want to store the input/output parameter of the custom web pages in EEPROM / Preferences, use the EEPROM / Preferences class directly to handle it by your sketches on its own. All you have to do is get the data entered from the custom web page in the AutoConnectAux's "on" handler, save it to the Preferreds with a specific key.
https://hieromon.github.io/AutoConnect/achandling.html#where-to-pick-up-the-values

mriksman, 2020-02-24:
Thanks; yeah that's what I am thinking. Your `mqttRSSI` uses the JSON file to save the `value`, and it appears the `mqttRSSI_NA` doesn't actually save the parameters at all?

Hieromon, 2020-02-25:
The `saveParms` function combined in the example `mqttRSSI_NA` is an `on` handler for the custom page `/mqtt_save` but does not save any parameters contrary to its name. It is provided to show how to handle custom web pages without using JSON.
As you would expect, you can improve the `saveParms` based on `mqttRSSI_NA` to be able to save parameters to EEPROM / Preferences.

For ESP32, the following partial sketch code shows the saving **Preferences** part added to the `saveParams` and the `loadParams` function.
**Note: The validity of the logic has not verified. It shows only the mindset.**

1. An include directive and `Preferences` declaration

```cpp
#include <Preferences.h>
Preferences prefs;

// The data structure of the persisted parameters definition
typedef struct {
  char server[32];
  char channelId[16];
  char userKey[16];
  char apiKey[16];
  int  period;
  bool unique;
} params_t;
```

2. Adds `loadParams` to load parameters from Preferences

```cpp
String loadParams(AutoConnectAux& aux, PageArgument& args) {
  params_t  params;

  if (prefs.begin("params")) {
    if (prefs.getBytes("params", &params, sizeof(params_t))) {
      prefs.end();
      mqttserver.value = params.server;
      channelid.value = params.channelId;
      userkey.value = params.userKey;
      apikey.value = params.apiKey;
      period.check(String(params.period));
      uniqueid.checked = params.unique;
    }
    else
      Serial.println("No saved Prefs");
  }
  else
    Serial.println("Failed Prefs open to load");
  return String("");
}
```

3. Modify `saveParams` and adds `savePrefs`

```cpp
// Add savePrefs function as the below
void savePrefs(const String serverId, const String channelId, const String userKey, const String apiKey, const int period, const bool useUnique) {
  params_t  params;

  if (!prefs.begin("params"))
    Serial.println("Prefs not initialized, allocate a new one");
  memset(&params, '\0', sizeof(params_t));
  strcpy(params.server, serverId.c_str());
  strcpy(params.channelId, channelId.c_str());
  strcpy(params.userKey, userKey.c_str());
  strcpy(params.apiKey, apiKey.c_str());
  params.period = period;
  params.unique = useUnique;
  prefs.putBytes("params", &params, sizeof(params_t));
  prefs.end();
}

// saveParams modification
String saveParams(AutoConnectAux& aux, PageArgument& args) {
  ...
  parameters.value = echo;
  // Save to Preferences
  savePrefs(mqttserver.value, channelid.value, userkey.value, apikey.value, updateInterval, uniqueid.checked);  // <-- Add
  return String("");
}
```

4. `setup()` modification

  ```cpp
void setup() {
  ...
  // Join the custom Web pages and register /mqtt_save handler
  portal.join({ mqtt_setting, mqtt_save });
  portal.on(AUX_SAVE_URI, saveParams);
  portal.on(AUX_SETTING_URI, loadParams);  // <-- Add an on handler to load parameters from Preferences
  ...
}
```

Hieromon, 2020-04-01:
Close once due to no activity record for one month.
Reopen when new reports are posted.

closed 2020-04-01 by Hieromon

haldi4803, 2022-02-26:
So....
This is 2 years old.
Do we have a never/other possibility to prestistently safe other settings? Maybe an working Example For ESP32?

I've modified the elements example to my needs, but then was kinda astonished that settings are not safed after unplugging the device. Thought that was included in the filesystem.

`/*
  Elements.ino, Example for the AutoConnect library.
  Copyright (c) 2020, Hieromon Ikasamo
  https://github.com/Hieromon/AutoConnect
  This software is released under the MIT License.
  https://opensource.org/licenses/MIT

  This example demonstrates the typical behavior of AutoConnectElement.
  It also represents a basic structural frame for saving and reusing
  values ​​entered in a custom web page into flash.
*/

// To properly include the suitable header files to the target platform.

#include <WiFi.h>
#include <WebServer.h>
#include <FS.h>
#include <SPIFFS.h>
using WebServerClass = WebServer;
fs::SPIFFSFS& FlashFS = SPIFFS;
#define FORMAT_ON_FAIL  true

#include <AutoConnect.h>

#define PARAM_FILE      "/elements.json"

static const char PAGE_ELEMENTS[] PROGMEM = R"(
{
  "uri": "/settings",
  "title": "Settings",
  "menu": true,
  "element": [
    {
      "name": "tablecss",
      "type": "ACStyle",
      "value": "table{font-family:arial,sans-serif;border-collapse:collapse;width:100%;color:black;}td,th{border:1px solid #dddddd;text-align:center;padding:8px;}tr:nth-child(even){background-color:#dddddd;}"
    },
    {
      "name": "text",
      "type": "ACText",
      "value": "Open Wheater API Settings",
      "style": "font-family:Arial;font-size:18px;font-weight:400;color:#191970"
    },
    {
      "name": "apikey",
      "type": "ACInput",
      "label": "API Key",
      "placeholder": "See: https://openweathermap.org"
    },
    {
      "name": "hr",
      "type": "ACElement",
      "value": "<hr style=\"height:1px;border-width:0;color:gray;background-color:#52a6ed\">",
      "posterior": "par"
    },
    {
      "name": "city",
      "type": "ACInput",
      "label": "City",
      "value": ""
    },
    {
      "name": "latitude",
      "type": "ACInput",
[... 170 lines cut ...]
      param.close();
      // Read the saved elements again to display.
      param = FlashFS.open(PARAM_FILE, "r");
      aux["echo"].value = param.readString();
      param.close();
    }
    else {
      aux["echo"].value = "Filesystem failed to open.";
    }
    return String();
  });

  portal.join({ elementsAux, saveAux });
  config.auth = AC_AUTH_NONE;
  config.ticker = true;
  config.autoReconnect = true;
  config.reconnectInterval = 1;
  portal.config(config);
  portal.begin();
}

void loop() {
  portal.handleClient();
}
`