Hieromon/AutoConnect #185: How do i pass a json in response?

shaikhfk, 2020-02-26:
Hello Hieromon,

I am using this [HandleClient example](https://github.com/Hieromon/AutoConnect/blob/master/examples/HandlePortal/HandlePortal.ino), Using the `http://ipaddress/io?v=low` am able to pass data from browser to ESP32 controller, now what i want is when i use the `http://ipaddress/io?v=getdata` the controller should send me data in JSON format Example:
```
{
  "userId": 1,
  "id": 1,
  "title": "delectus aut autem",
  "completed": false,
"Pin1": "High",
"Pin2": "Low"
}
```
Here `Pin 1&2 ` will read the `digitalRead(Pin1) ` data (high or low) and will send the data in JSON response.
Only when `http://ipaddress/io?v=getdata` is called controller should send JSON data rest other URL should always show HTML only as you designed.

Hieromon, 2020-02-27:
To do what you want, you need to know how to use the ESP32WebServer library, instead of AutoConnect.
The `/io` handler (it's for the WebServer class of the esp32 arduino-core, not for AutoConnect) To receive the query string `v=getdata` and respond with JSON, see the following example in the ESP32 arduino-core.
https://github.com/espressif/arduino-esp32/blob/13e02063d63146096952fc6a6ef0bc3ced5b1203/libraries/WebServer/examples/FSBrowser/FSBrowser.ino#L286-L295

shaikhfk, 2020-02-27:
Sir I tried multiple things its just when i get json data the autoconnect code gets stops i get the error Host not found so there a way to integrate [HandleClient](https://github.com/Hieromon/AutoConnect/blob/master/examples/HandlePortal/HandlePortal.ino) to send JSON response for specific URL only? AutoConnect Library should work as it is only for specific URL i want JSON response.

Hieromon, 2020-02-27:
> AutoConnect Library should work as it is only for specific URL i want JSON response.

I can not understand your intent. It's not AutoConnect that responds to certain URLs. It is responded by the user sketch code registered with [`WebServer::on`](https://github.com/espressif/arduino-esp32/blob/13e02063d63146096952fc6a6ef0bc3ced5b1203/libraries/WebServer/src/WebServer.h#L89) function of ESP32 arduino-core. When HTTP GET `/io` occurs, HandleClient.ino redirects to `/` with the handleGPIO function, the handleRoot as root page displays LED icons, all of which are webserver library class actions. Not AutoConnect.

shaikhfk, 2020-02-27:
Sir my intent is to get JSON response from ESP32 on specific URL without affecting autoConnect library so could you help me in integrating JSON response with [HandleClient](https://github.com/Hieromon/AutoConnect/blob/master/examples/HandlePortal/HandlePortal.ino)

Hieromon, 2020-02-27:
You can experience it using the following modification:

1. Modify the **handleGPIO**
```cpp
void handleGPIO() {
  String query = server.arg("v");
  if (query == "getdata") {
    sendJSON();
  }
  else {
    if (query == "low")    
      digitalWrite(BUILTIN_LED, LOW);
    else if (query == "high")
      digitalWrite(BUILTIN_LED, HIGH);
    sendRedirect("/");
  }
}
```

2. Adds the **sendJSON** as follows to respond with JSON
```cpp
void sendJSON() {
  String json = "{";
  json += "\"userId\":1,";
  json += "\"id\":1,";
  json += "\"title\":\"delectus aut autem\",";
  json += "\"completed\":false,";
  json += "\"Pin1\":\"" + String(digitalRead(Pin1) == HIGH ? "High" : "Low") + "\",";
  json += "\"Pin2\":\"" + String(digitalRead(Pin2) == HIGH ? "High" : "Low") + "\"";
  json += "}";
  server.send(200, "text/json", json);
  json = String();
}
```

shaikhfk, 2020-03-04:
Thanks alot Sir it worked very well

closed 2020-03-04 by shaikhfk