Hieromon/AutoConnect #186: Invalid catalog list:NoMemory error

wojtekdera, 2020-02-26:
Hello, I'm trying to use Auto Connect with Update feature but I'm getting always the same error:
Invalid catalog list:NoMemory

It appears always  when I put some *.bin files in bin folder.

**Message from the server:**
DEBUG:__main__:b'[{"name": "1.bin", "type": "bin", "date": "02/26/20", "time": "19:19:53", "size": 971120}]'
212.180.180.27 - - [26/Feb/2020 23:50:53] "GET /_catalog?op=list&path=/usr/home/wojtek/update/bin HTTP/1.1" 200 -

**Message from Autoconnect/Update site (http://192.168.1.169/_ac/update)**
Invalid catalog list:NoMemory

**Message when I delete *.bin files:**
Invalid catalog list:InvalidInput

**Message from console:**
23:50:53.469 -> [D][HTTPClient.cpp:1025] connect():  connected to mySite.pl:6123
23:50:53.469 -> [D][HTTPClient.cpp:1158] handleHeaderResponse(): code: 200
23:50:53.469 -> [D][HTTPClient.cpp:1161] handleHeaderResponse(): size: 90
23:50:53.469 -> [D][HTTPClient.cpp:361] disconnect(): still data in buffer (10), clean up.
23:50:53.509 -> 
23:50:53.509 -> [D][HTTPClient.cpp:370] disconnect(): tcp stop
23:50:53.509 -> 
23:50:53.509 -> [D][WiFiClient.cpp:509] connected(): Disconnected: RES: 0, ERR: 128

**files on server:**
[image]

Did you had this same error?
I can't make it work, please help me ;)

Hieromon, 2020-02-27:
> `Invalid catalog list:NoMemory`

It was produced by ArduinoJson, also `Invalid catalog list:InvalidInput` is same. ArduinoJson failed to parse the responded JSON from the server.
An updateserver.py is correctly generating the response JSON, but it has maybe corrupted during transmission.
You can take the responded JSON data natively on the Serial by `AC_DEBUG` turning on. Uncomment `AC_DEBUG` in the `AutoConnectDefs.h` header included in the library to enable debug logging.
https://github.com/Hieromon/AutoConnect/blob/ebb95cd8e5cae8e9823b0dc9cdcc32649c071a15/src/AutoConnectDefs.h#L13-L14

And about an `https`, AutoConnect uses WiFiClient and HTTPClient, it is not HTTPS. They do not support HTTPS. Under HTTPS we need to use the WiFiClientSecure class, but AutoConnect does not support it.

wojtekdera, 2020-02-27:
Thank you, so it can't work with https and I have to give it up?

**I did that Debug thing, now the message says:**
`[AC] Host:192.168.1.169,URI:/_ac/update,generated:/_ac/update, allocated
11:32:15.484 -> [AC] Host:192.168.1.169,URI:/_ac/update,already allocated
11:32:15.484 -> [AC] CB in AHEAD /_ac/update
11:32:15.484 -> [AC] Query mySite.pl:6123/_catalog?op=list&path=/usr/home/wojtek/update/bin
11:32:15.484 -> [D][HTTPClient.cpp:1025] connect():  connected to mySite.pl:6123
11:32:15.524 -> [D][HTTPClient.cpp:1158] handleHeaderResponse(): code: 200
11:32:15.524 -> [D][HTTPClient.cpp:1161] handleHeaderResponse(): size: 120
11:32:15.524 -> [AC] Update server responded:[AC] JSON:NoMemory
11:32:15.524 -> 
11:32:15.524 -> [D][HTTPClient.cpp:361] disconnect(): still data in buffer (19), clean up.
11:32:15.524 -> 
11:32:15.524 -> [D][HTTPClient.cpp:370] disconnect(): tcp stop
11:32:15.524 -> 
11:32:15.558 -> [D][WiFiClient.cpp:509] connected(): Disconnected: RES: 0, ERR: 128
`

Hieromon, 2020-02-27:
An HTTP response body is null. It is not an AutoConnect problem.
What happens if you access directly to the URL with query parameters from a web browser without using the ESP module? Did you see the JSON content like the below **on your web browser** with HTTP GET /_catalog?op=list&path=/usr/home/wojtek/update/bin?

```json
[{"name": "Elements.ino.nodemcu-32s.bin", "type": "bin", "date": "06/22/19", "time": "23:25:26", "size": 1106400}, {"name": "Elements.ino.nodemcu.bin", "type": "bin", "date": "05/20/19", "time": "23:58:58", "size": 472528}, {"name": "FileUpload.ino.nodemcu.bin", "type": "bin", "date": "04/17/19", "time": "21:29:39", "size": 409008}, {"name": "FSBrowser.ino.nodemcu.bin", "type": "bin", "date": "04/17/19", "time": "21:59:24", "size": 409952}, {"name": "HelloWorld.ino.nodemcu.bin", "type": "bin", "date": "04/17/19", "time": "22:10:07", "size": 408480}, {"name": "mqttRSSI.ino.nodemcu.bin", "type": "bin", "date": "04/17/19", "time": "21:27:51", "size": 448720}, {"name": "Simple.ino.nodemcu.bin", "type": "bin", "date": "09/19/19", "time": "00:26:01", "size": 425296}, {"name": "Update.ino.esp32.bin", "type": "bin", "date": "05/16/19", "time": "01:58:22", "size": 942048}, {"name": "Update.ino.nodemcu-32s.bin", "type": "bin", "date": "06/22/19", "time": "23:37:59", "size": 1134928}, {"name": "Update.ino.nodemcu.bin", "type": "bin", "date": "06/22/19", "time": "12:36:56", "size": 471536}]
```

wojtekdera, 2020-02-27:
http://mySite.pl/_catalog?op=list&path=. 
and 
https://mySite.pl/_catalog?op=list&path=. 
**returns:**
statusCode: 404

Why is that? Is this a correct path?
I use node-red (get method) and web browser with this same result

Screen from putty:
[image]

Hieromon, 2020-02-27:
Why do you specify "path=."?, I said it is the e.g.
Do you read the documentation? https://hieromon.github.io/AutoConnect/otaserver.html#update-server-for-the-autoconnectupdate-class

wojtekdera, 2020-02-28:
Yes I tried to red it for a few days. And Because this was the example. 
**The path:** 
https://mySite.pl/_catalog?op=list
http://mySite.pl/_catalog?op=list
http://mySite.pl/_catalog

also doesn't work.
The server side doesn't show any entries at all.

wojtekdera, 2020-02-28:
http://ipaddress:6123/_catalog?op=list
**shows** 
[{"type": "file", "name": "updateserver3.py"}, {"type": "file", "name": "updateserver2.py"}, {"type": "directory", "name": "bin"}]
but with my domain name doesn't work (maybe because https?)

**Server side shows:**
[image]

but still it isn't the forder with *.bin files I specified :/

Hieromon, 2020-02-28:
The update server works fine, your specified path is wrong. If you specify a relative path you may get some different results. However, this is only advice, not a solution.
The updateserver.py is implemented by the http.server module in the Python library. The inability to access this simple HTTP server from a browser is a problem in your network environment.
First, make sure that the Python [`http.server`](https://docs.python.org/3.7/library/http.server.html#module-http.server) library module works properly in your network environment.

> but with my domain name doesn't work (maybe because https?)

The reason that FQDN can not be resolved is that DNS is not transparent. http and DNS are different protocols. Or are you using DNS over HTTPS?, I do not know your network environment. 
What about checking DNS with nslookup from a different host?
In any case, the solution should be done on your own. The cause is in your network.

wojtekdera, 2020-02-28:
Thank you, I will try to find the solution in my network then. I will post a solution if there will be some.

Hieromon, 2020-02-28:
Ok, this topic keeps open until updateserver.py responds successfully in your network environment.

wojtekdera, 2020-03-09:
Sorry, I'm unable to setup this solution on my shared https server. I think it is something simple with https, but I can't find it and I don't have more time to do this. 
I used solution from your other post:
[https://github.com/Hieromon/AutoConnect/issues/127#issuecomment-531697629](url)
using normal httpUpdate.

Thank you, and still - AutoConnect is a great plugin!

closed 2020-04-26 by Hieromon