Hieromon/AutoConnect #188: Get wifi ssid and password?

Nagaraj4896, 2020-03-18:
@Hieromon **How do i get wifi ssid and password from autoconnect library?**
I want to get ssid and password from flash and start wifi with that credentials like below
**portal.begin(''saved ssid',"saved password",1000)**

Hieromon, 2020-03-18:
You can read credentials from the flash like as following code. (A part of the sketch)
It has two ways to read credentials.

1. Read all credentials
	```cpp
	AutoConnectCredential credt;
	station_config_t  credential;
	bool found = false;
	for (uint8_t i = 0; i < credt.entries(); i++) {
	  if (credt.load(i, &credential)) {
	    if (!strcmp((const char*)credential.ssid, "SSID_YOU_WANT"))) {
	      found = true;
	      break;
	    }
	  }
	}
	if (found) {
	  portal.begin((const char*)credential.ssid, (const char*)credential.password, 1000);
	  ...
	}
	```

2. Read a specified credential
	```cpp
	AutoConnectCredential credt;
	station_config_t  credential;
	if (credt.load("SSID_YOU_WANT", &credential)) {
	  portal.begin((const char*)credential.ssid, (const char*)credential.password);
	}
	```

AutoConnectCredential cannot be declared global in either case. Assign as a local variable in the function.

Hieromon, 2020-04-26:
Close due to no activity record for one month.
Reopen when new reports are posted.

closed 2020-04-26 by Hieromon

alvaroluiz2, 2020-09-30:
please I want too, to get de ssid and password for to use  in telegram Bot, or you have another way for to use in  myBot.wifiConnect(ssid, pass); //Conexao na rede wifi. 
Thank for your help.
AlvaroLuiz2@gmail.com (São Paulo-Brazil)