Hieromon/AutoConnect #189: Show Live Video

annak1n, 2020-03-26:
Hi..
Could you help me.
I can not connect the camera to ESSP 2 AI_THINKER
And display it using your web server.

Hieromon, 2020-03-27:
How did you implement the Sketch?
If you use the **CamaraWebServer.ino** example from the arduino-esp32 core as it is, you will not be able to get it to work. AutoConnect **cannot** coexist with the handling **startCameraServer** function used in CamareWebServer example.
AutoConnect uses [WebServer library of the arduino-esp32 core](https://github.com/espressif/arduino-esp32/tree/master/libraries/WebServer) internally. On the other hand, the startCameraServer function applies the [HTTP Server component of ESP-IDF](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/protocols/esp_http_server.html) natively. The two are different, and they **cannot** communicate for http using the **same port**.

So, declare AutoConnect object with the other than 80, 81 port in the Sketch. (Port 80 and 81 is already used by the camera server)

```cpp
#include <WebServer.h>
#include <AutoConnect.h>
WebServer   server(82);
AutoConnect portal(server);

void setup() {
  ...
  portal.begin();
  startCameraServer();
  ...
}

void loop() {
  portal.handleClient();
  // delay(10000); <-You can't do this. It blocks the WebServer behavior AutoConnect depends on.
}
```
Then, AutoConnect root path will be `YOUR_IP_ADDRESS:82/_ac`.
However, this workaround is not perfect. Most operating systems use HTTP port 80 for captive portal detection. Probably the sketch above will not be auto-popup the portal screen.
The correct way is to be able to specify any port number with the startCameraServer function. Besides that, you need to hack the camera_index.h which is gzip-compressed. It's too much hassle to it.

**ESP32-CAM with AutoConnect**
[image]

annak1n, 2020-03-30:
Good afternoon.
Thank you for quick answer.
I will try as you said.
I want to make my child a toy with control via a smartphone.
Of all the examples on the Internet, updating your ESP32 via wifi only your firmware works correctly.
In my project I want to use the L298n driver which will drive 2 servo motors.

annak1n, 2020-04-09:
Sry for stupid question..
But u can share Lib and full this exemple.. Cuz i try to make that u say and i ahve many error.

Hieromon, 2020-04-10:
The above sketch sample does not automatically pop up with the captive portal, but other AutoConnect features work. As I mentioned above, ESP32-CAM will work except as the port used by the camera server and the port used by AutoConnect (which is also the WebServer port) conflict. This restriction is not to using AutoConnect. If you use the WebServer library and startCameraServer at the same time, this restriction will remain.
The cause of your sketch error should not be AutoConnect. If you think that AutoConnect is the cause, please provide me the information according to the [documentation guide](https://hieromon.github.io/AutoConnect/faq.html#4-reports-the-issue-to-autoconnect-github-repository) so that I can reproduce the problem.

Hieromon, 2020-04-30:
Close due to no activity record for one month.
Reopen when new reports are posted.

closed 2020-04-30 by Hieromon