Hieromon/AutoConnect #190: How can I manually put WiFi credentials

Ferni492, 2020-04-01:
I need to add wifi networks "manually" to the library, in the documentation it speaks but no example is given .

Hieromon, 2020-04-01:
What "manually" means, and you didn't explain it. I don't understand what to do to help you.
However, the ordinary procedure will step by Sketch with hardcoded like as:

```cpp
#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <AutoConnect.h>

void setup() {
  delay(1000);
  Serial.begin(115200);
  Serial.println();

  AutoConnectCredential credt;
  station_config_t apConfig;
  memset(&apConfig, 0x00, sizeof(apConfig));  // For DHCP

  // Put SSID
  strncpy((char*)apConfig.ssid, "YOUR AP SSID", sizeof(apConfig.ssid));

  // Put Password
  strncpy((char*)apConfig.password, "YOUR AP password", sizeof(apConfig.password));

  // Put BSSID value of your SSID:
  // You must set the correct value of BSSID to match the specified SSID.
  // If you don't, the AutoConnect can't establish to the SSID.
  apConfig.bssid[0] = 0x??; // Change the correct value
  apConfig.bssid[1] = 0x??; // Change the correct value
  apConfig.bssid[2] = 0x??; // Change the correct value
  apConfig.bssid[3] = 0x??; // Change the correct value
  apConfig.bssid[4] = 0x??; // Change the correct value
  apConfig.bssid[5] = 0x??; // Change the correct value

  if (credt.save(&apConfig))
    Serial.println("Ok");
  else
    Serial.println("Save error");
}

void loop() {}
```

closed 2020-04-01 by Hieromon

LorenzoCerda, 2020-05-10:
Hi... just a little question here (I am little confused). Would this sketch example https://github.com/Hieromon/AutoConnect/issues/190#issuecomment-607056825 
create a new entry (and save it to the eeprom) each time the ESP boots?

Thank you in advance.

Hieromon, 2020-05-11:
@LorenzoCerda 
>create a new entry (and save it to the eeprom) each time the ESP boots?

It depends on the SSID. The key for a credential is the SSID. If the same SSID already exists, it will be overwritten. In that case, the new entries are not duplicates.

LorenzoCerda, 2020-05-11:
Ah, excellent! And thank you again, for your reply and a great library!

LorenzoCerda, 2020-11-05:
Hi Hieromon,

   Sorry to come back after all this time. I still don't get it work _as I want_. Let me explain. With the code you provided I get the credential saved automatically, so I don't need to insert it in the Portal. But this is half of what I need. 

  My case is the following: I use AutoConnect so someone else will use the ESP32 afterwards (I use a ESP32 btw.. not a ESP8266). But before giving the ESP32 to the third party, I check that it does what it should do. To do that, I connect the ESP32 to my own Wifi, check it and then I give it to the third party (then he or she will use AutoConnect to configure the ESP to connect to his/her Wifi). What I want to achieve is that I don't need to connect to the ESP32 (with my phone, for example), insert the AP password (the one that allows me to connect to the ESP from my phone), then sign-in, search for my Wifi, select it, insert its password and then restart. Instead I want that just after I flashed the ESP, I boot it and it should connect to my Wifi. It is ok for me if it doesn't connect at the first time, since I understand that AutoConnect will try to connect to the last connected Wifi (which doesn't exist for the first boot). 

   What I am getting now if the following: I get the credential saved in the ESP from the flashing. So I saved the steps of looking for my Wifi and inserting the password. Unfortunately, the ESP does not connect to the saved Wifi when I do the first boot (or a second one). Instead, I have to access the ESP from my phone (step I wanted to skip), I now go to the "Open SSIDs" menu and I find my Wifi network saved (the credential). I then can choose it and it will connect without asking me the password. The ESP will restart and everything will go as expected.

  I think I am missing some other configuration, but I don't know which one that is.

Just in case, here it is what I get from the serial:

`AC Address: MAC:240AC45C17A8`
`Credencial (Wifi) por defecto grabada OK.`
`[E][WiFiSTA.cpp:219] begin(): connect failed!`
`C.P. started, IP:172.217.28.1`
`Connected to `
`IP address:     0.0.0.0`
`WiFi.status() != WL_CONNECTED  -->>  Time information skipped.`
`Sunday, January 00 1900 00:00:00`
`Waiting test command from serial...`
` `
`Done waiting.`

  As usual, any help is highly appreciated. 

  Thanks again!