Hieromon/AutoConnect #191: ESPAsyncwebserver - Question

nelsontechnologies, 2020-04-02:
Is there any advantage to using espasyncwebserver instead?  If so, how would you implement it?

Thanks!

Hieromon, 2020-04-03:
Not blocking the ESP module process acting as a server to respond to HTTP requests addresses concurrent processing needs. It brings benefits but requires developer skills.
You can see it in the posted issue fo the ESPAsyncWebServer repository. Many people do not understand what "asynchronous" means.
I have developed AyncAutoConnect not implemented AutoConnectAux, but no plan to publish the repository, yet.

closed 2020-04-03 by Hieromon

v-c, 2020-06-29:
Any chance you will reconsider and publish AsyncAutoConnect?

Hieromon, 2020-09-24:
@v-c I hope to publish someday but it's still incomplete. In particular, there are few problems with the method of migrating WiFi connection waiting- process from while loop to event-driven.