Hieromon/AutoConnect #192: autoconnect_httpport not working

nelsontechnologies, 2020-04-02:
I am trying to change the port for the entire system to 54115.  I have changed it in the defs file and recompiled and still resorts to port 80.  
If I manually add it (ESP8266WebServer server(54115);) in the sketch it works but captive portal will not automatically pull up.  You have to manually goto the softap ip address and add the port to the end of the ip address.  How can this be changed with the autoconnect_httpport definition and can it automatically pull up on captive portal?

Hieromon, 2020-04-03:
> but captive portal will not automatically pull up.

Of course, depending on the client's OS behavior. Portal page automatic pop-up works with the trick of a client OS such as the iOS etc and does not pop up unless your client device's OS uses port 54115 for captive portal detection. Most operating systems do not scan all TCP ports for the captive portal detection with HTTP.
In particular, 54115 is a dynamic port range, and I think there are no uses for detecting the captive portal, and most devices use port 80 to detect the captive portal.

> manually goto the softap ip address and add the port to the end of the ip address. 

Yes, in other words, the client device needs to specify explicitly a port 54115 for detection to Internet availability when WiFi connected, and GET with an HTTP. The server simply responds to the request. If you need to use port 54115 for the ESP8266WebServer, need a TCP listener on the ESP8266 Server that listens to port 80 to be able to answer the captive portal detection from client devices.
Does this make sense?

> autoconnect_httpport not working

You misunderstand the mechanism of the captive portal in TCP sessions. It doesn't matter AutoConnect.

closed 2020-04-03 by Hieromon

nelsontechnologies, 2020-04-03:
Thank you. Like Apache, you can add a listener line for a separate port so it can listen on more than one port at a time. Can this be done so it can listen to port 80 and 54115? It’s ok to respond to both Behind the firewall. Just need to be able to answer to different port beyond the firewall.

Brad Nelson

Owner / Senior Developer
[Logo]<https://www.nelsont.com>

Nelson Technologies & Consulting, LLC.
361 Southwest Drive, Suite 170
Jonesboro, AR. 72401
870 219 1985
brad@nelsontc.com

www.nelsontc.com<http://www.nelsontc.com>
[Facebook icon]<https://www.facebook.com/nelsontcllc>   [LinkedIn icon] <https://www.linkedin.com/in/nelsontcllc/>

The content of this email is confidential and intended for the recipient specified in message only. It is strictly forbidden to share any part of this message with any third party, without a written consent of the sender. If you received this message by mistake, please reply to this message and follow with its deletion, so that we can ensure such a mistake does not occur in the future.

Closed #192<https://github.com/Hieromon/AutoConnect/issues/192>.

Hieromon, 2020-04-03:
It can not, and is a limitation of ESP8266WebServer. ESP8266WebServer has implemented only a single port listener using LwIP.
You can run multiple instances of ESP8266WebServer for each port using the Custom class, but AutoConnect will not work across ESP8266WebServer instances and does not port redirection.

nelsontechnologies, 2020-04-03:
Ok thank you. That makes sense. On the AUTOCONNECT_HTTPPORT definition, should you not be able to just change the definition in there, recompile and reboot and it start listening on that port?