Hieromon/AutoConnect #193: The sketch does not continue without wifi

FRANAIRBUS, 2020-04-04:
Good afternoon Hieromon,
First thank your library and the great work you have done in the program and in the documentation.
Thanks to people like you who contribute their knowledge altruistically, the world can move a little faster and people like me with some small programming knowledge can make codes adapted to our needs.

I have been adapting my code to your library for a few months and it is great I even have computers installed since December or November that are working without any problem although I have had to adapt it to my needs little by little with trial and error to reach an optimal point of what I need.

My program is based on your AutoConnect / mqttRSSI_FS example and currently it has more than 39 variables among which there are blynkserver, sqlserver, probe calibration, high and low temperature alarms, time regulator and other configuration variables by SPIFFS.

There are some details to highlight that I have not seen a solution

1º If we lose the Wi-Fi connection, the loop stops preventing my system from working as a thermostat as well as a sensor (this has a good side and that is that it continues to continually search for the Wi-Fi network to connect but the bad side is that it does not allow it to continue running the entire program.)

2º In the restart due to a general power failure of the entire installation, AutoConnect looks for a Wi-Fi signal and if it does not find it in 2 attempts, it blocks and goes into captive portal mode. (I have solved this by increasing AUTOCONNECT_TIMEOUT to give the wifi router time to recover from the power and signal cut before AutoConnect searches. (If only the esp restarts there is no problem but if the wifi router was restarted together with the esp The latter should continue to search for a signal without blocking as it would in case no. 1.)

3º I know that you are working on it and I have seen a closed case about security on the web server and configuration page. (In the absence of username and password I have provisionally solved it by removing the link to / _ac and leaving only a logo without a link.
I think that any basic user who does a search in the ethernet network even if he finds the ip of the sensors and enters the webserver will only see the logo and if he does not know the address / _ac or / mqtt_setting he would not know how to get to the configuration menu.)

Thank you very much again and forgive my English.

Hieromon, 2020-04-10:
@FRANAIRBUS, Thank you for using AutoConnect.
I'm currently spending time working for the v1.1.5 release, so I haven't been able to clean up your requirements yet. Please give me some time. Maybe I can offer you a satisfactory solution.

FRANAIRBUS, 2020-04-11:
Thank you very much @Hieromon I wait with great admiration for the news and improvements in version v1.1.5, and again congratulate you for the great work.

[image]

Hieromon, 2020-04-15:
@FRANAIRBUS I have released v1.1.5 and will tackle your issue.
So, let us arrange your mentions to solve your problem.

1. The loop stops when WiFi disconnected
What does it mean the "loop stops"?
Can you post a minimal amount of code that reproduces the problem?

2. The behavior of Sketch dependent on router power loss and recovery
Can you briefly organize your requirements below?
    - How the Sketch doing when WiFi disconnected
    The ESP8266 Arduino core has supported reconnection after WiFi disconnection via the router in the first place. (also ESP32 Arduino core)

    - Your requirement for reconnection or state in disconnected
    What are your requirements to have your own reconnection mechanism without expecting automatic reconnection by the Arduino core?

Depending on your requirements it may be a mistake to adapt mqtt RSSI. The program structure may not match in the first place.
First, we need to know what the program structure that can fit your requirements. So I need to understand your requirements more clearly.

FRANAIRBUS, 2020-04-19:
Hello, forgive the delay in answering. I have been testing your new version and you can see some improvements

Case No. 1 Fixed if you disconnect from the wifi router the void loop continues and keeps searching for wifi until connected 

Case No. 2 If the ESP8266 is restarted and it does not find any router or access point it looks for and if it does not find any it enters AP mode (Void Loop does not continue) it simply searches for an access point before entering AP mode Automatically. I wish I could continue with my sketch after rebooting and have it search for wifi from time to time without stopping the void loop 

[image]

Case No. 3 Username and password for customer portal configuration. I will try this solution

`void setup() {
  config.authentication = AC_AUTH_DIGEST;
  config.scope = AC_AUTHSCOPE_PARTIAL;  // authentication for /mqtt_setting only
  config.username = "admin";
  config.password = "password";
  portal.config(config);
  portal.load(AUX_mqtt_setting);  // Loads pages that will be authenticated according to the "auth" key specified.
  portal.begin();
} `

Hieromon, 2020-04-20:
@FRANAIRBUS So now is your only requirement for Case No.2?
> I wish I could continue with my sketch after rebooting and have it search for wifi from time to time without stopping the void loop

It's a matter of how the sketch is implemented. Similar requests have posted to past ISSUEs.
https://github.com/Hieromon/AutoConnect/issues/153#issuecomment-557396970

Apparently, some people have different requirements for WiFi connection than I realized.
Even if the ESP module reaches into a range of a known access point, it will not connect to the found one without human intervention. Not automatically. Because You can handle the portal and your Sketch with AutoConnectAux via http://172.217.28.1/_ac manually. In this case, the ESP module cannot reach the internet, but Sketch can be running the loop() with AutoConnectConfig::portalTimeout & AutoConnectConfig::retainPortal.
Also, you can call AutoConnect::begin on-demand from inside the loop() by driving an external switch or timers in this situation as well.
https://hieromon.github.io/AutoConnect/advancedusage.html#on-demand-start-the-captive-portal
```cpp
AutoConnect portal;
AutoConnectConfig config;
unsigned long tm;
#define SCAN_INTERVAL (3*60*1000) // Attempts WiFi connection interval [ms], as you wish

void setup() {
  config.portalTimeout = 1; // Timeout limits as you wish
  config.retainPortal = true;
  portal.config(config);
  tm = 0;
}

void loop() {
  if (WiFi.status() != WL_CONNECTED) {
    if (millis() - tm > SCAN_INTERVAL) {
        portal.begin();
        tm = millis();
    }
  }
  portal.handleClient();
}
```

Is your request to change this behavior? For what?, What is the usage scenario for completing the sketch process within the loop() without a WiFi connection to the router?

FRANAIRBUS, 2020-04-20:
@Hieromon 

> Is your request to change this behavior? For what?, What is the usage scenario for completing the sketch process within the loop() without a WiFi connection to the router?

My program currently works only as an iot sensor
I also want to turn it into a thermostat, in other words it must control the starts and stops of a refrigerator regardless of whether there is a wifi connection to a router.
So I need if the router is disconnected or loses signal or turns off. the controller should continue to operate and not enter captive portal mode, it should not stop the normal operation of the (void loop) and should continue to search for the signal from the router from time to time.

This happens very well if once connected to the router it is disconnected and the esp does not restart since the (void loop) continues to work and continues to search for the wifi connection.

but in the case there is no wifi signal from the router and by chance the esp restarts, it stops waiting for a wifi connection and after a while it enters the captive portal so you have to go to the customer's facility to collect it manually with the router.

FRANAIRBUS, 2020-04-20:
[image]

once connected with saved credentials it is great and even if you disconnect from the router it continues to work and search for wifi at the same time.

On the other hand, with the router switched off, if I restart the ESP, a failed connection appears and enters AP mode.

[image]

I would like it to work the same with router turned off even if ESP is restarted.

Hieromon, 2020-04-20:
@FRANAIRBUS Perhaps, your requirement is not about the AutoConnect library specification, but about how to implement your sketch application? If so, do you understand clearly the logic of the `mqttRSSI`?

To reconnect to the router once the connection has been established, simply issue `WiFi.setAutoReconnect(true)` at the appropriate time. It's a feature of the ESP Arduino core, not AutoConnect. The documentation is [here](https://arduino-esp8266.readthedocs.io/en/latest/esp8266wifi/station-class.html#setautoconnect).

To try the connection with the saved credentials when the ESP module starts, simply issue **AutoConnect::begin** in the `setup()`.
To return control from **AutoConnect::begin** without entering captive portal state if no known router can be found, simply set **AutoConnectConfig::portalTimeout** to an appropriate value.
If you want to access the SoftAP of AutoConnect without using the captive portal, set **AutoConnectConfig::retainPortal** to the true, and you can access the AutoConnect menu at `http://172.217.28.1/_ac`.

In the `loop()`, If WiFi has disconnected, you can detect it by **WiFi.status**. Its detection can be a trigger to measure a timer using `millis()`. When the measurement time exceeds a certain value, use **WiFi.scanNetworks** to scan the SSID broadcast. Then use **AutoConnectCredential::load** to match the scan results with the stored credentials. If a matching credential is stored, use it to natively issue **WiFi.begin**.

We have all the APIs and the documentation needed to implement these behaviors.
What is the problem?

FRANAIRBUS, 2020-04-21:
@Hieromon thank you very much I think the problem is already solved and the older I am the more foolish I have flown.

> WiFi.setAutoReconnect(true)

True by default by simply omitting.

As for the timer, it did not work for a very simple reason in the code and I have not realized it until now and it is to hit me against the wall.

> Serial.print("WiFi ");
  if (portal.begin()) {
    config.bootUri = AC_ONBOOTURI_HOME;
    Serial.println("connected:" + WiFi.SSID());
    Serial.println("IP:" + WiFi.localIP().toString());
    delay(1000);
  }
  else {
    Serial.println("connection failed:" + String(WiFi.status()));  
   /*
      **while (1) {
      delay(100);
      yield();**
    }
    */ 
  }
> 

Simply deleting (while) fixed the problem and now continue void loop without wifi connection

>    // while (1) {}

Your time solution can now be used to reconnect.

> void loop() {
 if (WiFi.status() != WL_CONNECTED) {
    if (millis() - tm > SCAN_INTERVAL) {
        portal.begin();
        tm = millis();
    } }

Thank you very much for the inconvenience and for your help, we can now close this case if you wish and thank you very much for your code.

It only remains to prove

> void setup() {  
  config.authentication = AC_AUTH_DIGEST; 
  config.scope = AC_AUTHSCOPE_PARTIAL; // authentication for /mqtt_setting only 
  config.username = "admin"; 
  config.password = "password"; 
  portal.config(config); 
  portal.load(AUX_mqtt_setting); 
 portal.begin(); 
}

closed 2020-06-11 by FRANAIRBUS