Hieromon/AutoConnect #194: Not redirecting after connection

MidnightInAurora, 2020-04-07:
I have a small sketch I'm using to test AC and I'm having an issue, actually two.

When the sketch runs with no saved creds, it does come up as an AP and I can select that SSID, enter a password and get to the AC status screen.  From there, I click the hamburger and select the router SSID, enter a password and AC connects to it.

The serial monitor shows WiFi connected with a DHCP-assigned IP.  All is good.

However, AC never moves past the Connecting screen (the one with the animated pulsating blue dot.)

Everything I've read leads me to believe that I'd get a redirect to the status screen showing the DHCP-assigned IP from the router.

Not happening.

I've tried changing AC from the 172 network to mine - 192.168.10.8.  Again, nothing.

`#include <WiFi.h>
#include <WebServer.h>
#include <AutoConnect.h>

WebServer Server;

AutoConnect Portal(Server);
AutoConnectConfig Config;

void rootPage() {
  char content[] = "WIFI CONNECTED";
  Server.send(200, "text/plain", content);
}

void setup() {
  delay(1000);
  Serial.begin(115200);
  Serial.println();

  Config.autoReconnect = false;
  Config.autoSave = AC_SAVECREDENTIAL_NEVER;

  Config.apid = "RX-TEST";
  Config.psk = "xxxxxxxx";

//IPAddress 
//  Config.apip = IPAddress(192,168,10,8);

  Config.hostName = "RX-1917";

  Portal.config(Config);

  Server.on("/", rootPage);

  if (Portal.begin()) {
    Serial.println("WiFi connected: " + WiFi.localIP().toString());
  }
}

void loop() {
  Portal.handleClient();
}`

The major problem with not getting back to the status screen is I have no idea what the assigned IP is.  I won't have a serial monitor and may not have access to the router to check DHCP assignments.

So, is this something I'm overlooking or doing wrong?  Are my expectations misguided?  Am I being set up for disappointment?

I think I'm only supposed to open one issue at a time so the second minor one can wait.

Thanks everyone.

Jim

Hieromon, 2020-04-16:
The below you mentioned is normal behavior because the WiFi connection has established with WIFI_STA mode.

> However, AC never moves past the Connecting screen (the one with the animated pulsating blue dot.)

AutoConnet will not enter WIFI_APSTA mode after a successful WiFi connection. Therefore, the ESP module is a client of the router and there is no way to explicitly know the IP address assigned by DHCP dynamically. It is equivalent to the behavior of WiFi.begin.
I'm not sure what you expect, but did you consider using mDNS?

Hieromon, 2020-09-28:
Closed due to no activity. It will resume with a new post.

closed 2020-09-28 by Hieromon