Hieromon/AutoConnect #196: autoRise and immediateStart

dumarjo, 2020-04-14:
Hi,
First this is a very good piece of software !

I try to get the feature of AP mode on demand with the **immediateStart** and I have some problem. 

From the doc, if the **immediateStart** is set to true, the ESP will be in AP_STA independently of **autoRise** value. But from my test this is not working. 

Here a sketch to try this. 

```
#include <WebServer.h>
#include <AutoConnect.h>
WebServer Server;
AutoConnectConfig portalConfig;
AutoConnect      Portal(Server);
void rootPage() {
  char content[] = "Hello, world";
  Server.send(200, "text/plain", content);
}
void setup() {
  delay(1000);
  pinMode(0, INPUT);
  portalConfig.title = R"(Custum Portal)";
  portalConfig.autoRise = true;
  portalConfig.immediateStart = true;

  Portal.config(portalConfig);
  Serial.begin(115200);
  Serial.println();
  Server.on("/", rootPage);
}

void loop() {
	if (digitalRead(0) == LOW) {
		while (digitalRead(0) == LOW)
		  yield();
		sleep(1);
		Portal.begin();
	  }
	Portal.handleClient();
}
```
When I push the button the AP start and serve AutoConnect. If I change the value of autoRise, The AP is not started when I press the button. 

From what I lokked in the code, if the autoRise is not set to true, the AP will never get UP and running. 

Do I not understand the documentation correctly ?

Here the section of the doc:
```
immediateStart:
Disable the first WiFi.begin() and start the captive portal. If this option is enabled, 
the module will be in AP_STA mode and the captive portal will be activated 
**regardless** of AutoConnectConfig::autoRise specification
```
Can you clarify this ?

Regards,

Jonathan

Hieromon, 2020-04-16:
> From the doc, if the immediateStart is set to true, the ESP will be in AP_STA independently of autoRise value. But from my test this is not working.

Sorry, there is an incorrect description in the document.  Also, the description of the [Inside AutoConnect::begin](https://hieromon.github.io/AutoConnect/lsbegin.html) is insufficient.
As you pointed out, specifying autoRise takes precedence over immediateStart. Even if immediateStart is true, the captive portal will not launch if autoRise is false.
Correct the documentation.

Thanks for pointing out my mistake.

dumarjo, 2020-04-16:
Hi,

No problem !

We will probably you this project in our next project. I will also do some PR to fix some corner case I see over my tests. Probably in the next couple of month. Keep your good work.

Jonathan

closed 2020-04-16 by dumarjo

Hieromon, 2020-04-16:
P.S.
This is the current specification. But now it may need to be reviewed.
<table>
  <tr>
    <th rowspan="2">immediateStart</th>
    <th colspan="2">autoRise</th>
  </tr>
  <tr>
    <td>true</td>
    <td>false</td>
  </tr>
  <tr>
    <td>true</td>
    <td>Skip 1st-WiFi.begin<br>ESP module becomes SoftAP and the captive portal starts immediately.<br></td>
    <td>Not attempt WiFi connection.<br>Only WebServer will start in STA mode.</td>
  </tr>
  <tr>
    <td>false</td>
    <td>Attempts WiFi connection in STA mode.<br>In some cases, the autoReconnect may restore the connection even if 1st-WiFiBeing fails.<br>If the connection is completely lost, the captive portal will be launched.<br><b>This is the default.</b></td>
    <td>Attempts WiFi connection in STA mode.<br>In some cases, the autoReconnect may restore the connection even if 1st-WiFiBeing fails.<br>WebServer will start in STA mode.</td>
  </tr>
</table>

dumarjo, 2020-04-16:
Looks good to me
Jonathan