Hieromon/AutoConnect #202: portal.begin() get stuck

jmcasimar, 2020-04-15:
Hello Hieromon,

I have to congratulate you first, the work you have done is amazing. I have been using your library since a year ago with the 0.9.7 version. Now I am updating my codes to take advantage of the new features. Nevertheless, I cannot get running even the simplest examples, this is what I have:

Version of libraries:
AutoConnect 1.1.5
ArduinoJson 6.15.1
PageBuilder 1.4.0

Example code:
```
/*
  HelloWorld.ino, Example for the AutoConnect library.
  Copyright (c) 2019, Hieromon Ikasamo
  https://github.com/Hieromon/AutoConnect

  This software is released under the MIT License.
  https://opensource.org/licenses/MIT
*/
/*
  To experience this example, upload the JSON file which is style.json
  from the data folder. Its file contains the attributes for the Caption
  of AutoConnectText. You can change the elements for your realization.
*/

#if defined(ARDUINO_ARCH_ESP8266)
#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
typedef ESP8266WebServer WEBServer;
#elif defined(ARDUINO_ARCH_ESP32)
#include <WiFi.h>
#include <WebServer.h>
#include <SPIFFS.h>
typedef WebServer WEBServer;
#endif
#include <FS.h>
#include <AutoConnect.h>

#define HELLO_URI   "/hello"
#define PARAM_STYLE "/style.json"

// Declare AutoConnectText with only a value.
// Qualify the Caption by reading style attributes from the SPIFFS style.json file.
ACText(Caption, "Hello, world");

//AutoConnectAux for the custom Web page.
AutoConnectAux helloPage(HELLO_URI, "Hello", true, { Caption });
AutoConnect portal;

// JSON document loading buffer
String ElementJson;

// Redirects from root to the hello page.
void onRoot() {
  WEBServer& webServer = portal.host();
  webServer.sendHeader("Location", String("http://") + webServer.client().localIP().toString() + String(HELLO_URI));
  webServer.send(302, "text/plain", "");
  webServer.client().flush();
  webServer.client().stop();
}

// Load the attribute of the element to modify at runtime from external.
String onHello(AutoConnectAux& aux, PageArgument& args) {
  aux.loadElement(ElementJson);
  return String();
}

// Load the element from specified file in SPIFFS.
void loadParam(const char* fileName) {
  SPIFFS.begin();
  File param = SPIFFS.open(fileName, "r");
  if (param) {
    ElementJson = param.readString();
    param.close();
  }
  SPIFFS.end();
}

void setup() {
  delay(1000);
  Serial.begin(115200);

  loadParam(PARAM_STYLE);     // Pre-load the element from JSON.
  helloPage.on(onHello);      // Register the attribute overwrite handler.
  portal.join(helloPage);     // Join the hello page.
  Serial.println("HELLO");
  portal.begin();
  Serial.println("WOLRD");
  
  WEBServer& webServer = portal.host();
  webServer.on("/", onRoot);  // Register the root page redirector.
}

void loop() {
  portal.handleClient();
}
```
I only added two prints to your example sketch and the serial output that I get is:
```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
HELLO
```
It seems that function portal.begin() get stucked, I continue tracking this error changing the file Autoconnect.cpp using more prints to try to find what is happening and I think is something related to the timeouts because in the next line:
```
cs = _waitForConnect(_connectTimeout) == WL_CONNECTED;
```
If I change the _connectTimeout for something different to zero it continue working

But then I get stucked in this lines:
```
while (WiFi.status() != WL_CONNECTED && !_rfReset) {
    handleClient();
    // Force execution of queued processes.
    yield();
    // Check timeout
    if ((hasTimeout = _hasTimeout(_apConfig.portalTimeout))) {
        AC_DBG("CP timeout exceeded:%ld\n", millis() - _portalAccessPeriod);
        break;
    }
}
```
Again is a while with a timeout condition to break it.

I believed that probably was a bug for the latest version but then I checked the 0.9.7 version and I saw that this lines are exactly the same.

Do you have any idea what it could be?

Hieromon, 2020-04-16:
@jmcasimar Thank you for using AutoConnect.
AutoConnect::begin enters a **captive portal state** when a WiFi connection is not possible using the credentials stored by AutoConnect or ESP8266 itself. In **this state**, the AutoConnect::begin **does not return unless WiFi is connected or the portal times out**.
To help you understand this flow, the documentation refers to AutoConnect::begin internal logic.
Refs: https://hieromon.github.io/AutoConnect/lsbegin.html

[image]

I think the cause of the problem is that the WiFi connection is failing, but I would like to see the debug trace output by AutoConnect with [**AC_DEBUG**](https://hieromon.github.io/AutoConnect/advancedusage.html#debug-print) to clarify why AutoConnect::begin does not return. 
Let defines AC_DEBUG, and post the output log, please.

jmcasimar, 2020-04-21:
@Hieromon everything was working fine. I just was expecting the debug output of the portal appears in the serial monitor, in older version this output appears without the AC_DEBUG, so I got confused.

Thanks for your help.

closed 2020-04-22 by Hieromon