Hieromon/AutoConnect #203: Lightweight AutoConnect version

acontini, 2020-04-15:
@Hieromon first of all congratulation for the beautiful library. I'm using it with PlatformIO and I've noticed that it is using around 40% more flash memory when active compared to the usual WiFi and WebServer libraries alone. Is there a way to increase its efficiency and reduce the memory imprint?

Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]  12.7% (used 41464 bytes from 327680 bytes)
Flash: [=======   ]  70.4% (used 922915 bytes from 1310720 bytes)
esptool.py v2.6

Hieromon, 2020-04-16:
Lightweight!, Yes, most definitely. I want to reduce too.
Much of what occupies the Flash comes from contained STL and external libraries, but I don't have an effective solution to reduce it yet. Instead, AutoConnect has a way to detach the ArduinoJson. However, it is provisional and does not provide a fundamental solution to the problem.
Refs: https://hieromon.github.io/AutoConnect/wojson.html

Here, the following are the statistics when I build the sketch below with ESP32DevKit-C in my environment.
```cpp
#include <Arduino.h>
#include <WiFi.h>
#include <WebServer.h>
#include <AutoConnect.h>

WebServer  server;
AutoConnect portal(server);

void setup() {
  server.on("/", []() {
    String content = R"(
<!DOCTYPE html>
<html>
<head>
  <meta charset="UTF-8" name="viewport" content="width=device-width, initial-scale=1">
</head>
<body>
Place the root page with the sketch application.
</body>
</html>
    )";
    server.send(200, "text/html", content);
  });
  portal.begin();
}

void loop() {
  portal.handleClient();
}
```

1. No build flags present
    RAM:   [=         ]  12.5% (used 40976 bytes from 327680 bytes)
    Flash: [=======   ]  70.8% (used 928343 bytes from 1310720 bytes)

2. `build_flags = -DAUTOCONNECT_NOUSE_JSON`
    RAM:   [=         ]  12.5% (used 40904 bytes from 327680 bytes)
    Flash: [=======   ]  69.6% (used 911703 bytes from 1310720 bytes)

It doesn't decrease much...

Incidentally, for ESP8266, the result as below. ESP-IDF seems to have a larger object size than ESP8266 non-os SDK, or a strange ghost lurks who spits a lot of object code.
```
Checking size .pio\build\nodemcuv2\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [====      ]  37.4% (used 30640 bytes from 81920 bytes)
Flash: [====      ]  39.8% (used 415248 bytes from 1044464 bytes)
```

atheros, 2020-12-14:
I was looking forward using AutoConnect, however using BLE_scan example with added AutoConnect from master gave me 118% of flash space used :(

It seems AutoConnect adds ~ 500kb on ESP32

Hieromon, 2020-12-15:
@atheros 
It can be due to related libraries, which need to be portable and stable. If you have any doubts about the size of the base libraries, it should be posted to ESP-IDF.

>  118% of flash space used

Well, you should have the opportunity to change the partition table.
Also, BLE library is big. https://github.com/espressif/arduino-esp32/issues/1075

So, 500kb... which one?

[image]

[image]

atheros, 2020-12-15:
Adding Wifi.h, WebServer.h and AutoConnect to BLE_scan example project results in a binary build ~500kb larger.
If I have a bit more time tonight I'll investigate it more.

Hieromon, 2020-12-16:
@atheros 
As you can see from the screenshot above, the bloat of the compiled ELF is largely due to the implementation of the esp-idf library. That is clear from the inspection results by me. The AutoConnect text block is slight 50KB and is similar in size to the AutoConnect for ESP8266, which is based on the ESP8266 non-OS SDK.

Many contributors are working on this problem and are mainly discussing it at the [arduino-esp32 repository](https://github.com/espressif/arduino-esp32/issues/339). It relies heavily on the program structure of the esp-idf library and FreeRTOS, and we need to diagnose the source code of the esp-idf library to solve the problem. Even if you identify a bloated text block in a compiled ELF, it is unlikely that you will be able to step into FreeRTOS and change its structure. The same is true for C ++ STDLib.

For now, the practical workaround we can adopt is to adjust the partition size. And the contributors of the arduino-esp32 core provide an easy way to adjust the partition size, and the contribution is commendable.

atheros, 2020-12-16:
I did some tests yesterday and it turned out that BLE_scan + Wifi + WebServer serving a static page uses 103% of the flash, so AutoConnect uses additional 15%. Since I cannot fit BLE, WiFi and Webserver in the default partition, I guess making AutoConnect smaller won't help much :)

I do not want to play with partition size as I want it to be a plug and play solution, without re-configuring arduino environment and the esp32 itself.

h2zero, 2020-12-31:
Just found this repo, looks very nice, looking forward to trying it out!

I wanted to comment here for those of you who want to use BLE and WiFi and want to save space. Try using NimBLE, it uses half  the flash and ram of the default bluedroid library with nearly 100% backward compatibility. 

https://github.com/h2zero/NimBLE-Arduino

Hieromon, 2021-01-04:
@h2zero, Thank you a information.
Original Bluedroid used a memory pool, but esp32 uses osi_malloc () to allocate it dynamically from HEAP. (Maybe esp-idf inherits from the Bluedriod contained in Android 5.1) Original Bluedroid managed the size and number of each memory pool, but esp32 does not. (I don't like this implementation)
Therefore, after a certain amount of time, the problem of "HEAP fragmentation occurs and memory cannot be allocated" may occur.
Has NimBLE-Arduino already solved this point?

h2zero, 2021-01-05:
Hi @Hieromon, NimBLE (the core) creates it's own mempools for it's operations (size definable in config). The C++ wrapper, due to necessity of a multi-use-case library, does dynamically allocate from the heap. This has been minimized however, heap use and fragmentation is very small, due to some significant optimizations from the original. For instance, with the original library 1 connection as a client used over 10kb of ram, with NimBLE it's less than 1kb, depending on the number of attributes obtained from the peripheral.

So to answer the question;  almost :laughing: ,but it's better for sure.

If you decide to check it out I would appreciate any feedback (please note, a significant release is coming soon).

Hieromon, 2021-01-05:
@h2zero Thank you for your guidance.

> with the original library 1 connection as a client used over 10kb of ram, with NimBLE it's less than 1kb, depending on the number of attributes obtained from the peripheral

It's awesome! It will save many drowning in a small pond.
I will try it out and will report to you.

Thank you again.

h2zero, 2021-01-05:
Sounds good! Yes it's a significant reduction in resources, shocks many people haha.

I've been working on some stability and other improvements that I hope to release soon. Hopefully I'll get some time to test this repo soon as well.

Koxx3, 2021-03-16:
> optimizations

+1000 for NimBLE. Works great (even better in fact) and use wayyyy less storage/memory.
ESP32 (2 big OTA partitions of 1.9MB) with NimBLE + AutoConnect + TFT_eSPI = 88% for me

closed 2021-10-09 by Hieromon