Hieromon/AutoConnect #204: OTA translation not working

dmaixner, 2020-04-16:
Hello Hieromon,
this new builtin OTA is awesome, exactly what I needed. :)

I have one issue though, it seems translations are not working for OTA, maybe the problem is in `AutoConnect.h`: first is included `AutoConnectPage.h` and then `AutoConnectOTA.h`, but `AC_LABELS` is checked only in `AutoConnectPage.cpp`, which seems to be too late for OTA. If I move
```
#ifdef AC_LABELS
#include  AC_LABELS
#endif
```
to `AutoConnectPage.h`, it's working fine. Is it ok to create PR for this or does it need some more general solution?

Hieromon, 2020-04-17:
@dmaixner Hello,
Oh, I leaked its condition from the test case. Please PR.

Notes:
The [current master](https://github.com/Hieromon/AutoConnect/commit/2818c6cd022605e2a594acfecc2e26ab37eeb72f) is advanced than the [release](https://github.com/Hieromon/AutoConnect/commit/f6bac9ba9baa43a1007ae565c2c05c1479585437).
Please fork the latest before your PR.

dmaixner, 2020-04-17:
Created #205 for this, closing issue.
Thank you.

Hieromon, 2020-04-17:
@dmaixner Thank you very much!
I released v1.1.6.

closed 2020-04-17 by Hieromon