Hieromon/AutoConnect #208: Color/Image Enhacement

jmcasimar, 2020-04-21:
I would like to take this space to propose you a little enhancement. I am developing multiples IOT modules based on ESP8266/ESP32 this modules are installed in some reduced space (for example, my house). I have faced with the problem that internet connections failed, or that I changed my WiFi password or that my internet provider changed the router and I have to reconfigure all my modules. [AutoConenct](https://github.com/Hieromon/AutoConnect) makes this problem easy to resolve since I can use my cellphone and reconnect all my modules in a few minutes.

However as the number of modules increase and they have differents purposes the configuration of each module is also different, the problem is that if I have more than 2-3 portals open at the same time it is not easy to recognize what module I am configuring. One solution would be use a keyword to identify each module and name the SSID of the portal with this keyword inside, or maybe put this keyword in the header of the portal, but I was thinking to develop a color based solution, because I also want to feel that each module and its portal is unique. Actually to make this I have to modify the file [AutoConnectLabels.h](https://github.com/Hieromon/AutoConnect/blob/master/src/AutoConnectLabels.h) changing colors definitions. I was thinking to make a method that could modify the color without rewriting this file, similar to the way AutoConnectConfig works or maybe even integrate this option in the same class.

In the same line, I have thought the possibility to add images to the html pages using base64 format, to add, for example, a personal logo to the portal. Probably the memory could be a problem but it is a feature that also could be evaluated, I am not sure if it is even possible to do this with your PageBuilder library.

This enhacement is only for the appereance, I would like to improve the UI/EX and I know this is not a priority but eventually could be a nice configuration option. 

Would you consider add this enhacement at some time in the library?

Hieromon, 2020-04-22:
I understand your situation. It is possible to change the UI color scheme to identify individuals even with the current AutoConnect. It does not require modification of `AutoConnectLabels.h`.
You can use [PlatformIO IDE](https://platformio.org/platformio-ide) and [Microcost Visual Studio Code](https://code.visualstudio.com/) instead of the standard Arduino IDE.
AutoConnect's #define directives are considered so that you can use PlatformIO's excellent development environment to change the internal #define directive for each build unit without the library source code modification. (That can be individual for each project) This way, you can change the color scheme from build to build without modifying the source code.
Also, I think that your request will be satisfied by linking the color scheme by a unit to the menu title with [AutoConnectConfig::title](https://hieromon.github.io/AutoConnect/apiconfig.html#title) of the individual module.

**platformio.ini**
```ini
build_flags =
  -DAUTOCONNECT_MENUCOLOR_TEXT='"#ffa500"'
  -DAUTOCONNECT_MENUCOLOR_BACKGROUND='"#1a237e"'
  -DAUTOCONNECT_MENUCOLOR_ACTIVE='"#283593"'
```
See https://hieromon.github.io/AutoConnect/colorized.html
Refs: https://github.com/Hieromon/AutoConnect/issues/143#issuecomment-543182727

jmcasimar, 2020-04-22:
That will work, thank you! 

And about the option to add base64 images to the portal, do you think is possible?

Hieromon, 2020-04-22:
Where do you insert your mentioned base64 image additions in the AutoConnect-generated dynamic HTML? You should be able to get the AutoConnect generated HTML source code, and I would like to you indicate where to place it on the HTML source.

Its CSS structure of each part depends on each other. If you insert an unknown size base64 encoded image carelessly, the layout may be broken. How do you guarantee the static layout for AutoConnect?
Once that specific solution is presented, I will consider generalizing it as a library.

jmcasimar, 2020-04-29:
I want to put the image in the header of the page, so the maximun height it would be the height of the header. I sent you an [example](https://github.com/Hieromon/AutoConnect/files/4555088/AutoConnect.statistics.zip) attached.

Hieromon, 2020-04-30:
What you sent to me is not the answer as I intended the question.
If you want to extend it as a library we need an interface with user sketches, so how to ensure layout as an API is my question.
Your attached example is the width and height are only embedded, also fixed, and they are not specified arbitrarily. It's enough to modify AutoConnectPages.cpp individually. I have no intention of generalizing it as a library.

jmcasimar, 2020-04-30:
Ok, I understand what you want. 

I do not know AutoConnect as deeply as you, AutoConnect is also built on the top of PageBuilder that also is currently supported by you. The example I give you is only the expected result of the development of this API, and of course I would want to be able to modify this property inside some sketch.

Give me some time to study how AutoConnect works and think in a way to integrate this.

jmcasimar, 2020-05-01:
Here is what I have so far. I focus on creating a new AutoConnect Image Element, similar to the previously defined like Text, Input, File, CheckBox, etc. I took the Text element as basis to make this new element.

In [AutoConnectElement.h](https://github.com/Hieromon/AutoConnect/blob/master/src/AutoConnectElement.h) I added these lines:
```
#ifndef _AUTOCONNECTELEMENT_H_
#define _AUTOCONNECTELEMENT_H_

#include "AutoConnectElementBasis.h"
#ifdef AUTOCONNECT_USE_JSON
#include <ArduinoJson.h>
#include "AutoConnectElementJson.h"
using AutoConnectElement  = AutoConnectElementJson;
using AutoConnectButton   = AutoConnectButtonJson;
using AutoConnectCheckbox = AutoConnectCheckboxJson;
using AutoConnectFile     = AutoConnectFileJson;
using AutoConnectInput    = AutoConnectInputJson;
using AutoConnectRadio    = AutoConnectRadioJson;
using AutoConnectSelect   = AutoConnectSelectJson;
using AutoConnectStyle    = AutoConnectStyleJson;
using AutoConnectSubmit   = AutoConnectSubmitJson;
using AutoConnectText     = AutoConnectTextJson;
using AutoConnectImage     = AutoConnectImageJson; # NEW LINE
#define AUTOCONNECT_JSON_BUFFER_SIZE  256
#else
using AutoConnectElement  = AutoConnectElementBasis;
using AutoConnectButton   = AutoConnectButtonBasis;
using AutoConnectCheckbox = AutoConnectCheckboxBasis;
using AutoConnectFile     = AutoConnectFileBasis;
using AutoConnectInput    = AutoConnectInputBasis;
using AutoConnectRadio    = AutoConnectRadioBasis;
using AutoConnectSelect   = AutoConnectSelectBasis;
using AutoConnectStyle    = AutoConnectStyleBasis;
using AutoConnectSubmit   = AutoConnectSubmitBasis;
using AutoConnectText     = AutoConnectTextBasis;
using AutoConnectImage    = AutoConnectImageBasis; # NEW LINE
#endif // !AUTOCONNECT_USE_JSON

/**
 * Support declare the AutoConnectElement variable with reducing the
 * arguments. These macros declare the AutoConnectElement variable
 * with the same name as a "name" argument.
 */
#define ACElement(n, v)    AutoConnectElement n(#n, v)
#define ACButton(n, ...)   AutoConnectButton n(#n, ##__VA_ARGS__)
#define ACCheckbox(n, ...) AutoConnectCheckbox n(#n, ##__VA_ARGS__)
#define ACFile(n, ...)     AutoConnectFile n(#n, ##__VA_ARGS__)
#define ACInput(n, ...)    AutoConnectInput n(#n, ##__VA_ARGS__)
#define ACRadio(n, ...)    AutoConnectRadio n(#n, ##__VA_ARGS__)
#define ACSelect(n, ...)   AutoConnectSelect n(#n, ##__VA_ARGS__)
#define ACSubmit(n, ...)   AutoConnectSubmit n(#n, ##__VA_ARGS__)
#define ACStyle(n, ...)    AutoConnectStyle n(#n, ##__VA_ARGS__)
#define ACText(n, ...)     AutoConnectText n(#n, ##__VA_ARGS__)
#define ACImage(n, ...)    AutoConnectImage n(#n, ##__VA_ARGS__) # NEW LINE

#endif // _AUTOCONNECTELEMENT_H_
```
In [AutoConnectElementBasis.h](https://github.com/Hieromon/AutoConnect/blob/master/src/AutoConnectElementBasis.h):
```
typedef enum {
  AC_Button,
  AC_Checkbox,
  AC_Element,
  AC_File,
  AC_Input,
  AC_Radio,
  AC_Select,
  AC_Style,
  AC_Submit,
  AC_Text,
  AC_Image,
[... 223 lines cut ...]
void setup() {
  delay(1000);
  Serial.begin(115200);

  loadParam(PARAM_STYLE);     // Pre-load the element from JSON.
  helloPage.on(onHello);      // Register the attribute overwrite handler.
  portal.join(helloPage);     // Join the hello page.
  portal.begin();
  
  WEBServer& webServer = portal.host();
  webServer.on("/", onRoot);  // Register the root page redirector.
}

void loop() {
  portal.handleClient();
}
```

The result:

[image]

Please ignore the text on the upper left corner "NEW_CHANGE" I am trying to figure out how to add an image to the header in a dinamically way (that is my personal objective).

Please let me know what you think about it.

Hieromon, 2020-09-28:
@jmcasimar 

Such an extension that is not guaranteed to maintain the default style of AutoConnect is less versatile and leads to unexpected behavior. On the other hand, modifications similar to the UI decoration you pointed out can be achieved by modifying the library individually by a specific person with a specific purpose.
Therefore, there are currently no plans to support this extension as a library.

Hieromon, 2020-12-02:
No specific solution provided. There is only an idea.
Close.

closed 2020-12-02 by Hieromon